meshtastic/firmware #4340: [Bug]: GPS Checksum Failure HeltecV3 (workaround hint)

MrSiO, 2024-07-28:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

3.5.15

### Description

Hi.
My HeltecV3 with GPS module NEO6MV2 locked to satellites fine on version 2.3.11.  (tracker mode)

Since I first upgraded to 2.3.13 and later tried 2.3.15 I was not able to get any GPS Lock anymore.

Using Serial debug, In the logs (see below) I got a few GPS logs inluding:

`[GPS] 1 new GPS checksum failures`
Doubled check my wires, solders and all seemed fine.

As a workaround, I got it to work by changing the "Position broadcast Interval" from 30 to 10 and same thing for the "GPS update interval" setting.  Now I got sats lock and GPS coordinates are sent....  (???)

Unable to go back to 30 seconds wihout issues comming back.

### Relevant log output

```Shell
DEBUG | ??:??:?? 1 [GPS] Probing for GPS at 9600 
INFO  | ??:??:?? 3 [GPS] Found a UBlox Module using baudrate 9600
DEBUG | ??:??:?? 3 [GPS] Module Info : 
DEBUG | ??:??:?? 3 [GPS] Soft version: 6.02 (36023)
DEBUG | ??:??:?? 3 [GPS] Hard version: 00040007
DEBUG | ??:??:?? 3 [GPS] Extensions:0
WARN  | ??:??:?? 3 [GPS] Got NAK for class 06 message 39
WARN  | ??:??:?? 3 [GPS] Unable to enable interference resistance.
WARN  | ??:??:?? 4 [GPS] Got NAK for class 06 message 3B
WARN  | ??:??:?? 4 [GPS] Unable to enable powersaving details for GPS.
INFO  | ??:??:?? 6 [GPS] GNSS module configuration saved!
---------------------------------------------------------
DEBUG | ??:??:?? 6 [GPS] NMEA GPS time 2024-07-28 01:21:48
DEBUG | ??:??:?? 6 [GPS] Upgrading time to quality GPS
DEBUG | 21:21:48 6 [GPS] Read RTC time as 1722129708
DEBUG | 21:21:48 6 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | 21:21:48 6 [GPS] No GPS lock
DEBUG | 21:21:48 6 [GPS] onGPSChanged() pos@0 time=1722129708 lat=0 lon=0 alt=0
INFO  | 21:21:48 6 [GPS] updatePosition LOCAL pos@0 time=1722129708 lat=0 lon=0 alt=0
DEBUG | 21:21:48 6 [GPS] Setting local position: lat=0 lon=0 time=1722129708 timestamp=0
DEBUG | 21:21:48 6 [GPS] Node status update: 3 online, 6 total
---------------------------------------------------------------------------
DEBUG | 21:22:12 31 [GPS] WANT GPS=0
DEBUG | 21:22:12 31 [GPS] Initial GPS lock took 31s
DEBUG | 21:22:12 31 [GPS] GPS::powerState=STANDBY
DEBUG | 21:22:12 31 [GPS] Sleep Time: 60000
DEBUG | 21:22:12 31 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | 21:22:12 31 [GPS] onGPSChanged() pos@0 time=1722129732 lat=0 lon=0 alt=0
INFO  | 21:22:12 31 [GPS] updatePosition LOCAL pos@0 time=1722129732 lat=0 lon=0 alt=0
DEBUG | 21:22:12 31 [GPS] Setting local position: lat=0 lon=0 time=1722129732 timestamp=0
DEBUG | 21:22:12 31 [GPS] Node status update: 3 online, 6 total
---------------------------------------------------------------------
DEBUG | 21:23:12 91 [GPS] WANT GPS=1
DEBUG | 21:23:12 91 [GPS] GPS::powerState=ACTIVE
-----------------------------------------------------------------------
DEBUG | 21:23:43 122 [GPS] WANT GPS=0
DEBUG | 21:23:43 122 [GPS] GPS Lock took 31s
DEBUG | 21:23:43 122 [GPS] GPS::powerState=STANDBY
DEBUG | 21:23:43 122 [GPS] Sleep Time: 28963
DEBUG | 21:23:43 122 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | 21:23:43 122 [GPS] onGPSChanged() pos@0 time=1722129823 lat=0 lon=0 alt=0
INFO  | 21:23:43 122 [GPS] updatePosition LOCAL pos@0 time=1722129823 lat=0 lon=0 alt=0
DEBUG | 21:23:43 122 [GPS] Setting local position: lat=0 lon=0 time=1722129823 timestamp=0
DEBUG | 21:23:43 122 [GPS] Node status update: 3 online, 6 total
--------------------------------------------------------------------------
DEBUG | 21:24:13 152 [GPS] WANT GPS=1
DEBUG | 21:24:13 152 [GPS] GPS::powerState=ACTIVE
WARN  | 21:24:13 152 [GPS] 1 new GPS checksum failures, for a total of 1.
```

todd-herbert, 2024-07-28:
It looks like what you're seeing on 2.3.13 and 2.3.15 was also [expected behavior in 2.3.11](https://github.com/meshtastic/firmware/blob/2740a56944d6647e53320a7e235b61ae0207a143/src/gps/GPS.cpp#L1039-L1049), although that code probably wasn't doing what it claimed back then.

With #4070 and #4161, I didn't really try to reinterpret the purpose of that section of the code, just got it doing "what it says on the tin" again. I suspect the original intention was to allow GPS to sleep, if it is unable to get a GPS lock.

___

There's probably an argument for keeping the GPS awake and searching, even if it fails to find a lock within the "position broadcast interval", although there's possibility for conflict with other use cases here too.

@GPSFan, @Nestpebble
Any thoughts on this one?

jp-bennett, 2024-07-28:
@todd-herbert the timeout sleep is intended to save power if a node is in a location where it can't get a GPS lock. Searching for GPS is expensive for the power budget, so we try to avoid burning through the whole battery if it doesn't look like it'll get a lock.

todd-herbert, 2024-07-28:
Ah sorry, forgot to tag you! Thank you for the insight!

GPSFan, 2024-07-28:
The 39 and 3b commands should not have been NAKd.
DEBUG | ??:??:?? 3 [GPS] Soft version: 6.02 (36023)
DEBUG | ??:??:?? 3 [GPS] Hard version: 00040007
DEBUG | ??:??:?? 3 [GPS] Extensions:0
WARN  | ??:??:?? 3 [GPS] Got NAK for class 06 message 39  <<--------------------------
WARN  | ??:??:?? 3 [GPS] Unable to enable interference resistance.
WARN  | ??:??:?? 4 [GPS] Got NAK for class 06 message 3B  <<-------------------------
WARN  | ??:??:?? 4 [GPS] Unable to enable powersaving details for GPS.

So something is definitely wrong with the init sequence.
Checksum failures are usually due to buffer over runs when the receiver is not initialized correctly and many more sentences are being sent to Meshtastic than what it needs.

I'll look into it in the morning.

todd-herbert, 2024-07-28:
@GPSFan Just keep in mind too that this is reported on 2.3.15, prior to #4161, although I'm not sure if it's relevant to those init sequence logs.

GPSFan, 2024-07-28:
I would suggest as a next step to update to a later version of Meshtastic. I just tested with 2.3.16 and saw no such issue with NAKs for the init messages. Also try a factory reset as some settings saved to flash in the receiver may have messed things up.
Firmware 6.02 is very old for a Neo-6 and should be updated to 7.03. I also only have a Neo-6 with firmware 7.03 so there may be some corner case with that firmware. You should also look at using a newer GPS receiver, the Neo-6 is outdated and there are lots of newer parts with better performance, better price and better power consumption.

MrSiO, 2024-07-28:
> I would suggest as a next step to update to a later version of Meshtastic. I just tested with 2.3.16 and saw no such issue with NAKs for the init messages. Also try a factory reset as some settings [...]

Will try the factory reset and updgrade to 2.3.16.

But about the NEO firmware update it seems this not an option... Am I wrong? [image]

GPSFan, 2024-07-28:
No, you are correct, the Neo-6M is ROM based. My mistake. 7.03 was the last ROM version.

MrSiO, 2024-07-28:
> I would suggest as a next step to update to a later version of Meshtastic. I just tested with 2.3.16 and saw no such issue with NAKs for the init messages. Also try a factory reset as some settings [...]

Hi. Thx for your support.

Where can I download 2.3.16?
From the repo (https://meshtastic.org/downloads/) I can download 2.3.15 or the Alpha 2.4.X... no traces of 2.3.16

GPSFan, 2024-07-28:
2.3.16 is the version I have on my test machine, I built it myself from source and hasn't been updated in a while, 2.4.1 is out, try that.
https://github.com/meshtastic/firmware/releases/tag/v2.4.1.394e0e1

GPSFan, 2024-07-28:
Just built and tested the latest 2.4.1 and it behaves as 2.3.16 WRT the GPS init for a Neo-6 fw 7.03.

MrSiO, 2024-07-28:
Factory reset followed by the installation of 2.4.0 worked for me!  Problem solved!

GPSFan, 2024-07-29:
Glad to hear it!!

closed 2024-07-29 by thebentern