meshtastic/firmware #4341: Help needed, heartbeat led not blinking

ponzano, 2024-07-28:
Hello! I’m developing a new board for Meshtastic based on Nrf52840 and ebyte e22900m22s (sx1262). I got the first prototypes built and successfully compiled and flashed the adafruit bootloader. I’m now trying to add support in the Meshtastic firmware adding the variant files (platformio.ini, variant.h, variant.cpp) and it’s being executed correctly. I’m able to connect to the board using both Bluetooth and usb and send messages to the mesh. There’s a small issue I can’t figure out, the status led that is supposed to flash once every second, like on other boards, is kept turned on and doesn’t blink. I'm almost certain it's not a hardware issue as it behaves and blinks (and fades) correctly when in DFU mode or during flashing. I tried changing the #define LED_STATE_ON but both 0 and 1 give the same result. If I disable heartbeat led from the app it also doesn’t change the behavior. I defined only 1 led in variant.h, and if in variant.cpp I write pinmode(..) and digitalwrite(pin, LOW) or ledoff(pinled_1) it seems to turn off the led briefly at startup. Is there some other file I should edit ?
https://github.com/user-attachments/assets/f89455f3-ad60-4563-8ae4-8eb2cca6dc23

ponzano, 2024-07-29:
After an afternoon of fiddling, I could get the led blinking BUT only if I tweak the adc_multiplier so that the battery voltage is always read above 4.2v. If it goes at or below 4.2, the debug reports ‘loss of power in powered’ and the led stops blinking (and is kept on) even if usb is not plugged in. 
Does anyone know why this happens?

rjDipcord, 2024-07-31:
With the LED staying _off_ when the board _thinks_ the battery voltage is low and only flashing when it _thinks_ it is high, it sounds to me like there is some sort of power-saving feature at play here. I would also suffice to say the LED output is inverted, and the logic intends to keep the LED in the _off_ state when the battery is low.

ponzano, 2024-08-06:
For future reference, I got it working correctly but adding in variant.h: 

#ifndef _VARIANT_RAK4630_
#define _VARIANT_RAK4630_

#define RAK4630

I'm now looking at what this does in the firmware to adapt it to my variant.

closed 2024-08-06 by ponzano