meshtastic/firmware #4346: GPS Inaccurate Across Multiple Platforms 2.4.0[Bug]:

BobsBlueNorth, 2024-07-29:
### Category

Hardware Compatibility, Other

### Hardware

T-Beam, Rak4631

### Firmware Version

2.4.0

### Description

I have been moving up in firmware as it has been getting released. Somewhere along the line the GPS system got out of wack. I am running 2.4.0 on multiple RAK 4631s, multiple T-Beam 1.1s and Wireless Trackers 1.1. The only one that seems to be tracking correctly is the Heltec. Both the RAKs and T-Beams are off by about a 1/4 of a mile and all of them have the exact same coordinates down to the last integer. I find that highly suspect.

Can anyone else confirm this?

I rolled a T-Beam back to 2.3.2 and the issue seems to be gone.

BobsBlueNorth, 2024-07-29:
I take that back. 2.3.2 just failed as well. It was doing great for about 5 to 10 minutes. It was getting repeated locks and reporting my position to within 20 feet then it jumped about 1/4 mile away to the exact same spot as all the other nodes. 

Some kind of of calculation error? Could this be an app related issue? I am running Android 2.3.15

BobsBlueNorth, 2024-07-29:
The only node that is reporting and displaying correctly is a remote node that is running 2.2.XX firmware and hasn't been updated in about 10 months.  Even when I put in a fixed position dataset into a device running 2.4.0, it reports and/or displays wrong.

jp-bennett, 2024-07-29:
That's a feature, not a bug! =D It sounds like Position Precision got turned on for those devices. It's a per-channel setting to do some position masking. You can set the offending devices back to full precision to fix it.

BobsBlueNorth, 2024-07-30:
Besides the CLI, how do I change that? I don't see that function in the web app nor the android app.

BobsBlueNorth, 2024-07-30:
I see it now, it's the Low, Medium, High Precision inside the Channel configuration screen.

rjDipcord, 2024-07-31:
> I see it now, it's the Low, Medium, High Precision inside the Channel configuration screen.

Did changing that setting affect position reporting in a positive way for you?

BobsBlueNorth, 2024-08-01:
It did. It seems to be functioning as intended.

closed 2024-08-01 by jp-bennett