meshtastic/firmware #4367: [Bug]: Seeed Tracker T1000-E tracker doesn't detect when USB power is applied

geeksville, 2024-08-01:
### Category

Other

### Hardware

Other

### Firmware Version

master (private build)

### Description

Some of our behaviors change when the device realizes it is connected to USB power.  I just connected my T1000-E to my desktop and happened to notice this (low priorty/minor) problem:

```
DEBUG | 15:45:39 155470 [Power] Battery: usbPower=0, isCharging=0, batMv=4114, batPct=94
```

usbPower should be 1 in this message (because I was connected to USB power)
I suspect we need to add some variant code so that this detection happens (but I haven't yet looked at the schematic).  If someone else wants to investigate/fix this then great, otherwise I'll give it a shot sometime in the next few days

(not a big problem by cc @Dylanliacc )

### Relevant log output

```Shell
DEBUG | 15:45:39 155470 [Power] Battery: usbPower=0, isCharging=0, batMv=4114, batPct=94
```

thebentern, 2024-08-01:
Nordic has some API methods to determine this that might work. The RAK 4631 uses this macro to enable them. It should work for the tracker card as well I would imagine
`#define NRF_APM
`

geeksville, 2024-08-02:
Thanks @thebentern.  I tried turning this on for my Seeed Tracker T1000-E tracker and alas it didn't work (still said no USB power).  So I tried firing it up on my rak4631 and same behavior/problem.  So I think something might be slightly busted in general.  I'm debugging now.

geeksville, 2024-08-02:
btw - I turned on NRF_APM and that didn't help.  Which is odd because the code looks like it should.  I then went back and confirmed: yep still works great on a RAK4631.  I'm going to test on a Wio Tracker 1110 board now (because easier to debug than the 1000-E)

PR #4376 (2024-08-03): fix #4367 make USB power detection work correctly on seeed trackers
for wio tracker 1110 and 1000-E and possibly other nrf52 boards. The problem was that nrf52 power stuff wasn't generating regular powerstatus notifications (because that code was guarded by a batteryLevel check which was null for those boards).  So I've cleaned up the battery status stuff a bit and we now have fewer special cases.
Tested on a 1000-E, tracker 1110 and a rak4631 board.

(fix #4367)
commit dd552a99 (2024-08-03): fix #4367 make USB power detection work correctly on seeed trackers (#4376)

for wio tracker 1110 and 1000-E and possibly other nrf52 boards.
The problem was that nrf52 power stuff wasn't generating regular
powerstatus notifications (because that code was guarded by a batteryLevel
check which was null for those boards).  So I've cleaned up the battery status stuff
a bit and we now have fewer special cases.
Tested on a 1000-E, tracker 1110 and a rak4631 board.
closed 2024-08-03 by thebentern