meshtastic/firmware #4368: [Bug]: When toPhone queue is full we are dropping the newest packet - we should instead drop the oldests

geeksville, 2024-08-01:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

2.3.15.45bb9531

### Description

Heh.  This has been this way for a long time (I think forever) but I just noticed this while using a board that has been sitting on my desk (in a busy area with like 100 visible nodes).

```
DEBUG | 17:29:58 161730 [Router] Forwarding to phone (id=0x1af87b26 fr=0xad to=0xff, WantAck=0, HopLim=2 Ch=0x0 Portnum=3 rxtime=1722533398 rxSNR=46.25 rxRSSI=-93 hopStart=3)
WARN  | 17:29:58 161730 [Router] ToPhone queue is full, dropping packet.
```

When we receive packets from the mesh, we queue them for delivering to the local PC/phone.  However if the local ToPhone queue is full we are dropping the NEW packet we just received.  Is that correct?  I don't think so - wouldn't it make more sense to drop the OLDEST packet already in the queue?  Because if a user connects to their device they probably want the most recent 30 messages (or whatever) to come to their phone - not the oldest ones (which could be days old).

Thoughts?

geeksville, 2024-08-01:
Oh nevermind - already fixed in master ( the load on this T1000-E tracker was old)

GUVWAF, 2024-08-02:
What you're seeing even holds for current master, although the behavior depends on the packet type:
https://github.com/meshtastic/firmware/blob/4b4c1669a964c7e3f543952a2633db09afbbf2a1/src/mesh/MeshService.cpp#L311-L319

When the queue is full, it gets overwritten by text (and RangeTest) packets only. This was upon request of users that wanted higher priority for those type of packets.

Ideally we would replace the oldest non-text message, but I haven't found a way to iterate and replace packets in this special FreeRTOS `queue`.

geeksville, 2024-08-03:
oh yes - I had forgotten this thanks for the message @GUVWAF

closed 2024-08-03 by geeksville