meshtastic/firmware #4371: [Bug]: CardKB canned message module keyboard fails

sbusby9, 2024-08-02:
### Category

Hardware Compatibility

### Hardware

Heltec V3

### Firmware Version

2.4.0.46d7b82

### Description

CardKB canned module keyboard not functional under 2.4.0 stable release. Revert to 2.3.13 and it works again perfectly. CardKB has some problems but not complete failure under 2.3.15: long keypress delays, left arrow key function gone. Sometimes unresponsive.

HarukiToreda, 2024-08-06:
I am using a heltec V3 with the cardkb on version 2.4.0 just fine, works fine on 2.4.1, try doing a full wipe and reinstall the firmware or upgrade to 2.4.1

sbusby9, 2024-08-06:
Thanks for your response. My friend and I both have this problem, and we
both have telemetry boards BMP280 connected. My friend believes the
telemetry boards are what causes the CardKB fail in 2.4.0.
I2C address conflict?
Scott

HarukiToreda, 2024-08-06:
that is a factor not mentioned. There is a problem with the Enviromental screen, it is know to dissable the left and right buttons on the cardkb when enabled. If you dissable it, it will work just fine. 

see issue #3684 
https://github.com/meshtastic/firmware/issues/3684

closed 2024-08-06 by thebentern

VigibotDev, 2024-08-07:
2.4.0 stable on Station G2, canned message ON and cardkb connected, it write good, say message send, and no message on the mesh (broadcast@LongMod)

HarukiToreda, 2024-08-07:
> 2.4.0 stable on Station G2, canned message ON and cardkb connected, it write good, say message send, and no message on the mesh (broadcast@LongMod)

That is not the issue, cardkb works fine, the problem happens when you enable the environmental screen for a temperature sensor, it dissables the left and right button.