meshtastic/firmware #4378: [Bug]: nrf52 boards are not entering deep sleep when battery is critically low

geeksville, 2024-08-03:
### Category

Hardware Compatibility

### Hardware

Not Applicable

### Firmware Version

master (private build)

### Description

I just happened upon this code in Power.cpp:

```
#ifdef ARCH_NRF52
                // We can't trigger deep sleep on NRF52, it's freezing the board
                LOG_DEBUG("Low voltage detected, but not triggering deep sleep\n");
#else
                LOG_INFO("Low voltage detected, triggering deep sleep\n");
                powerFSM.trigger(EVENT_LOW_BATTERY);
#endif
```

Freezing boards is bad, but IMO it is **super** important to enter deep sleep anytime we are able to detect that the battery is critically low.  Because to not do this will kill lipos.  

I'm happy to look into this as I continue my adventure of improving nrf52 power consumtion.  But opening an issue in case any of ya'll have comments on this theory?

todd-herbert, 2024-08-03:
I built one node with a small solar enclosure from RAK. [(Unify Enclosure Solar IP65 100x75x38)](https://store.rakwireless.com/products/unify-enclosure-ip65-100x75x38-solar) 

* RAK4631
* RAK19007
* 1000mAh LiPo cell (to fit claimed 55x34x6mm space below mounting plate)

I was concerned that this battery could run flat. I made a [quick-and-dirty modification](https://github.com/meshtastic/firmware/compare/master...todd-herbert:meshtastic-firmware:nasty-solar-hack) to the firmware to add some level of software protection.

In essence: 
* Early in boot, check voltage
  - If below 3.7V, enter *pseudo deep-sleep*: `delay()` 30 minutes, then reboot
  - If above 3.7V, continue boot
* If `powerStatus` reports below 3.45V, reboot the node. This triggers the low voltage check.

This current draw in this *pseudo deep-sleep* state is very good; a few µA from memory? *However*, below a certain voltage, I believe the buck converter on the RAK19007 baseboard switches to a linear mode. Between a certain voltage range (3.5V down to 3.3V?) the current is higher; almost 1mA at worst.

Even in this 1mA region, the current is low enough to protect the battery from many days of poor weather. Despite the small battery, I haven't actually had this happen yet since deployment, although it worked very well in testing.

The *pseudo deep-sleep* state is the same one used with NRF52 tracker and sensor roles, however the current draw in those cases (after normal use, instead of early in boot) is higher (~1mA, even with good voltage). I suspect some piece of hardware isn't shutting down perfectly, but I couldn't identify exactly what.

___

This doesn't entirely fix the situation, but I found it to help in my specific use case. I'm not sure whether something like this would be suitable for general use though.

geeksville, 2024-08-03:
see related comments here: https://github.com/meshtastic/firmware/pull/2040

geeksville, 2024-08-04:
Progress notes (mostly for future searchers/reference on state of power as of today):

* I've found why waking system-off wasn't working - because it could only work for 'wake due to USB connect', I'm working on expanding that to also use wake based on ADC comparision of the battery voltage.  Will include in the fix for this bug.  For boards that can't do either of those things I'll have system-off mode failback to just super long sleeps with the existing old behavior.

Test configs:

RAK4630 with OLED display and GPS modules:

* Power consumption in system off (with the fixes I'll be submitting) is **excellent** when the battery voltage is at 3.7V.  It is drawing only 5uA which is almost the optimal possible per datasheet.

[image]

markbirss, 2024-08-05:
@geeksville it was someone else but i cannot find mention, on discord group ?

geeksville, 2024-08-13:
added this note from a discord discussion:

(better to shut down while we still have battery than to start browning out when we can't control the shutdown process.  And all bets get wonky in those super low v cases, possibly solar panel charges battery up some and the old write to meshtastic.txt (which I just removed in #4395) was highly likely to be writing when the board is coming-up then dying repeatedly when battery is below v3.3)

geeksville, 2024-08-20:
note to self: turn EVENT_LOW_BATTERY back on!

Joseph-DiGiovanni, 2025-07-02:
I'm confused. This thread seems to imply we should deep sleep sooner than normal for nrf52 platforms to minimize current draw, but the code was then updated specifically to not deep sleep at all on the nrf52. Could you please clarify the status of this issue?

Joseph-DiGiovanni, 2025-07-02:
Also is this possibly fixed in newer hardware/firmware? I had my RAK19003 run down to 2.75v (according to the Meshtastic app) and it seemed incredibly stable all the way up until somewhere past 2.65v when it eventually stopped responding.

Joseph-DiGiovanni, 2025-07-03:
> they could have instead passed the battery v straight to the cpu VDDH input (bypassing that regulator)

I'm honestly a little over my head here, so forgive my ignorance if I'm mistaken, but aren't the _current_ schematics for the RAK4630 and the RAK4631 module doing exactly this?

[image]

[image]

phaseloop, 2025-12-03:
You may be interested in this: https://github.com/meshtastic/firmware/pull/8793

I have a custom NiceNano/Faketec board and it works fine down to 2.5V but I tracked down flash corruption happening which is essentially ourselves shooting in the foot.

danielcharrua, 2026-09-09:
I think I may have found a related issue with the current implementation, specifically on `nrf52_promicro_diy_tcxo`.

I'm testing firmware 2.7.26.54e0d8d on an nRF52840 ProMicro-compatible board. Battery voltage is measured through a 1M/1M divider, and `adc_multiplier_override` is calibrated against a multimeter, so the reported battery voltage is accurate.

With the current firmware, when the battery reaches the low-voltage condition, the nRF52 eventually enters deep sleep / SYSTEM OFF as intended. The problem is that the node does not automatically recover when battery voltage rises again.
This is particularly problematic for unattended solar nodes.

Our board has an external undervoltage supervisor (MASN V2) which cuts power at ~2.9 V and automatically restores it when the battery recovers.

We performed a useful control test:

- Normal/calibrated ADC → Meshtastic detects low battery → nRF52 shuts down → node does not recover when battery voltage rises.
- ADC temporarily adjusted so Meshtastic never sees the low-battery condition → external supervisor reaches its ~2.9 V cutoff → physically removes power → restores power after battery recovery → nRF52 boots normally.

So the external power supervisor and hardware recovery work correctly. The unrecoverable state only occurs when Meshtastic initiates the low-battery shutdown first.

Looking at the current nRF52 code, there appears to already be support for using LPCOMP as a battery recovery wake source when `BATTERY_LPCOMP_INPUT` is defined. However, as far as I can see, `nrf52_promicro_diy_tcxo` does not currently define a battery LPCOMP wake source.

Since the nRF52840 hardware supports LPCOMP and the ProMicro DIY target already measures the battery through P0.31/AIN7, could LPCOMP wake support be added to this variant?

More generally, perhaps an nRF52 target that can automatically enter SYSTEM OFF due to low battery should also have a defined way to automatically wake when the battery recovers.

I'm happy to perform additional tests on the ProMicro hardware if useful, including testing older firmware versions or different LPCOMP recovery thresholds.