meshtastic/firmware #4381: SCD-30 CO2 sensor

pete4radio, 2024-08-03:
### Platform

NRF52, ESP32, RP2040

### Description

CO2 sensors are used indoors to know when ventilation is adequate, and used outdoors in fire monitoring.  The hardware and libraries are readily available thanks to Adafruit.

pete4radio, 2024-08-04:
in telemetry.proto can I use
  /*
   * SCD30 Scale Chip or compatible
   */
  SCD30 = 26;

fifieldt, 2024-11-07:
Sorry this was automatically closed.

This will happen, but is currently waiting on:

https://github.com/meshtastic/firmware/pull/4601

gatlinnewhouse, 2025-03-05:
I have a version working with Environmental and AirQuality by sharing access to the Sensor object [here](https://github.com/coffee-and-telesense/meshtastic-firmware/tree/scd30-stable-airquality). It's based on an older firmware version, and I think we also have a branch for the RAK but I cannot recall if there were any changes outside of `variants/*` for the RAK4631.

closed 2026-04-05 by oscgonfer