meshtastic/firmware #4384: [Bug]: "Wake On Tap Or Motion" setting not being honored

blarson, 2024-08-04:
### Category

Other

### Hardware

Rak4631

### Firmware Version

2.4.0

### Description

The device does not appear to honor the Display "Wake On Tap Or Motion" configuration setting when running in Client role (other roles not tested).

I've tried setting and clearing this flag via the Apple iPhone app and via the web client, and neither appears to work. The device continues to wake the screen on motion, regardless of whether the option is set or cleared.

It also does not appear to reboot after adjusting "Wake On Tap Or Motion" alone. It does, however, reboot when applying other Display settings.

garthvh, 2024-08-04:
Have you installed an accelerometer?

blarson, 2024-08-04:
Yes, the ST LIS3DH accelerometer on the RAK19026 base board.

It always wakes on motion, even with the setting disabled.

Ruledo, 2024-08-26:
I just realized this same bug with v2.5 and RAK1904 installed.
The setting in the client is not honored by the device.

Ruledo, 2024-08-26:
I tested the web client as well, same result.
The toggle appears to be set off but it still wakes on motion and when rebooted the setting toggle is enabled again.

closed 2024-09-06 by thebentern