meshtastic/firmware #4403: [Feature Request]: Add SOS Alert when user button is pressing 1 time with GPS on Android IoS and Web client

Niranjoyyengkhom, 2024-08-06:
### Platform

NRF52

### Description

Hi I want an Feature to send an SOS Alert to Mesh network Using User Button this will help us in use cases like Coal mining, Personal Tracking and other highly Risky areas

thebentern, 2024-08-06:
This feature has come up before an unfortunately it's problematic from a legal perspective for the project
#3561 

My guidance would be to look at the detectionsensor module in the docs. Someone recently made "man down" switch using this.

closed 2024-08-06 by thebentern

Niranjoyyengkhom, 2024-08-06:
> This feature has come up before an unfortunately it's problematic from a legal perspective for the project #3561
> 
> My guidance would be to look at the detectionsensor module in the docs. Someone recently made "man down" switch using this.

can you give me his contact info