meshtastic/firmware #4407: [Feature Request]: Five-way button based on ADC

gjelsoe, 2024-08-06:
### Platform

ESP32

### Description

Devices from Radio Master aka Bandit lineup has a five-way button connected to a ADC input
adding it would increase the number of buttons and open up for Canned Messages on those devices.

For the Bandit ADC values are [3227, 0, 1961, 2668, 1290, 4095] and corresponds to [UP, DOWN, LEFT, RIGHT, ENTER, IDLE]

Could also be used on other devices which have free ADC pins.

garthvh, 2024-08-06:
Should map fine to canned messages

gjelsoe, 2024-08-06:
**WIP.**

It also needs to be working with both Canned and "Main Menu" that is normally is defined with BUTTON_PIN.
Got some code that needs some testing.

gjelsoe, 2024-08-07:
This is how Radio Masters Bandits are using Five-way buttons.

[image]

gjelsoe, 2024-08-07:
@todd-herbert Has/is working on some code for it and it looks very promising..

todd-herbert, 2024-08-07:
Don't have the hardware, but had a go on behalf. Working from the [ExpressLRS source code](https://github.com/ExpressLRS/ExpressLRS/tree/d9f56f8bd6f9f7144d5f01caaca766383e1e0950/src/lib/SCREEN/FiveWayButton).

https://github.com/todd-herbert/meshtastic-firmware/tree/radiomaster-joystick
According to @gjelsoe, what's there is working for now. If anyone actually has the hardware themselves and wants to take over, please do!

Not sure what to do about "adhoc ping" / "toggle GPS". Is it worth implementing double / triple press for consistency with user button, or just do something else, like map to long-press up/down?

PR #4476 (2024-08-17): Radio Master Joystick
Closes #4407. Don't have the hardware myself, but well tested by @gjelsoe 

Creates a canned message input for the in-built joystick on Radio Master Bandit Nano.
Ports the `FiveWayButton` class from the radio's official [ExpressLRS firmware](https://github.com/ExpressLRS/ExpressLRS/tree/d9f56f8bd6f9f7144d5f01caaca766383e1e0950/src/lib/SCREEN/FiveWayButton)
Fully guarded to build solely for Radio Master Bandit Nano hardware (I hope)

No opportunity to use interrupts(?)
Supposedly, interrupts on ES32 GPIO39 cause issues with WiFi / BLE.
[image]
closed 2024-08-17 by thebentern