meshtastic/firmware #4410: [BUG] Infinite reboot loop after configuring as REPEATER

vk2diy, 2024-08-07:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.4.0.46d7b82

### Description

1. Get new device.
2. Update to latest stable firmware with web flasher.
3. Install app on phone.
4. Can't see anyone, so enable up/down and MQTT
5. Successful interaction
6. Try turning off phone proxying, up/down and MQTT and ignoring MQTT.
7. Can't see anyone.
8. Try changing system over to either REPEATER mode
9. System gets locked in infinite reboot/crash loop as follows.

```
DEBUG | ??:??:?? 36 [Router] Forwarding to phone (id=0x68ab6268 fr=0x18 to=0x18, WantAck=0, HopLim=3 Ch=0x0 Portnum=6 requestId=91fb59fe priority=70)
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d36b9  PS      : 0x00060230  A0      : 0x80119ff2  A1      : 0x3ffd85f0  
A2      : 0x3ffe3164  A3      : 0x3ffc8d4c  A4      : 0x3ffc8468  A5      : 0x000009bc  
A6      : 0x000009bc  A7      : 0x00000000  A8      : 0x00000100  A9      : 0x3ffd85c0  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x00000004  A13     : 0x3ffe329c  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000161  LBEG    : 0x40093e14  LEND    : 0x40093e2a  LCOUNT  : 0x00000000  

Backtrace: 0x400d36b6:0x3ffd85f0 0x40119fef:0x3ffd8610 0x4011778d:0x3ffd8630 0x4012ae65:0x3ffd8690 0x4012b0fd:0x3ffd86e0 0x4013b4f6:0x3ffd8700 0x401324f9:0x3ffd8720 0x400e2df5:0x3ffd8760

ELF file SHA256: 5daac71e23ca1618

E (40526) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
```
10. Re-flash the thing
11. Infinite reboot continues.

### Relevant log output

```Shell
/\ E S H T /\ S T / C

INFO  | ??:??:?? 0 Booted, wake cause 0 (boot count 1), reset_reason=reset
DEBUG | ??:??:?? 0 Filesystem files (507904/1048576 Bytes):
DEBUG | ??:??:?? 0  /prefs/channels.proto (57 Bytes)
DEBUG | ??:??:?? 0  /prefs/config.proto (99 Bytes)
DEBUG | ??:??:?? 0  /prefs/db.proto (428 Bytes)
DEBUG | ??:??:?? 0  /prefs/module.proto (116 Bytes)
DEBUG | ??:??:?? 0  /static/.gitkeep (0 Bytes)
DEBUG | ??:??:?? 0  /static/Logo_Black.svg.gz (602 Bytes)
DEBUG | ??:??:?? 0  /static/Logo_White.svg.gz (610 Bytes)
DEBUG | ??:??:?? 0  /static/apple-touch-icon.png.gz (3164 Bytes)
DEBUG | ??:??:?? 0  /static/favicon.ico.gz (2270 Bytes)
DEBUG | ??:??:?? 0  /static/icon.svg.gz (852 Bytes)
DEBUG | ??:??:?? 0  /static/index-CGqDWSD_.js.gz (348 Bytes)
DEBUG | ??:??:?? 0  /static/index-CcIXWgqL.js.gz (213944 Bytes)
DEBUG | ??:??:?? 0  /static/index-DfHMdodT.css.gz (16434 Bytes)
DEBUG | ??:??:?? 0  /static/index.html.gz (549 Bytes)
DEBUG | ??:??:?? 0  /static/maplibre-gl-DofehgOG.js.gz (210431 Bytes)
DEBUG | ??:??:?? 0  /static/robots.txt.gz (42 Bytes)
DEBUG | ??:??:?? 0  /static/site.webmanifest.gz (197 Bytes)
[   967][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
[   982][I][XPowersCommon.tpp:248] begin(): SDA:21 SCL:22
[   987][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
INFO  | ??:??:?? 0 AXP2101 PMU init succeeded, using AXP2101 PMU
[  1010][D][XPowersAXP2101.tpp:2791] setInterruptImpl(): DISABLE - HEX:0xffffffff BIN:
DEBUG | ??:??:?? 1 =======================================================================
DEBUG | ??:??:?? 1 DC1  : +   Voltage:3300 mV 
DEBUG | ??:??:?? 1 DC2  : -   Voltage:500 mV 
DEBUG | ??:??:?? 1 DC3  : -   Voltage:500 mV 
DEBUG | ??:??:?? 1 DC4  : -   Voltage:1800 mV 
DEBUG | ??:??:?? 1 ALDO1: -   Voltage:1800 mV 
DEBUG | ??:??:?? 1 ALDO2: +   Voltage:3300 mV 
DEBUG | ??:??:?? 1 ALDO3: +   Voltage:3300 mV 
DEBUG | ??:??:?? 1 ALDO4: -   Voltage:3300 mV 
DEBUG | ??:??:?? 1 BLDO1: -   Voltage:1800 mV 
DEBUG | ??:??:?? 1 BLDO2: -   Voltage:3300 mV 
DEBUG | ??:??:?? 1 =======================================================================
DEBUG | ??:??:?? 1 Battery: usbPower=1, isCharging=0, batMv=0, batPct=0
INFO  | ??:??:?? 1 Scanning for i2c devices...
[  1110][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
DEBUG | ??:??:?? 1 Scanning for I2C devices on port 1
DEBUG | ??:??:?? 1 I2C device found at address 0x34
INFO  | ??:??:?? 1 axp192/axp2101 PMU found
DEBUG | ??:??:?? 1 I2C device found at address 0x3c
INFO  | ??:??:?? 1 ssd1306 display found
INFO  | ??:??:?? 1 ssd1306 display found
DEBUG | ??:??:?? 1 0x4 subtype probed in 2 tries 
INFO  | ??:??:?? 1 2 I2C devices found
DEBUG | ??:??:?? 1 acc_info = 0
INFO  | ??:??:?? 1 S:B:4,2.4.0.46d7b82
DEBUG | ??:??:?? 1 Setting random seed 2890875052
DEBUG | ??:??:?? 1 Total heap: 189624
DEBUG | ??:??:?? 1 Free heap: 155148
DEBUG | ??:??:?? 1 Total PSRAM: 4192107
DEBUG | ??:??:?? 1 Free PSRAM: 4191863
DEBUG | ??:??:?? 1 NVS: UsedEntries 90, FreeEntries 540, AllEntries 630, NameSpaces 3
DEBUG | ??:??:?? 1 Setup Preferences in Flash Storage
DEBUG | ??:??:?? 1 Number of Device Reboots: 68
DEBUG | ??:??:?? 1 OTA firmware version 0.2.1.ceca52c
INFO  | ??:??:?? 1 Initializing NodeDB
INFO  | ??:??:?? 1 Loading /prefs/db.proto
INFO  | ??:??:?? 1 Loaded /prefs/db.proto successfully
INFO  | ??:??:?? 1 Loaded saved devicestate version 23, with nodecount: 1
INFO  | ??:??:?? 1 Loading /prefs/config.proto
INFO  | ??:??:?? 1 Loaded /prefs/config.proto successfully
INFO  | ??:??:?? 1 Loaded saved config version 23
INFO  | ??:??:?? 1 Loading /prefs/module.proto
INFO  | ??:??:?? 1 Loaded /prefs/module.proto successfully
INFO  | ??:??:?? 1 Loaded saved moduleConfig version 23
INFO  | ??:??:?? 1 Loading /prefs/channels.proto
INFO  | ??:??:?? 1 Loaded /prefs/channels.proto successfully
INFO  | ??:??:?? 1 Loaded saved channelFile version 23
[  1514][E][vfs_api.cpp:105] open(): /littlefs/oem/oem.proto does not exist, no permits for creation
INFO  | ??:??:?? 1 File /oem/oem.proto not found
DEBUG | ??:??:?? 1 cleanupMeshDB purged 0 entries
DEBUG | ??:??:?? 1 Using nodenum 0x1f9ffa18 
DEBUG | ??:??:?? 1 Number of Device Reboots: 68
DEBUG | ??:??:?? 1 Expanding short PSK #1
INFO  | ??:??:?? 1 Wanted region 6, using ANZ
INFO  | ??:??:?? 1 Saving /prefs/db.proto
DEBUG | ??:??:?? 1 Using GPIO38 for button
DEBUG | ??:??:?? 1 SPI.begin(SCK=5, MISO=19, MOSI=27, NSS=18)
DEBUG | ??:??:?? 1 Set Timezone to GMT0
DEBUG | ??:??:?? 1 Read RTC time as 327
INFO  | ??:??:?? 1 Turning on screen
DEBUG | ??:??:?? 1 haveGlyphs=1
DEBUG | ??:??:?? 1 RF95Interface(cs=18, irq=26, rst=23, busy=33)
INFO  | ??:??:?? 1 Starting meshradio init...
DEBUG | ??:??:?? 1 (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=0, time 231 ms
DEBUG | ??:??:?? 1 (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=253, time 2115 ms
INFO  | ??:??:?? 1 Radio freq=919.875, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 1 Set radio: region=ANZ, name=LongFast, config=0, ch=19, power=30
INFO  | ??:??:?? 1 Radio myRegion->freqStart -> myRegion->freqEnd: 915.000000 -> 928.000000 (13.000000 MHz)
INFO  | ??:??:?? 1 Radio myRegion->numChannels: 52 x 250.000kHz
INFO  | ??:??:?? 1 Radio channel_num: 20
INFO  | ??:??:?? 1 Radio frequency: 919.875000
INFO  | ??:??:?? 1 Slot time: 42 msec
INFO  | ??:??:?? 1 Set radio: final power level=20
DEBUG | ??:??:?? 1 Current limit set to 100.000000
DEBUG | ??:??:?? 1 Current limit set result 0
INFO  | ??:??:?? 1 RF95 init result 0
INFO  | ??:??:?? 1 Frequency set to 919.875000
INFO  | ??:??:?? 1 Bandwidth set to 250.000000
INFO  | ??:??:?? 1 Power output set to 20
INFO  | ??:??:?? 1 RF95 Radio init succeeded, using RF95 radio
INFO  | ??:??:?? 1 Not using WIFI
DEBUG | ??:??:?? 1 (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=237, time 2000 ms
DEBUG | ??:??:?? 1 LoRA bitrate = 118.500000 bytes / sec
�
 INFO  | ??:??:?? 1 PowerFSM init, USB power=1
DEBUG | ??:??:?? 1 Enter state: BOOT
[  1989][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 320 / 4 = 80 Mhz, APB: 80000000 Hz
DEBUG | ??:??:?? 2 [Power] Battery: usbPower=1, isCharging=0, batMv=0, batPct=0
DEBUG | ??:??:?? 2 [Screen] Screen: Started...
DEBUG | ??:??:?? 2 [Screen] haveGlyphs=1
DEBUG | ??:??:?? 3 [Screen] haveGlyphs=1
DEBUG | ??:??:?? 4 [Screen] haveGlyphs=1
INFO  | ??:??:?? 5 [PowerFSM] Initialise the NimBLE bluetooth module
DEBUG | ??:??:?? 5 [Screen] haveGlyphs=1
INFO  | ??:??:?? 6 [PowerFSM] Turning off screen
```

vk2diy, 2024-08-07:
On further investigation it seems it's every time I try to reconfigure the device using the app that it crashes.

vk2diy, 2024-08-07:
On further investigation, no it's just crashing. Maybe I'll see if it's the app. If I turn off bluetooth it might go stable. Hang on ...

vk2diy, 2024-08-07:
Nope just seems to be crashing. I think the device is just stuck in an infinite reboot loop. It's confusing because the screen goes off and on but there's never any data displayed just the logo or black, and a momentary screen update looks partially corrupted before reboot. Flashing behind the screen continues as normal. 

After awhile there is a long red flash followed by a period of no LED, followed by a reboot.

If I press the left button the screen doesn't wake and the flashing continues. If I press the middle button the screen wakes for awhile then goes back to sleep. If I press the right button it resets.

I can't use the android interface to do anything because it's not up long enough.

I've already tried resetting with the web flasher.

Please help me unbrick, otherwise I will have to send the hardware back.

vk2diy, 2024-08-14:
No help after 1 week, I am returning the hardware. Your bug remains.

GUVWAF, 2024-08-15:
You can probably still reflash it with the CLI: https://meshtastic.org/docs/getting-started/flashing-firmware/esp32/cli-script/
You'll likely need to try to connect to it multiple times, because it only has a small amount of time to react before it reboots.

Not sure why it's crashing, this is the decoded stack trace: 

> 0:
  0x400d36b6: _ZN11MeshService11sendToPhoneEP22_meshtastic_MeshPacket$constprop$918 at ??:?
0:
  0x40119fef: RoutingModule::handleReceivedProtobuf(_meshtastic_MeshPacket const&, _meshtastic_Routing*) at :?
0:
  0x4011778d: ProtobufModule<_meshtastic_Routing>::handleReceived(_meshtastic_MeshPacket const&) at :?
0:
  0x4012ae65: Router::handleReceived(_meshtastic_MeshPacket*, RxSource) at :?
0:
  0x4012b0fd: Router::runOnce() at ??:?
0:
  0x4013b4f6: concurrency::OSThread::run() at ??:?
0:
  0x401324f9: loop() at ??:?
0:
  0x400e2df5: loopTask(void*) at :?

vk2diy, 2024-08-15:
Sorry already returned the hardware.

fifieldt, 2024-09-16:
This looks like a duplicate of https://github.com/meshtastic/firmware/issues/3066