meshtastic/firmware #4414: Enable regional satellite navigation systems based on region

fifieldt, 2024-08-08:
### Platform

other

### Description

At the moment we don't have a lot of structure about which satellite constellations we enable. Typically, we'll default to enabling as many as possible +/- any power considerations.

For  Beidou, Galileo, GPS, and GLONASS, this is fine - these are global systems. However, there are two regional systems: NAVIC, and QZSS, that are supported by an increasing number of chips. These systems are only available in certain areas, but in the areas they operate can provide excellent results.

We should enable NAVIC and QZSS where it makes sense, and disable them if we're clearly outside the coverage area. It's suggested in a gps.cpp todo comment that we could do this based on the selected LoRa region.

Primary coverage areas and suggested settings:
NAVIC: (5, 55) to (50, 110)
[image]
IN
TH
MY
SG

QZSS: (95, -40) to (160, 50)
[image]
ANZ
JP
KR
TH
TW
MY
SG

fifieldt, 2024-08-08:
Related to meshtastic/meshtastic#1351

GPSFan, 2024-08-09:
This is a hard call as both those systems exclude a lot of Europe, the UK and the US. The systems are not complete (yet) and may have issues with power management on certain receivers. With that said, both can contribute to better location accuracy, so they should be on the road map for support. Then of course you have folks wanting to degrade GPS accuracy by default. Maybe Meshtastic needs a high accuracy/low latency role as well as a low accuracy "location secure" role.

gergy008, 2024-08-25:
Not sure about all of them, but the GPS chip I'm using in my tracker doesn't allow for individual satellite owners to be selected, it's either receiving data from all that the chip supports, or none. Is this something that would be controllable by Meshtastic?

GPSFan, 2024-08-25:
And which GPS chip might that be?

fifieldt, 2024-09-11:
An alternative to trying to do this automatically is to surface the option to the user to select constellations. What do people think about that?

closed 2024-11-17 by github-actions[bot]