meshtastic/firmware #4429: [Bug]: No GPS on Heltec wireless tracker

brloomis, 2024-08-09:
### Category

Other

### Hardware

Heltec Wireless Tracker

### Firmware Version

2.4.2.5b45303

### Description

After #4319 my Heltec Wireless tracker v1.1 shows "No GPS", and the logs indicate that it could not probe the GPS unit.

Patching this line: https://github.com/meshtastic/firmware/pull/4319/files#diff-c9e9d981789c05f2633c8bc91527d5c33f714ea43d106684250598ec0d4487ccR1194
to look for "UM600" instead of "UC6580" works, and the GPS seems to work OK afterwards.

I'm not sure if this is an issue specific to my device (I tried two with the same issue).

### Relevant log output

```Shell
Originally I got these logs:

DEBUG | ??:??:?? 5 [GPS] Probing for GPS at 115200 
DEBUG | ??:??:?? 5 [GPS] Setting Baud to 115200
WARN  | ??:??:?? 10 [GPS] Failed to find UBlox & MTK GNSS Module using baudrate 115200

I added debug output and here is the full response from $PDTINFO:

$PDTINFO,FB2S UM600,G1B1L1E1,V2.0,R6.0.0.0Build1280,N/A,N/A*74
```

brloomis, 2024-08-10:
For reference, my test patch is here: https://github.com/brloomis/meshtastic-firmware/commit/1f7242eeab381117f59936257cdfa0e5d2bc60d6 but there may be a better way to do it.

fifieldt, 2024-08-10:
@brloomis I think that's worth sending in. Sorry for breaking your device!

fifieldt, 2024-08-12:
@brloomis do you have time to send in the pull request? If not, let us know :D I'm just worried others may need this fix quickly :)

fifieldt, 2024-08-12:
Thanks @brloomis , yes -I think there's a decent amount of optimisation work that could be done on that probe method :)

I was curious about that too and also could not find a chip with that model. My best guess is that maybe UM600 was a 'series' marker and UC6580 is somehow related to eg UM620?

madeofstown, 2024-08-14:
You possibly have a bad unit. Does it get really warm when in use?

closed 2024-09-15 by fifieldt