meshtastic/firmware #4434: Nara: Virtual Pet Game for Meshtastic

eljojo, 2024-08-10:
## What is it?
Nara (Not A Robotic Animal) is a virtual pet game, built as a module for Meshtastic. You battle other nodes and win points.

The idea is to have fun while also encouraging people to join and grow the Meshtastic network.

Watch video here: https://youtu.be/MWfdFIDvkeY

https://github.com/user-attachments/assets/6cd1a610-c8c0-4f2c-8517-1abb95137a72

## How to play?
All you have to do is enable the module and go outside!

- **Find nodes:** You'll start making points by simply finding new Meshtastic nodes.
  - Every 10 minutes, Nara picks up a new online node from the local meshtastic database and invites it to battle.
  - This happens only once per day, per node in the db.
- **Battle:** If you find another node playing Nara, the two will enter a battle.
  - Naras exchange random numbers, and an algorithm decides who wins.

<details><summary>

### How does the algorithm work?
</summary>
<p>
The algorithm ensures fair play by using cryptographic hashes to determine winners, regardless of hardware differences.

First, both nodes exchange randomly generated numbers and start the battle.

| Message          | From    | To      |
|------------------|---------|---------|
| Battle Invite (123)  | Aaron   | Britney |
| Battle Accept (456)  | Britney | Aaron   |

Once numbers have been exchanged, they are combined to do a hash computation. The formula is:

```
SHA256("%d/%d/%0x/%d", aaronMessage, britneyMessage, localNodeNum, counter)
```

Nodes will compute hashes, increasing the counter, until one of the following happens:
- A counter is found, for which the hash ends in **four** zeros
    - Input: `"123/456/4359109c/54633"`
    - Hash: `7724bec8bb34ae39a85af34402f4b4a08db7a8edd6d0230f166b014cc1950000`
- The counter reaches 100,000 iterations -- Nodes give up.
- The other node finds a counter better than our current best iteration (i.e., even if we find a matching hash, our counter will lose).

Once Naras find a matching counter, or give up searching, they'll message the other one letting them know of their counter. The counter is verified upon reception by the other node.

| Message            | From    | To      | Meaning                          |
|--------------------|---------|---------|----------------------------------|
| Battle Turn (432)  | Britney | Aaron   | (waiting for Aaron)              |
| Battle Turn (1234) | Aaron   | Britney | Britney wins: 432 < 1234         |

</p>
</details> 

## What are the next steps?
I'm sharing this project to see if there's potential in merging some form of this into the Meshtastic codebase.
- **Short-term:** Finalize protobufs, gather community feedback, and refine the core mechanics.
- **Long-term:** Expand the game features (e.g., collaborative modes, special node roles) and integrate graphics and sound.

## Project status?
Currently, there's a working proof of concept. I'm at the point of getting community feedback before continuing developing this idea. The game could go in multiple different directions. I don't have experience with C++, so my code may need re-writing.

You can find the code here: https://github.com/meshtastic/firmware/compare/master...eljojo:meshtastic-firmware:nara-module

**Working features:**
- Battle mechanics described above using Protobufs
- Module that shows Nara status on Screen, including battle + points
- "Hello Nara" message sent to mesh on boot, this helps discover other Nara and recover points from previous battles (allows for unpersisted distributed database)

### FAQ:
- **Will this congest the network?** The game will auto-disable on congested networks. The idea is to help density in sparse areas, not add noise in areas with good meshing. The game should auto-regulate, maybe decreasing hops or battling less often.
- **Will this turn into a hashing arms race?** Despite hashing being used, it's not a game of who computes the fastest. Hashes are only used to make it hard to predict winning moves. Imagine playing a game of rock-paper-scissors over instant-messaging! Devices of diverse capabilities have an equal footing, and often no one wins.
- **Will my battery die when it's enabled?** In my stress tests, it hasn't added noticeable battery drain to normal Heltec v3 devices. For wireless paper, it drains significantly faster due to e-ink refresh. More optimizations and real scientific testing can be done here.

### Other Ideas:
- There's no concept of rankings or leveling up yet, but it'd be great to develop such mechanics.
- **Website:** I'd like a website similar to the online maps that show statistics about each Nara.
- **Collaborative game mode:** Two (or more) Naras join forces in trying to find the best hash.
- **Special node mode:**
    - **Game node:** Meant for battery-powered remote nodes. This node doesn't play itself; instead, it broadcasts challenges, where one or more nodes can compete. Ideally SPI backed.
    - **Gateway node:** Allows two-way sync over RF or MQTT between nodes and some central game database.
- **DNA and mutation:** Have ways for Nara to grow/mutate/reproduce.
- **Graphics:** I'd really like the game to have some cool graphics to show what's going on, and the status of your pet.
- On devices with buzzers, it should make sound during battles.

### Requirements:
- CryptoEngine changes to add SHA256 hashing, platform dependent
- [Not yet implemented] Some storage space to store module settings/db of node battle status
    - If space is too limited, it may be an optional setting for nodes with an SPI attached
- Some RAM to store Nara DB of battle status
- New protobufs added
- would like to have: mobile and web versions, CLI support!

garthvh, 2024-08-10:
Very cool!

jp-bennett, 2024-08-11:
This sounds so good. Looking forward to giving it a closer look, and thinking about how to integrate!