meshtastic/firmware #4447: [Bug]: Fixes needed in 3.0 timeframe for NRF52 filesystem reliability

geeksville, 2024-08-12:
### Category

Hardware Compatibility

### Hardware

Not Applicable

### Firmware Version

2.4.1.3

### Description

(making this bug now for linking purposes.  Will populate details in a few minutes)

esev, 2025-01-18:
> So fundamentally: if we get reset while doing any portion of that page erase/write operation we will lose not just the 1 (ish) LFS block which was being synced, but up to 31ish random filesystem blocks who were unlucky to be nearby.

Adding to this thought. If any of the physical flash pages goes bad, we lose 4096 bytes of the file system and LittleFS is only able to recover and relocate 128 of those bytes to a new page.

fifieldt, 2025-10-08:
Filesystem reliability has improved substantially since this bug was lodged. Is this still needed?