meshtastic/firmware #4466: [Feature Request]: Please Enable More Frequent Device Metrics/Telemetry for Router Role Devices ONLY in Crowded Mesh

Deuteranomalous1, 2024-08-15:
### Platform

NRF52

### Description

The improvements to Metrics/Telemetry broadcasting for busy meshes are great at reducing congestion. 

However, it's also created a bit of an issue for busy mountain top Routers. Routers are by definition meant to be located high and clear of obstructions. This means they are very likely to see more than 40 nodes and also see the highest channel utilization in any given mesh. 

This has created somewhat of an issue after upgrading my main router to 2.4.x. I am now seeing metrics updated maybe once every few days due to the exceptional viewshed it possesses which overlooks the entirety of the south Salish Sea from Vancouver down to Puget Sound.  The channel util at the site was perpetually hovering around 50% based on my last on site visit. This has grown substantially since the unit was installed in March when it was reliably around 20%. Since then I have had a steady decline in frequency of updates corresponding with increase in channel utilization but the metrics were received more than often enough to be useful data. Upon upgrading to 2.4.0 a few weeks ago the updates have almost ceased entirely.

Routers mode devices are also remote and the metrics/telemetry are instrumental in preemptively spotting issues such as under charging and moisture ingress. Having multiple days pass between received telemetry information significantly hampers this ability. 

My ask is that the frequency scaling for Metrics/Telemetry on Router mode ONLY nodes be altered in a future release. Perhaps an interval of every 3 hours regardless of number of online nodes and channel utilization? In a low traffic mesh the current scaling could still be observed. 

I've attached screenshots from my device and another user to illustrate the infrequency of useful telemetry reception. 
[image]
[image]

Thank you!

b8b8, 2024-08-15:
My recommendation a few months ago on Discord was to have ROUTERS send telemetry (including an environmental packet) every 6 hours even when channel utilization is high.

GUVWAF, 2024-08-15:
Even at high channel utilization, you can still request the telemetry with the Python CLI command `--request-telemetry`. This is not (yet) in the phone apps, but technically it can.

Currently `ROUTER`s only send their telemetry every 12 hours because they are assumed to be a permanent install, either powered or with a good solar set-up. I'm not sure if making an exception for `ROUTER` would be a good idea, because it would be another incentive for people to use this role while they maybe shouldn't, especially if your mesh is already crowded.

thebentern, 2024-08-15:
Actually Routers _should_ already be exempt from this scaling logic, so this might indeed be a bug rather than feature request. I'll have to do some digging, as the code in Default.cpp straightforward, but perhaps it's not picking up the value somehow. 
[image]

GUVWAF, 2024-08-15:
Probably its then just the high channel utilization that's causing it to not send it out:
https://github.com/meshtastic/firmware/blob/efc27f205106f7caf7b1541b64770e98e8b40872/src/modules/Telemetry/DeviceTelemetry.cpp#L23

GUVWAF, 2024-08-15:
Yeah, I think that's fair. Then it will send it still up to 40% instead of 25% channel utilization. 

Just don't document it as a feature for the `ROUTER` role ;)

closed 2024-08-15 by thebentern

Deuteranomalous1, 2024-08-15:
> @Deuteranomalous1 could you post a config dump of one of these routers (without sensitive information)?

Sorry its a 5-6 hour round trip hike this time of year but I can get it for you next trip if you still need it but it seems like its well in hand :)

ashleycawley, 2025-01-04:
> Actually Routers _should_ already be exempt from this scaling logic, so this might indeed be a bug rather than feature request. I'll have to do some digging, as the code in Default.cpp [...]

What ever came of this? In our area we are seeing issues with Routers not reporting their Device Metrics / battery levels for numerous days (if your lucky).

Through some prior discussions on Discord I think I determined that the Router role should be broadcasting out this information once every 12 hours, however what I've read here makes me think this will only occur if its under a certain Channel Util % threshold. I've tried to find where that threshold is set in the code base but I'm not confident with reading this language.

Could someone confirm if the above assumption is correct and if so what the current % threshold is set to?

Also, if the Router role is set to broadcast **so infrequently (12hrs)** would it not make more sense to bypass Channel Util checks? One packet every 12 hours isn't going to exacerbate or cause congestion issues.

GUVWAF, 2025-01-04:
Routers are now considered "impolite", meaning they will send their DeviceTelemetry up until 40% channel utilization instead of 25%: 
https://github.com/meshtastic/firmware/blob/9afadde2f4ef8c59c80c7926d1c77153f6f9ca33/src/modules/Telemetry/DeviceTelemetry.cpp#L22

You can also request DeviceTelemetry using the Python CLI `--request-telemetry` command, which would still work even at high channel utilization.

ashleycawley, 2025-01-04:
Yes I am aware that there is a manual work-around method using the CLI as you covered that in this convo back August, thank you for mentioning that back then as it was useful to me in a pinch.

However, the vast majority of users will interact with their nodes and do quick health-checks via mobile Apps and that alternative method does not help those majority of users there.

> ... they will send their DeviceTelemetry up until 40% channel utilization instead of 25%:

Two points from this I guess:

1. Are you able to point me to the file / line number where that % is specified? **I'm convinced we're seeing it blocked at values <40%** (perhaps in the 25-40% bracket).
2. **What is the rationale behind preventing the broadcast of one very useful packet which is dispatched very infrequently (12hrs)?**

GUVWAF, 2025-01-04:
1. It's here: https://github.com/meshtastic/firmware/blob/2c654454cffabdd165311ac7bac3770ba975b53b/src/airtime.h#L70
2. If your mesh is really above 40% channel utilization, then it will already be very unreliable. The reason you might miss the packets now might be simply because of collisions due to the congestion. Also, the 12 hour interval is just the default, you can set another [interval](https://meshtastic.org/docs/configuration/module/telemetry/#device-metrics-update-interval) as long as it's not less than the minimum (30 min.).

ashleycawley, 2025-01-04:
1. Thank you!
2. As I've tried to emphasise I don't think our channel util is above 40%, I think its below that. Our small mesh in my vicinity is very reliable, we don't have congestion, we live a very rural area with no major cities nearby. We have a small budding mesh building in our area but 8-14hrs can easily go by with not a single user generated message on the primary/public channel, just to give you a sense for how quiet / busy it is. Device Metrics are reliably gathered from other _Client_ nodes in the local mesh.

> the 12 hour interval is just the default, you can set another [interval](https://meshtastic.org/docs/configuration/module/telemetry/#device-metrics-update-interval) as long as it's not less than the minimum (30 min.).

This is one of the first things we tried and had no joy with, it seemingly made no difference. Is that setting adhered to the same when a device is using the Router role?

If the remedy was that easy to change that setting then why was that not proposed to the original author on this issue? My prior testing suggested it was having no effect when changed, which is what led me to going down this path of finding other hidden attributes discussed in here (channel util %) which can get in the way and block this.

If the device metric broadcast for Routers was permitted once every 12 hours irrespective of channel util % then this would all be a mute point and people would be able to easily get the data they need. I'm struggling to see the rationale behind that limitation.

GUVWAF, 2025-01-04:
> Is that setting adhered to the same when a device is using the Router role?

I believe so, I can't fins in the code where it wouldn't. 

> If the remedy was that easy to change that setting then why was that not proposed to the original author on this issue?

The original issue was specifically about the high channel utilization. Back then it was still 25% and the interval is scaled with the number of online nodes. Now, for routers it's 40% and not scaled with the number of online nodes.

> If the device metric broadcast for Routers was permitted once every 12 hours irrespective of channel util % then this would all be a mute point

So why would relaxing this constraint help for you if your channel utilization is very low?

ashleycawley, 2025-01-04:
> So why would relaxing this constraint help for you if your channel utilization is very low?

Because the broadcast for the Router role has been demoted to be very infrequent (12hrs) and I think sometimes our channel utilization may range around 20-35%. And as I've been trying to state from testing I think lower than 40% thresholds have still been imposing a block potentially.

**I have LoS to a Router 1 mile away which I can communicate with flawlessly, remote admin etc. but I can go many days with no Device Metrics being retrieved. I can on-demand, manually use the CLI on my PC just to retrieve them reliably, so the link is good.**

My understanding of channel utilisation from a [official Meshtastic video](https://www.youtube.com/watch?v=aDud_5Bxtvc) led me to believe that non-Meshtastic traffic on the band (other LoRaWAN / Helium traffic) can also contribute to the channel utilisation figure, so I believe that some of what we're seeing locally is not Meshtastic traffic but other stuff on the air and due to Meshtastic's code logic choices its blocking informational broadcasts in a very quiet Meshtastic zone.

GUVWAF, 2025-01-04:
> I can go many days with no Device Metrics being retrieved.

Are you always connected with an app to a Meshtastic device? A node can only store around 30 packets of all types, and text messages get priority when the buffer is full. 

> My understanding of channel utilisation from a official Meshtastic video led me to believe that non-Meshtastic traffic on the band (other LoRaWAN / Helium traffic) can also contribute to the channel utilisation figure

This is highly unlikely if not impossible. They have to be using the exact same frequency and LoRa settings, as well as our custom sync word.

ashleycawley, 2025-01-04:
> Are you always connected with an app to a Meshtastic device?

Yes I have tried leaving it connected for days.

> This is highly unlikely if not impossible.

Does that mean there is a mistake in the first 35 seconds of this official Meshtastic video: [https://www.youtube.com/watch?v=aDud_5Bxtvc](https://www.youtube.com/watch?v=aDud_5Bxtvc) or perhaps I have misinterpreted the opening statements?

ashleycawley, 2025-01-04:
Just to reiterate - I can at any point change this node from Router to Client and start getting Device Metrics reliably every hour, as soon as its switched back to Router it stops, no updates every 12hrs, even if you're patient and wait 48-72hrs.

GUVWAF, 2025-01-04:
> Does that mean there is a mistake in the first 35 seconds of this official Meshtastic video

Yes, while it's theoretically possible, I don't think there is currently any other system that uses the exact same LoRa settings and sync word as we do. 

> I can at any point change this node from Router to Client and start getting Device Metrics reliably every hour, as soon as its switched back to Router it stops, no updates every 12hrs, even if you're patient and wait 48-72hrs.

Strange, I don't see how this can occur. Can you try setting it to 30 min.?

b8b8, 2025-01-04:
i have a router running firmware from the last month (cant remember which one exactly) and have it set to telemetry every 30min. In my busy mesh i normally end up getting telemetry every 1hr approx. Just set to to 30min (1800) and it will work as expected.

Deuteranomalous1, 2025-01-05:
> i have a router running firmware from the last month (cant remember which one exactly) and have it set to telemetry every 30min. In my busy mesh i normally end up getting telemetry every 1hr approx. Just set to to 30min (1800) and it will work as expected. The defaults have changed, but the minimum for us (you) that know what we're doing it still very good.

Agreed this is what I ended up doing as well. Surprised this is still open. 

FWIW requesting telemetry over CLI does not provide the whole picture. Just the basic telemetry built in to the node. At least not for me:

PS C:\Users\user> py -m  meshtastic --request-telemetry --dest '!6e2718a'
Connected to radio
Sending device_metrics telemetry request to !6e2718a on channelIndex:0 (this could take a while)
Telemetry received:
Battery level: 91.00%
Voltage: 4.07 V
Total channel utilization: 12.09%
Transmit air utilization: 2.93%
Uptime: 710430 s

This node is equipped with an INA219 for voltage/current and a RAK 1901 for Temp/Humidity. The telemetry for those items sends just fine with the regular interval updates but apparently can not be requested over CLI. 

Not complaining, just seems worth noting in case someone looks this up later.

GUVWAF, 2025-01-05:
> The telemetry for those items sends just fine with the regular interval updates but apparently can not be requested over CLI.

I believe you can if you update the CLI and use `--request-telemetry power` or `--request-telemetry environment`.

Deuteranomalous1, 2025-01-05:
> > The telemetry for those items sends just fine with the regular interval updates but apparently can not be requested over CLI.
> 
> I believe you can if you update the CLI and use `--request-telemetry power` or `--request-telemetry environment`.

Hot diggidy thank you sir! Worked like a charm.

ashleycawley, 2025-01-05:
OK, thanks for confirming that video had some false-statements.

> Strange, I don't see how this can occur. Can you try setting it to 30 min.?

Certainly, I'd be happy to try new suggestions. I have tried this in the past but happy to try again, just for clarity as I know Android / iOS lay things out differently, on my iOS App I'm going to:
Settings > Module Configuration > Telemetry (Sensors) > Update Interval > Device Metrics > Thirty Minutes
If that is the wrong place then let me know.

After puttning that setting in place via Remote Admin I confirmed it was successful by querying via the CLI on another device and got back:
```telemetry:
device_update_interval: 1800
power_update_interval: 1800
```
I have also cleared the log on my mobile app for the node so I can monitor what comes in over time.

My belief is that the isImpoliteRole logic block which only exists in DeviceTelemetry.cpp is getting triggered too easily (possibly below 40%) and this is causing issues, if someone more capable than myself who can interpret / understand the code is able to double-check the logic in that code block it would be greatly appreciated.

IIRC the video said the Channel Util was measured every minute, is that correct or false as well? If it is a % being measured over the period specified by the update interval, then a longer period update interval (ie. 12hrs) would gather a higher % over the time? I realise its probably not this last bit.

GUVWAF, 2025-01-05:
> My belief is that the isImpoliteRole logic block which only exists in DeviceTelemetry.cpp is getting triggered too easily (possibly below 40%) and this is causing issues, if someone more capable [...]

I double-checked the logic and I don't see anything wrong.

> IIRC the video said the Channel Util was measured every minute, is that correct or false as well?

This is correct.

ashleycawley, 2025-01-05:
Thank you for confirming those two things @GUVWAF 👍

After switching the Device Metrics to thirty minutes earlier this morning and verifying it via the CLI, so far today I've received just one metric broadcast at 12:00pm. This is to that LoS Router 1 mile away with no hops. 

Stats from that one Device Metric were:
Channel Util: 22.04%
Airtime: 1.09%

Manually running `meshtastic --request-telemetry --dest '!433ae044' --no-time` now at 17:30 just to get another perspective on channel util / airtime later in the day I get:
```
Telemetry received:
Battery level: 86.00%
Voltage: 4.03 V
Total channel utilization: 27.05%
Transmit air utilization: 6.20%
```

Are you able to confirm that the logic which changes Router's device metric broadcasts to be every 12 hours does not override and take precedence over the manual configuration of the setting shown in my screenshot above?

If you have any further suggestions on things to try then I'm happy help in testing what I can.

GUVWAF, 2025-01-05:
> Are you able to confirm that the logic which changes Router's device metric broadcasts to be every 12 hours does not override and take precedence over the manual configuration of the setting shown in my screenshot above?

In DeviceTelemetry it calls `getConfiguredOrDefaultMsScaled()` with the setting:
https://github.com/meshtastic/firmware/blob/02a5a91da0813858c5fd0c27282f3a01271968cd/src/modules/Telemetry/DeviceTelemetry.cpp#L25-L26
And this eventually uses the configured interval if you have set it:
https://github.com/meshtastic/firmware/blob/02a5a91da0813858c5fd0c27282f3a01271968cd/src/mesh/Default.cpp#L7-L8

So, yes, that seems good to me. Also b8b8 confirmed it's working as expected for his router.

> If you have any further suggestions on things to try then I'm happy help in testing what I can.

What's the firmware version of your router actually? If it's <2.4.3 the limit would still be at 25% channel utilization, which you seem to exceed sometimes.

ashleycawley, 2025-01-05:
> 1. It's here: https://github.com/meshtastic/firmware/blob/2c654454cffabdd165311ac7bac3770ba975b53b/src/airtime.h#L70

I've just noticed the line beneath that one you highlighted above is:
https://github.com/meshtastic/firmware/blob/2c654454cffabdd165311ac7bac3770ba975b53b/src/airtime.h#L71

Doesn't that variable sound more relevant to the whole polite / impolite logic block we have been discussing?
Also as I've mentioned, it feels like through testing Chan Util in between 25-40% has been getting blocked, so when I spotted this line it got me excited. Apologies if its a red herring.

ashleycawley, 2025-01-05:
> What's the firmware version of your router actually?

2.5.15

GUVWAF, 2025-01-05:
> Doesn't that variable sound more relevant to the whole polite / impolite logic block we have been discussing?

If it's a router `isTxAllowedChannelUtil()` is called with `false` and then it uses `max_channel_util_percent`:

https://github.com/meshtastic/firmware/blob/5196ee39cb8d8122f794fe0b100e4757887028ea/src/airtime.cpp#L123-L125

ashleycawley, 2025-01-06:
OK, so I'm around **28hrs** into my Router being set to a Device Metric Interval of every **thirty minutes** now and this is what I have so far:
[image]
Just one metric. When potentially there could have been over 50 which could have come through.

As mentioned if I switch it from router to client they will start coming through more regularly and more successfully. Given the results I'm seeing you can appreciate how I began to think:
a) The device metric interval setting was being ignored when the router role was used.
b) The impolite logic block may have been catching chan util <40% but I appreciate you've checked that.

I'm not the only one seeing this either, I have heard the same from another router operator around 10 miles from me.

Any thoughts on what else I can try or test @GUVWAF ? Thanks for your time and help.

garthvh, 2025-01-06:
Router and repeater send metrics every 12 hours.

ashleycawley, 2025-01-06:
> Router and repeater send metrics every 12 hours.

Yes, I learnt that some time ago, however @GUVWAF gave me the impression recently that the Device Metric Update Interval setting would still over-ride that and take precedence. Are you saying that's not the case?

garthvh, 2025-01-06:
> > Router and repeater send metrics every 12 hours.
> 
> Yes, I learnt that some time ago, however @GUVWAF gave me the impression recently that the Device Metric Update Interval setting would still over-ride that and take precedence even when the role was Router. Are you saying that's not the case?

You have to make a manual request, the broadcast interval is unchanged

ashleycawley, 2025-01-06:
> You have to make a manual request, the broadcast interval is unchanged

OK, so there we have it then: the Device Metric Update Interval setting (aka _telemetry.device_update_interval_) **has no effect** when the role is ROUTER.

We still have the mystery of why some router operators in a very quiet Meshtastic area are not metrics every 12 hours. I'll continue to investigate...

GUVWAF, 2025-01-06:
> has no effect when the role is ROUTER.

It does, you can still set it as long as it's at least 30 min. @b8b8 confirmed this is working as expected. 
To be completely sure, I just lowered the minimum to 30 seconds and hard-coded my channel utilization to 30%. I set my role to `ROUTER` and it's happily sending DeviceTelemetry every 30 seconds (see logs below). 

I think the only way you can debug this is checking the serial logs of your router.

> INFO  | ??:??:?? 276 [DeviceTelemetry] Send: air_util_tx=0.094194, channel_utilization=30.000000, battery_level=101, voltage=-0.001000, uptime=276
DEBUG | ??:??:?? 276 [DeviceTelemetry] Partially randomized packet id 886124043
DEBUG | ??:??:?? 276 [DeviceTelemetry] updateTelemetry LOCAL
DEBUG | ??:??:?? 276 [DeviceTelemetry] Node status update: 3 online, 3 total
INFO  | ??:??:?? 276 [DeviceTelemetry] Send packet to mesh
DEBUG | ??:??:?? 276 [DeviceTelemetry] Update DB node 0xbff18ce4, rx_time=0
DEBUG | ??:??:?? 276 [DeviceTelemetry] handleReceived(LOCAL) (id=0x34d12e0b fr=0xbff18ce4 to=0xffffffff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 priority=10)
DEBUG | ??:??:?? 276 [DeviceTelemetry] No modules interested in portnum=67, src=LOCAL
DEBUG | ??:??:?? 276 [DeviceTelemetry] Add packet record (id=0x34d12e0b fr=0xbff18ce4 to=0xffffffff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 priority=10)
DEBUG | ??:??:?? 276 [DeviceTelemetry] Expand short PSK #1
DEBUG | ??:??:?? 276 [DeviceTelemetry] Use AES128 key!
DEBUG | ??:??:?? 276 [DeviceTelemetry] enqueue for send (id=0x34d12e0b fr=0xbff18ce4 to=0xffffffff, WantAck=0, HopLim=3 Ch=0x8 encrypted len=49 hopStart=3 priority=10)
DEBUG | ??:??:?? 276 [DeviceTelemetry] txGood=5,txRelay=1,rxGood=5,rxBad=0
DEBUG | ??:??:?? 276 [DeviceTelemetry] Use channel 0 (hash 0x8)
DEBUG | ??:??:?? 276 [DeviceTelemetry] Expand short PSK #1
DEBUG | ??:??:?? 276 [DeviceTelemetry] Use AES128 key!
DEBUG | ??:??:?? 276 [DeviceTelemetry] decoded message (id=0x34d12e0b fr=0xbff18ce4 to=0xffffffff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 hopStart=3 priority=10)
DEBUG | ??:??:?? 276 [PowerFSM] Radio wait to sleep, txEmpty=0
DEBUG | ??:??:?? 276 [RadioIf] Started Tx (id=0x34d12e0b fr=0xbff18ce4 to=0xffffffff, WantAck=0, HopLim=3 Ch=0x8 encrypted len=49 hopStart=3 priority=10)
DEBUG | ??:??:?? 276 [RadioIf] Packet TX: 600ms
INFO  | ??:??:?? 276 [PowerFSM] setup LORA_DIO1 (GPIO33) with wakeup by external source
INFO  | ??:??:?? 276 [PowerFSM] setup RF95_IRQ (GPIO26) with wakeup by gpio interrupt
INFO  | ??:??:?? 277 [PowerFSM] Exit light sleep gpio: btn=0
INFO  | ??:??:?? 277 [PowerFSM] Exit state: LS
DEBUG | ??:??:?? 277 [PowerFSM] State: NB
DEBUG | ??:??:?? 277 [RadioIf] Completed sending (id=0x34d12e0b fr=0xbff18ce4 to=0xffffffff, WantAck=0, HopLim=3 Ch=0x8 encrypted len=49 hopStart=3 priority=10)
DEBUG | ??:??:?? 280 [RadioIf] Lora RX (id=0x34d12e0b fr=0xbff18ce4 to=0xffffffff, WantAck=0, HopLim=2 Ch=0x8 encrypted len=49 rxSNR=10.25 rxRSSI=-65 hopStart=3)
DEBUG | ??:??:?? 280 [RadioIf] Packet RX: 600ms
DEBUG | ??:??:?? 280 [Router] Rx someone rebroadcasting for us (id=0x34d12e0b fr=0xbff18ce4 to=0xffffffff, WantAck=0, HopLim=2 Ch=0x8 encrypted len=49 rxSNR=10.25 rxRSSI=-65 hopStart=3)
DEBUG | ??:??:?? 280 [Router] Didn't find pending packet
DEBUG | ??:??:?? 280 [Router] Found existing packet record for fr=0xbff18ce4,to=0xffffffff,id=0x34d12e0b
DEBUG | ??:??:?? 280 [Router] Add packet record (id=0x34d12e0b fr=0xbff18ce4 to=0xffffffff, WantAck=0, HopLim=2 Ch=0x8 encrypted len=49 rxSNR=10.25 rxRSSI=-65 hopStart=3)
DEBUG | ??:??:?? 280 [Router] Ignore dupe incoming msg (id=0x34d12e0b fr=0xbff18ce4 to=0xffffffff, WantAck=0, HopLim=2 Ch=0x8 encrypted len=49 rxSNR=10.25 rxRSSI=-65 hopStart=3)
DEBUG | ??:??:?? 280 [Router] Incoming msg was filtered from 0xbff18ce4
WARN  | ??:??:?? 281 [Position] Ch. util >25%. Skip send
WARN  | ??:??:?? 286 [Position] Ch. util >25%. Skip send
INFO  | ??:??:?? 287 [PowerFSM] lsEnter begin, ls_secs=86400
INFO  | ??:??:?? 287 [PowerFSM] setup LORA_DIO1 (GPIO33) with wakeup by external source
INFO  | ??:??:?? 287 [PowerFSM] setup RF95_IRQ (GPIO26) with wakeup by gpio interrupt
INFO  | ??:??:?? 317 [PowerFSM] Exit light sleep cause: 4
INFO  | ??:??:?? 317 [PowerFSM] setup LORA_DIO1 (GPIO33) with wakeup by external source
INFO  | ??:??:?? 317 [PowerFSM] setup RF95_IRQ (GPIO26) with wakeup by gpio interrupt
INFO  | ??:??:?? 317 [PowerFSM] Exit light sleep cause: 4
DEBUG | ??:??:?? 317 [Power] Battery: usbPower=1, isCharging=0, batMv=-1, batPct=0
WARN  | ??:??:?? 317 [Position] Ch. util >25%. Skip send
WARN  | ??:??:?? 317 [GPS] 1 new GPS checksum failures, for a total of 1
INFO  | ??:??:?? 317 [PowerFSM] setup LORA_DIO1 (GPIO33) with wakeup by external source
INFO  | ??:??:?? 317 [PowerFSM] setup RF95_IRQ (GPIO26) with wakeup by gpio interrupt
INFO  | ??:??:?? 347 [PowerFSM] Exit light sleep cause: 4
INFO  | ??:??:?? 347 [PowerFSM] setup LORA_DIO1 (GPIO33) with wakeup by external source
INFO  | ??:??:?? 347 [PowerFSM] setup RF95_IRQ (GPIO26) with wakeup by gpio interrupt
INFO  | ??:??:?? 348 [PowerFSM] Exit light sleep cause: 4
DEBUG | ??:??:?? 348 [Power] Battery: usbPower=1, isCharging=0, batMv=-1, batPct=0
WARN  | ??:??:?? 348 [Position] Ch. util >25%. Skip send
INFO  | ??:??:?? 348 [DeviceTelemetry] Send: air_util_tx=0.110861, channel_utilization=30.000000, battery_level=101, voltage=-0.001000, uptime=348
DEBUG | ??:??:?? 348 [DeviceTelemetry] Partially randomized packet id 2507858444
[... cut ...]
DEBUG | ??:??:?? 349 [RadioIf] Completed sending (id=0x957ae20c fr=0xbff18ce4 to=0xffffffff, WantAck=0, HopLim=3 Ch=0x8 encrypted len=49 hopStart=3 priority=10)
DEBUG | ??:??:?? 351 [RadioIf] Lora RX (id=0x957ae20c fr=0xbff18ce4 to=0xffffffff, WantAck=0, HopLim=2 Ch=0x8 encrypted len=49 rxSNR=11.5 rxRSSI=-64 hopStart=3)
DEBUG | ??:??:?? 351 [RadioIf] Packet RX: 600ms
DEBUG | ??:??:?? 351 [Router] Rx someone rebroadcasting for us (id=0x957ae20c fr=0xbff18ce4 to=0xffffffff, WantAck=0, HopLim=2 Ch=0x8 encrypted len=49 rxSNR=11.5 rxRSSI=-64 hopStart=3)

GUVWAF, 2025-01-06:
I already showed you the code and now I proved that it's working as expected for me, and b8b8 confirmed this as well.

Indeed the default is 12 hours, but you can still set it differently so "the broadcast interval is unchanged" is then not true.

ashleycawley, 2025-01-06:
> I already showed you the code and now I proved that it's working as expected for me, and b8b8 confirmed this as well.

You're assuming you're talking to someone who can interpret the code, which I can't reliably. Also even what I can understand I don't know how it fits into the bigger picture exactly with the rest of the code, so I don't know the nuances like you folks.

> Indeed the default is 12 hours, but you can still set it differently so "the broadcast interval is unchanged" is then not true.

I understand the default for router is 12 hours, I don't know how many times I have to say that.

So what do you make of my router + the other router operator in my county who have both set their telemetry.device_update_interval to 1800 but aren't receiving anymore than one or two a day?

The above would suggest it is being limited to once every 12 hours as @garthvh said and possibly encountering chan util issues.

Do you think the Router needs a reboot after changing the setting? Traditionally I've been used to it rebooting itself if it requires it. I'm just trying to think of whatever else could possibly be causing a disparity between what you're seeing versus what we're seeing. We are in the UK on 868MHz if that makes any difference to logic used in the firmware.

GUVWAF, 2025-01-06:
> You're assuming you're talking to someone who can interpret the code

You were asking for me to confirm the logic. I can of course say "yes, checks out", but then I'd rather also show it. I also tried to reproduce your issue, but I couldn't.

> So what do you make of my router + the other router operator in my county who have both set their telemetry.device_update_interval to 1800 but aren't receiving anymore than one or two a day?

I'm sorry, but at this point I don't know why you're having these issues and I think the only way would be to check the serial logs of your router.

> Do you think the Router needs a reboot after changing the setting? 

No, shouldn't be necessary and since you got the correct setting back when by querying from the CLI, you should be good.

GUVWAF, 2025-01-06:
@ashleycawley Ah, finally found something. Your own transmit air utilization is quite high, when you requested it, it was 6.20%. Since you are using a region with 10% duty cycle, it won't transmit DeviceTelemetry above 5% transmit utilization to avoid reaching the point where you cannot send anything at all anymore.

The reason this happens only in `ROUTER` role, is because then you're rebroadcasting way more packets.

ashleycawley, 2025-01-06:
Interesting find! I see, so that's the "_Transmit air utilization_", there's some logic that says not to transmit device metrics if its above 5%? Nice find. Would that be stated in the same _airtime.h_ file we were talking about previously?

GUVWAF, 2025-01-06:
Yes, it's 50% of the duty cycle, so 50% of 10% is 5%:
https://github.com/meshtastic/firmware/blob/4c3a3ca47d4ee50e2d946956fe614d6e7549df8a/src/airtime.h#L72

ashleycawley, 2025-01-06:
I think you've found it @GUVWAF - Over the past 2.5 hours I've been requesting and logging data via the CLI from my Router and look at the Airtime Utilization:
```
Transmit air utilization: 6.58%
Transmit air utilization: 6.79%
Transmit air utilization: 6.83%
Transmit air utilization: 6.78%
Transmit air utilization: 6.41%
```
All over 5%, meaning the logic in that code-block would prevent Device Metrics from being published.

I've then looked at the other Router owned by someone else 10 miles away and although the device metrics is still vastly lacking in frequency, it transmitted more metrics than my router but all the data has a airtime of 5% or less, not a single one over that mark.

b8b8, 2025-01-07:
So i may have missed it in the convo (and my router has been acting as expected up in Canada) but where/how-busy is your router? As GUVWAF established above, the air time cut-off of 5%, its there not still the 40% channel utilization hard cut-off. If your router is really busy, you may never get telemetry based on that.

PR #4468 (2024-08-15): Router and sensor are impolite
Closes #4466