meshtastic/firmware #4470: Title: High Frequency Mesh Communication Observed in Version 2.4.x

Oliver0804, 2024-08-15:
Description:
I've noticed a significant increase in mesh communication frequency in devices running version 2.4.x. Specifically, in version 2.4.1, devices like the T-Echo and T-Beam are frequently engaging in RF interactions, even when there are no other users or messages being exchanged. This behavior was not present in version 2.3.x, despite having more devices in the mesh network at that time.
[image]

Steps to Reproduce:

Set up two devices (e.g., T-Echo and T-Beam) with firmware version 2.4.1.
Ensure there are no active users or messages being sent in the mesh network.
Observe the frequency of RF interactions between the devices.
Expected Behavior:
The mesh communication should only occur when necessary, similar to the behavior in version 1.x.

Actual Behavior:
Devices on version 2.4.x are frequently communicating via RF, even when there is no apparent reason to do so.

Environment:

Devices: T-Echo, T-Beam
Firmware Version: 2.4.1
Previous Version with Expected Behavior: 2.3.x
Additional Information:
This issue could potentially lead to unnecessary power consumption and might affect the overall performance of the mesh network.

thebentern, 2024-08-15:
@Oliver0804 can you post a meshtastic --info configuration dump of the nodes?

Oliver0804, 2024-08-15:
> @Oliver0804 can you post a `meshtastic --info` configuration dump of the nodes?

I've made some progress. I encountered the same issue while using version 2.3.x, so I went back to check my iPhone. I found that the problem disappears when the MQTT feature is disabled—there are no excessively frequent TX transmissions. However, when MQTT is enabled, the T-Echo sends messages abnormally, 5 to 7 times in quick succession.

App version 2.4.1(1121)

Oliver0804, 2024-08-15:
> configuration dump of the nodes?

[log.txt](https://github.com/user-attachments/files/16628218/log.txt)

fifieldt, 2024-08-20:
@Oliver0804 did you see https://github.com/meshtastic/firmware/issues/4502 ?

perhaps it was that?

thebentern, 2024-08-20:
> @Oliver0804 did you see #4502 ?
> 
> perhaps it was that?

I believe this issues pre-dates that bug, and they have neighbors disabled. 
@Oliver0804 nothing looks suspicious in the info dump from what I can see. It's possible that msh/TW is an extremely busy topic on MQTT. If you can post some `meshtastic --noproto` logs from when this is happening, we might be able to tell one way or another.