meshtastic/firmware #4484: [Bug]: no timestamp in 2.4.3 alpha

OE5DRO, 2024-08-17:
### Category

Other

### Hardware

Rak4631

### Firmware Version

2.4.3 alpha

### Description

i am very fast in updating new firmware. so i did update my rak4631 and tbeam to 2.4.3.
after that i was waiting for the timestamp in meshtastic app (mac, iPhone, Android), but nothing happened.
i made reboot, factory reset but nothing helped. then i went back to 2.4.2 and time was visible again in my apps.
i use no GPS hardware and location is fixed. maybe not an issue because of alpha, but want you to know ...
tnx 4 this nice firmware!

Spogster, 2024-08-18:
I have been suffering with this since 2.4.1 does seem to intermittently work though check out my bug report #4416 to see if it looks similar.

thebentern, 2024-08-18:
This is fixed in the latest 2.4.3 hotfix release
https://github.com/meshtastic/firmware/releases/tag/v2.4.3.1bbc273

closed 2024-08-18 by thebentern