meshtastic/firmware #4501: [Bug]: GPS buffer size check off-by-one?

fifieldt, 2024-08-19:
### Category

Serial

### Hardware

Seeed Wio Tracker 1110

### Firmware Version

2.4.3.7129cee9

### Description

In `arch/nrf52/nrf52.ini`, we set `SERIAL_BUFFER_SIZE=1024`. and serial connections are created with this as the buffer limit.

and in src/gps/GPS.cpp we check that it's not overflowing:
```
1774 #ifdef SERIAL_BUFFER_SIZE
1775     if (_serial_gps->available() >= SERIAL_BUFFER_SIZE - 1) {
1776         LOG_WARN("GPS Buffer full with %u bytes waiting. Flushing to avoid corruption.\n", _serial_gps->available());
1777         clearBuffer();
1778     }
1779 #endif
```

I noticed in logs for a wio-tracker-wm1110 that each time the GPS went active, the below message was repeated:
`[GPS] GPS Buffer full with 1023 bytes waiting. Flushing to avoid corruption.`

I'm not very familiar with this code, but 1023 bytes in a 1024 byte buffer seems fine? Perhaps the extra "- 1" in the check should be removed?

### Relevant log output

```Shell
[GPS] GPS Buffer full with 1023 bytes waiting. Flushing to avoid corruption.
```

jp-bennett, 2024-08-19:
I think that missing byte is the null terminator. The problem is that with the buffer completely full, the GPS output has almost certainly overwritten itself in some places, leading to some very weird results when trying to parse it.

fifieldt, 2024-08-19:
That makes sense, thanks. I guess now to investigate why a GPS without a lock and probably not receiving a whole heap of signals is blasting through its buffer.

closed 2024-08-19 by fifieldt

GPSFan, 2024-08-19:
A GPS without a lock will still output NMEA messages, if improperly configured (or not configured at all) the chip may output more messages than the buffer can handle.
For example, the L76K defaults to the following:
RMC, GGA, GSV, GSA, VTG, GLL, TXT, and ZDA
The u-blox M8N defaults to these:
GGA, GLL, GSA, GSV, RMC, VTG, TXT
Even with no data this could overwhelm the serial buffer.
When a u-blox module is detected, the first thing we do is turn OFF all NMEA messages.