meshtastic/firmware #4502: [Bug]: 2.4.3 Hotfix introduces NeighbourInfo Loop - Sending every second.

Talie5in, 2024-08-19:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

2.4.3.1 (Hotfix)

### Description

2.4.3.1 Hotfix introduces NeighbourInfo Loop - Sending every second

Noted on Graphs Collected causing high RF use and mention also in Discord thanks to tjr1
https://discord.com/channels/867578229534359593/919642584480112750/1274887561541713941

From my Graphs, you can see where I turned on NeighbourInfo at 01/01:30 AM and turned off again at 15:00 PM
[image]

Last 6 hours data for another node I have on 2.4.3.1 (Hotfix)
[image]

Talie5in, 2024-08-19:
Note: After re-enabling with Debug Log it's no longer doing it, I have the same report from another user on the Mesh, dropped ChUtil from 40% back to sub 1%.

Just messing with device to see if it waits before it breaks/loops.

Talie5in, 2024-08-19:
Logs attached to Original Post.

thebentern, 2024-08-19:
Ironically, two characters differences produced the opposite of desired behavior

PR #4504 (2024-08-19): We need Milliseconds not... rapid fire NeighbourInfo!
Fixes #4502

Thanks to @GUVWAF  for the pointer.
closed 2024-08-19 by thebentern