meshtastic/firmware #4518: Fresh install communications issues

Lolagatorade, 2024-08-21:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

2.5.0

### Description

When upgrading things worked fine for the most part on the aspect of sending to other devices, other than some key exchange issues and mismatch

Doing a fresh install, which means erasing and flashing resulted in only the public channel being functional at times it will go through, but it says either no channel or acknowledge by another node even though I am well within range of the target radio

[image]
[image]
[image]

Lolagatorade, 2024-08-21:
[image]
[image]

thebentern, 2024-08-21:
This was an issue with reporting ACKs in the iOS app fixed here:
https://github.com/meshtastic/Meshtastic-Apple/pull/882

It should be in test flight shortly (if not already)

Lolagatorade, 2024-08-21:
[image]

Recently got this error too might be related to this issue.