meshtastic/firmware #4521: [Bug]: Critical fault #11

uguraltinsoy, 2024-08-21:
### Category

Other

### Hardware

DIY

### Firmware Version

2.4.2.5b45303

### Description

I made the connections as shown below and completed the setup, but I am encountering an error during installation. Despite the error, the device powers on, but I get a "Critical fault #11" error, and I cannot complete the setup. I tried two different ESP32s and two different Ra-02 modules, so it's unlikely to be a hardware-related issue.

https://adrelien.com/blog/meshtastic-diy-how-to-build-your-own-meshtatic-node-esp32-lora-radio/amp/
[image]

[image]
[image]
[image]

Lolagatorade, 2024-08-22:
Had the error too but my device was functional after a reboot and it was only a one time error.

Lolagatorade, 2024-08-22:
[image]