meshtastic/firmware #4524: [Bug]: Remote Administration on the Technical Preview not working via CLI

DRCIT-Tony, 2024-08-21:
### Category

BLE, Serial

### Hardware

T-Beam, T-Echo, Rak4631, Heltec V3

### Firmware Version

2.5.0.ab7de7f

### Description

Since this is a tech preview, I was not sure if the remote administration was broken.

I have 5 nodes that I am testing out: A T-Echo, 2x Heltec v3, T-Beam v1, and a RAK 4631.

Under the stable versions, I can do remote node admin via CLI, but under the tech preview, I am not able to. All nodes were factory reset and flashed to the technical preview fw, locally set region, owner, primary channel (set something other than the default as I did not want to send random msgs out to others during the testing), and secondary channel set to admin. After setting up the RAK, I locally set the URL to the rest of the devices.

They can all see each other, but I cannot get any responses remotely. I then tried doing it over my iPhone and had the same issue. I can confirm that this does work on the stable version. Setting the device to managed did not have any effect. 

I will get one of two responses when I send a command using --dest !<node-id> --any command:
```
Connected to radio
Requesting current config from remote node (this can take a while).
Error on response: NO_RESPONSE
Completed getting preferences
```
or, not as often, but:
```
Connected to radio
Requesting current config from remote node (this can take a while).
Aborting due to: Timed out waiting for an acknowledgment
```
Since I have already spent many hours on this, I figured I would ask if this is a known issue or not. Also, I am not sure if the CLI version of Meshtastic app is on or supports the technical preview like the iPhone, so I am going to do one more test where I factory reset everything and manage it solely from the iPhone instead of CLI.

Thanks for your help!

python v3.12.5 | pip3 v24.2 | meshtastic v2.3.15 (pwsh) | meshtastic v2.5.0 (iOS)

garthvh, 2024-08-21:
You will need to enable legacy admin to use an admin channel

jp-bennett, 2024-08-21:
We've gotten a couple reports of folks having issues. When using the channel named "admin", you will need to enable "Legacy Administration" on the remote node. There's also an added security piece that isn't fully working in the apps yet, that will cause problems with actually making changes on the remote node.

DRCIT-Tony, 2024-08-21:
One more question, is there a way to set Legacy Admin via CLI? When I set it via iOS, it immediately unsets on save/reboot.

jp-bennett, 2024-08-21:
> One more question, is there a way to set Legacy Admin via CLI? When I set it via iOS, it immediately unsets on save/reboot.

I'm reasonably sure iOS is actually setting it, and then failing to toggle the slider when it reads state back. The Python CLI should update before too much longer, with support for the new stuff.

DRCIT-Tony, 2024-08-21:
Ah ok... thank you.  I will check --info.

Thanks again!

DRCIT-Tony, 2024-08-21:
So, I can confirm that under iOS v2.5.0, the setting for both Legacy Administration, as well as Managed Device, has no effect on the RAK4631 and remains unset. I am able to set it to Managed Device via CLI. But, since I am not able to set the flag for Legacy Administration, I am not able to remote admin under v2.5.0. Since the RAK was the stationary node we were testing, it was the only device that needed the remote admin. I'll switch back to the stable versions for now, but will look forward to the stable release of v2.5.0.

Thanks again for your help!

closed 2024-08-21 by DRCIT-Tony

W1LMS, 2024-08-24:
Hello everyone. I was interested in knowing if there is a way to downgrade to the previous version of Meshtastic on iOS?

It seems that the new v2.5.x on iOS does not allow me to remote admin my devices, and that the Legacy Admin setting is inoperative. It seems to only allow the new PKI Admin regardless of turning on Legacy Admin, and does not transmit the Legacy Admin setting to the device, nor the Device Managed setting to of the nodes on the v2.5.0. So I downgraded the devices back to the latest stable version, and I can now remote manage from CLI, but I can no longer use the Meshtastic app on iOS to remote manage since it will only support PKI, despite setting Legacy Admin. Unless there is some special trick to getting the LA to work, the standard methods of saving changes to the device is not working.

Prior to the update, I had no issues, so I am hoping to be able to downgrade.

I also have a feature request (though I do not know what is involved in this method), but I recommend that technical previews, or similar,  be released through iOS Test Flight. That way, the working stable app is not bricked by an update, and in effect, you can test the apps out side by side (if you wanted to). Or just have 2 different apps, one stable, one alpha or something.

Well, thanks for your time. Have a great weekend!

73,
W1LMS

djholt, 2024-09-13:
In the current state of 2.5, should I expect the admin channel to work? I'm still seeing NO_RESPONSE even with security.adminChannelEnabled set to true.

I'm also comfortable to test the new signed administrative controls if the API and firmware work are far enough along. I'd rather keep with 2.5 and follow along with current development efforts.

jp-bennett, 2024-09-13:
> @jp-bennett In the current state of 2.5, should I expect the admin channel to work? I'm still seeing NO_RESPONSE even with security.adminChannelEnabled set to true.

It is expected to work, with an updated client. We added a session key to the legacy admin channel, to prevent replay attacks.

There is a potential issue, where if both the local and remote nodes are 2.5, the local node will try to admin using PKI rather than the admin channel. We're working on a fix for that case.

> 
> I'm also comfortable to test the new signed administrative controls if the python API and firmware work are far enough along. I'd rather keep with 2.5 and follow along with current development efforts.
With the Python 2.5 release, that should be working, too. Copy the local node's public key into the remote's admin key, and if they have successfully key exchanged, it should work.

djholt, 2024-09-13:
Oh, good timing. I was just updating my comment as you posted.

> There is a potential issue, where if both the local and remote nodes are 2.5, the local node will try to admin using PKI rather than the admin channel. We're working on a fix for that case.
> 
Ah, this explains it. Both nodes are 2.5 latest.

I did just try what you'd suggested. https://github.com/meshtastic/python/pull/672 helped to me set the admin key on the remote node, and it worked perfectly. This is awesome.

Quick question, while you're here: what's the best method to generate a new key pair programatically?

Thanks @jp-bennett !

jp-bennett, 2024-09-13:
> what's the best method to generate a new key pair programatically?

The web client has key generation working. Or you can use the Python client to set the private key to blank using `base64:`  and the node will generate a new keypair on next boot.

djholt, 2024-09-13:
Oh, very nice. That's great. I'm loving all of the upgrades coming with 2.5. Thanks for the replies and all that you're doing here!

djholt, 2024-09-14:
I've been spending some time testing PKI-based admin (admin channel disabled), and from time to time, I encounter the following timeout (also mentioned in the original issue above):

`Aborting due to: Timed out waiting for an acknowledgment`

I'm having a hard time diagnosing what's causing it, but it tends to always happen when I update keys. Occasionally, the issue will subside on its own after some number of minutes, and I can always recover to a working state by re-flashing. Restarting doesn't help. It doesn't appear to be at all related to signal intermittency as I had originally thought, but of course the error will always occur if channel or radio configs are mismatched.

Are there any temporary state configs or session-like keys that expire or reset after some amount of time? That might explain the behavior I'm seeing, but it's just a thought. Thanks for any suggestions or ideas!

jp-bennett, 2024-09-15:
@djholt We don't currently have any sort of re-keying solution in place. Specifying the admin key isn't enough to guarantee decryption, as the public key still has to be in the nodeDB.

The model is really that nodes should keep their public/private key for as long as possible, and we encourage the end user to back up their private key, because trying to publish a new key is a pain. I'm curious if there's a use case we haven't considered, that would necessitate re-keying.

The sessionkey is only required when making a change, not requesting current config. There's even an admin message that returns an empty object, that is strictly for getting a fresh session key. That session key expires after 300 seconds.

djholt, 2024-09-15:
Okay, that makes sense, and answers a few questions. I feel like the situation I ran into is not typical or common. Following some testing and experimentation, I was trying to consolidate my node configs to use a single admin key and afterward ran into the errors described above. Flushing out the node DB on a couple of nodes was ultimately the trick to clear it up. The process of debugging was a good opportunity to learn more. I'll chime in if I experience issues during typical use, but otherwise I can't identify changes that are needed.

Thank you! I appreciate the discussion and opportunity to learn.