meshtastic/firmware #4525: [Feature Request]: Public MQTT location sharing

daviesgeek, 2024-08-21:
### Platform

NRF52, ESP32, RP2040, Linux Native, other

### Description

My proposal for fixing public mqtt location sharing:

Split out the MQTT config to have a specific section to report to the public mqtt server. Change the password, make a new server, however it needs to happen, but make it very hard if not impossible to set up the public mqtt server in the main mqtt settings.

- Make it VERY clear with that your location data is being shared publicly and without knowledge of who is picking it up
- Default to showing low precision and maybe even prompt the user to say yes/no when changing said precision
- ONLY ever send your node's nodeinfo packet. No more forwarding other packets to the public server, etc. (obv you can still setup a second mqtt server in the main mqtt settings screen, but this is just for the public server

Also maybe setup some controls and application keys to access the mqtt server. I.e., public map developers have to request access and provide a privacy policy clearly outlined so the end user(s) can see what's happening with their data.

GUVWAF, 2024-08-21:
I think most of this is already in place with the "Map Reporting" setting, which is a collection of information that's only published to MQTT and not on RF.

daviesgeek, 2024-08-21:
The "map reporting" as best as i understand is based off your mqtt server settings, whereas what I'm proposing is a specific public mqtt config where you choose to send data to the public server (or not) in addition to whatever you have configured in your mqtt server settings.
While the map reporting flag could be repurposed for this use, i would suggest that it be made a completely new section to be very clear about what it's doing and how it's working.

closed 2024-08-21 by garthvh

daviesgeek, 2024-08-22:
@garthvh It'd be nice to offer an explanation rather than just closing with no comment. There were a TON of comments in the #map channel in discord today around how and why this might need to be solved. This wasn't necessarily meant to be "do it this way", this can be a discussion thread around how to handle this issue.

GUVWAF, 2024-08-22:
The issue builds up from the premise of "make it harder to set the public MQTT server", but I don't think this will be a long-term solution. Other public MQTT servers are becoming popular, even though people have to manually change to this server already. E.g. a [public MQTT server in The Netherlands](https://map.meshnet.nl/) has around 750 nodes, versus around 900 on Meshtastic's one.

Here are my comments on the other parts of your issue:

> Default to showing low precision and maybe even prompt the user to say yes/no when changing said precision

The first part is already done in #4467.

> Make it VERY clear with that your location data is being shared publicly and without knowledge of who is picking it up

Yes, maybe this can be made more clear, but (1) this is only true when sharing it on a channel with known key and (2) not a unique problem because of the public MQTT or even MQTT in general. 

> ONLY ever send your node's nodeinfo packet. 

You have to explicitly enable uplink on a public channel already for all packets to show up. "Map Reporting" does roughly what your proposing here.

> Also maybe setup some controls and application keys to access the mqtt server.

I believe this is already in the works.

daviesgeek, 2024-08-22:
@GUVWAF No, my premise was "help the user understand what they're doing", the "make it harder" portion was a side effect of just making sure end users don't shoot themselves in the foot.

I think overall my proposal was misunderstood. I'm a bit disheartened by the lack of understanding and collaboration on this, but that's alright, it seems like the dev team has a way forward and am interested to see what that is when the work is complete.