meshtastic/firmware #4529: Favorites not sticking

Lolagatorade, 2024-08-22:
### Firmware Version

2.5.0

### What did you do?

Favorites not sticking after favoriting 

### Expected Behavior

Favorite nodes should stick 

### Current Behavior

After some time it will forget favorite nodes 

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-08-22:
@jp-bennett I am seeing this issue on 2.5, the app side add and remove is all working, but we are losing favorites from the node db over time.  I started with 6 and now have 4

Lolagatorade, 2024-08-22:
> @jp-bennett I am seeing this issue on 2.5, the app side add and remove is all working, but we are losing favorites from the node db over time.  I started with 6 and now have 4

Yeah, there seems to be some strange paradoxical things happening in this new update 

Same thing with the signal values I don't know if you've experienced it yet or with the GPS

Talie5in, 2024-08-22:
> Same thing with the signal values I don't know if you've experienced it yet

Is this a LR1110 based radio? (Seeed T1000-E for example?)

Lolagatorade, 2024-08-22:
> > Same thing with the signal values I don't know if you've experienced it yet
> 
> 
> 
> Is this a LR1110 based radio? (Seeed T1000-E for example?)

From what I know it is a no. Ime using a heltec v3 whatever comes stock with it

Lolagatorade, 2024-08-22:
> > Same thing with the signal values I don't know if you've experienced it yet
> 
> 
> 
> Is this a LR1110 based radio? (Seeed T1000-E for example?)

https://github.com/meshtastic/Meshtastic-Apple/issues/889

garthvh, 2024-08-23:
> > > Same thing with the signal values I don't know if you've experienced it yet
> > 
> > 
> > Is this a LR1110 based radio? (Seeed T1000-E for example?)
> 
> [meshtastic/Meshtastic-Apple#889](https://github.com/meshtastic/Meshtastic-Apple/issues/889)

The signal meter is only on nodes you can connect to directly over Lora

jp-bennett, 2024-08-23:
I did some local verbose debugging, and noticed a few things, at least on my devices. The number of favorites in firmware didn't actually change. iOS was dropping a node as a favorite, when the remote node sent in a position packet. And when manually disconnecting and reconnecting Bluetooth, the list of favorite nodes is restored, until the next position packet comes in.

closed 2024-10-07 by thebentern