meshtastic/firmware #4530: [Feature Request]: "Screen Paused" on eink is a useless feature

battlehax, 2024-08-22:
### Platform

other

### Description

There's not much reason to refresh an e-ink display to show "Screen Paused" messages as the display (should) only update when there's new cards to show.

Firmware for devices with an e-ink display should set:

Radio Configuration -> Display -> Screen Timeout (seconds) to 2147483647
or
display.screen_on_secs = 2147483647

todd-herbert, 2024-08-22:
> "Screen Paused" on eink is a useless feature
 
I certainly don't like the screensaver, although I'm not sure about just disabling it by default either.

> Radio Configuration -> Display -> Screen Timeout (seconds) to 2147483647

As a quick note: for E-Ink displays specifically, setting this to 0 will also disable the timeout.

___

The `Screen` class was originally written for OLED displays, early in Meshtastic's history(?). Support for other types of display (TFT, E-Ink) has been tacked-on over the years. All rendering takes place as if drawing for an OLED display, then the resulting image is fed out to a TFT / E-Ink driver.

This is a bit of a logistical challenge with E-Ink displays, as the underlying driver is fed a steady stream of images at a relatively high frame-rate, with no real context. The E-Ink code jumps through quite a few hoops attempting to identify which of these images represent meaningful changes in display state, and decide how to handle that.

The information on the current `Screen` class tends to be fairly "dynamic": given the opportunity, the UI would continue to feed new images to the E-Ink display, performing a few updates a minute. Display wear was a concern here, although battery would also be impacted (by the display refresh, but also possibly by image processing).

When display timeout it reached, an OLED display very clearly turns off. For quite some time, E-Ink displays would simply cease updating, leaving stale information on the display. The screensaver was a low-impact attempt to alert users that the display information is out of date.

___

Unofficially, I am in the early stages of re-working the E-Ink UI and driver. I'm hoping to cut the screensaver problem off "at the source", with "event driven" rendering, rather the current system of interpreting output from the OLED display.

battlehax, 2024-08-22:
Thank you for the explanation. I was wrong, and it's not a good idea if it can cause hardware damage.

closed 2024-08-22 by battlehax