meshtastic/firmware #4531: [Bug]: 2.4.3.91d6612 Tbeam failed to find radio

fifieldt, 2024-08-22:
### Category

Hardware Compatibility

### Hardware

T-Beam

### Firmware Version

2.4.3.91d6612

### Description

After updating from 2.4.0 to 2.4.3.91d6612, tbeam can no longer detect a radio.

No other changes were made to the device.

### Relevant log output

```Shell
DEBUG | ??:??:?? 2 RF95Interface(cs=18, irq=26, rst=23, busy=33)

INFO | ??:??:?? 2 Starting meshradio init...

DEBUG | ??:??:?? 2 (bw=0, sf=0, cr=4/0) packet symLen=2147483647 ms, payloadSize=0, time -1 ms

DEBUG | ??:??:?? 2 (bw=0, sf=0, cr=4/0) packet symLen=2147483647 ms, payloadSize=253, time -1 ms

INFO | ??:??:?? 2 Radio freq=920.000, config.lora.frequency_offset=0.000

INFO | ??:??:?? 2 Set radio: region=TW, name=Custom, config=0, ch=-1397824096, power=27

INFO | ??:??:?? 2 Radio myRegion->freqStart -> myRegion->freqEnd: 920.000000 -> 925.000000 (5.000000 MHz)

INFO | ??:??:?? 2 Radio myRegion->numChannels: -1 x 0.000kHz

INFO | ??:??:?? 2 Radio channel_num: -1397824095

INFO | ??:??:?? 2 Radio frequency: 920.000000

INFO | ??:??:?? 2 Slot time: 42 msec

INFO | ??:??:?? 2 Set radio: final power level=20

DEBUG | ??:??:?? 2 Current limit set to 100.000000

DEBUG | ??:??:?? 2 Current limit set result 0

INFO | ??:??:?? 2 RF95 init result -8

INFO | ??:??:?? 2 Frequency set to 920.000000

INFO | ??:??:?? 2 Bandwidth set to 0.000000

INFO | ??:??:?? 2 Power output set to 20

WARN | ??:??:?? 2 Failed to find RF95 radio

DEBUG | ??:??:?? 2 SX126xInterface(cs=18, irq=33, rst=23, busy=32)

DEBUG | ??:??:?? 2 SX126X_DIO3_TCXO_VOLTAGE defined, using DIO3 as TCXO reference voltage at 1.800000 V

INFO | ??:??:?? 2 Starting meshradio init...

DEBUG | ??:??:?? 2 (bw=0, sf=0, cr=4/0) packet symLen=2147483647 ms, payloadSize=0, time -1 ms

DEBUG | ??:??:?? 2 (bw=0, sf=0, cr=4/0) packet symLen=2147483647 ms, payloadSize=253, time -1 ms

INFO | ??:??:?? 2 Radio freq=920.000, config.lora.frequency_offset=0.000

INFO | ??:??:?? 2 Set radio: region=TW, name=Custom, config=0, ch=-1397824096, power=27

INFO | ??:??:?? 2 Radio myRegion->freqStart -> myRegion->freqEnd: 920.000000 -> 925.000000 (5.000000 MHz)

INFO | ??:??:?? 2 Radio myRegion->numChannels: -1 x 0.000kHz

INFO | ??:??:?? 2 Radio channel_num: -1397824095

INFO | ??:??:?? 2 Radio frequency: 920.000000

INFO | ??:??:?? 2 Slot time: 42 msec

INFO | ??:??:?? 2 Set radio: final power level=22

INFO | ??:??:?? 28 SX126x init result -2

WARN | ??:??:?? 28 Failed to find SX1262 radio

DEBUG | ??:??:?? 28 SX126xInterface(cs=18, irq=33, rst=23, busy=32)

DEBUG | ??:??:?? 28 SX126X_DIO3_TCXO_VOLTAGE defined, using DIO3 as TCXO reference voltage at 1.800000 V

INFO | ??:??:?? 28 Starting meshradio init...

DEBUG | ??:??:?? 28 (bw=0, sf=0, cr=4/0) packet symLen=2147483647 ms, payloadSize=0, time -1 ms

DEBUG | ??:??:?? 28 (bw=0, sf=0, cr=4/0) packet symLen=2147483647 ms, payloadSize=253, time -1 ms

INFO | ??:??:?? 28 Radio freq=920.000, config.lora.frequency_offset=0.000

INFO | ??:??:?? 28 Set radio: region=TW, name=Custom, config=0, ch=-1397824096, power=27

INFO | ??:??:?? 28 Radio myRegion->freqStart -> myRegion->freqEnd: 920.000000 -> 925.000000 (5.000000 MHz)

INFO | ??:??:?? 28 Radio myRegion->numChannels: -1 x 0.000kHz

INFO | ??:??:?? 28 Radio channel_num: -1397824095

INFO | ??:??:?? 28 Radio frequency: 433.125000

INFO | ??:??:?? 28 Slot time: 42 msec

INFO | ??:??:?? 28 Set radio: final power level=22

INFO | ??:??:?? 55 SX126x init result -2

WARN | ??:??:?? 55 Failed to find SX1268 radio

DEBUG | ??:??:?? 55 Initializing MQTT

ERROR | ??:??:?? 55 NOTE! Recording critical error 3 at src/main.cpp:1040
```

fifieldt, 2024-08-22:
2.4.2.5b45303 - same issue.
2.4.1.394e0e1 - same issue
2.4.0.46d7b82 - same issue

so it looks like it's not the firmware's fault.

GUVWAF, 2024-08-22:
> (bw=0, sf=0, cr=4/0)
> Set radio: region=TW, name=Custom, config=0, ch=-1397824096, power=27

The LoRa settings are invalid. This can probably be avoided by #4167.

fifieldt, 2024-08-22:
Thanks @GUVWAF ! Closing this to concentrate the issue into #4167.

closed 2024-08-22 by fifieldt