meshtastic/firmware #4533: [Feature Request]: Cannot Change Time Zone

uguraltinsoy, 2024-08-22:
### Platform

ESP32

### Description

When displaying the GPS instant time, it appears as GMT-0 time, we use GMT-3 in my country, how can I show it as GMT-3?

[image]

closed 2024-08-22 by garthvh

uguraltinsoy, 2024-08-22:
> There is a setting under device config

Can you explain in more detail please

ponzano, 2024-08-30:
> > There is a setting under device config
> 
> 
> 
> Already set according to GMT-3 but it doesn't look like this on the oled screen
> 
> 
> 
> [image]
> 
> 

I think it only works if you use this format 
https://stackoverflow.com/questions/73935736/time-zone-format

https://www.gnu.org/software/libc/manual/html_node/TZ-Variable.html

So for you it should be

CET-5CEST,M3.5.0/2:00:00,M10.4.0/3:00:00

Since Central European time is 2h ahead of gmt + your 3 hour difference from gmt