meshtastic/firmware #4536: [Bug]: 2.5.0 Technical Preview Not Publishing Location

bob109, 2024-08-22:
### Category

Other

### Hardware

T-Echo

### Firmware Version

2.5.0.ab7de7f

### Description

Got a T-echo and I flashed 2.5.0.ab7de7f on.   Got a GPS lock and the t-echo knows the direction and distance to my other two nodes.  However neither of the other nodes know where the t-echo is... ie, it's not publishing its location over RF.  (I'm neither uploading nor downloading MQTT) 

I re-flashed back to 2.4.3.91d6612 and everything works as I expect it should.  All three nodes know where the other two are.

Szetya, 2024-08-22:
This is indeed the case. To me, the log doesn't say much.
Position request from another node:
DEBUG | 21:31:26 3523 [RadioIf] Lora RX (id=0xcf0b7d6f fr=0x68 to=0xad, WantAck=0, HopLim=3 Ch=0x8 encrypted rxSNR=6.5 rxRSSI=-50 hopStart=3)

DEBUG | 21:31:26 3523 [RadioIf] AirTime - Packet received : 387ms

DEBUG | 21:31:26 3523 [Router] Add packet record (id=0xcf0b7d6f fr=0x68 to=0xad, WantAck=0, HopLim=3 Ch=0x8 encrypted rxSNR=6.5 rxRSSI=-50 hopStart=3)

DEBUG | 21:31:26 3523 [Router] Using channel 0 (hash 0x8)

DEBUG | 21:31:26 3523 [Router] Expanding short PSK #1

DEBUG | 21:31:26 3523 [Router] Using AES128 key!

DEBUG | 21:31:26 3523 [Router] decoded message (id=0xcf0b7d6f fr=0x68 to=0xad, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 WANTRESP rxtime=1724362286 rxSNR=6.5 rxRSSI=-50 hopStart=3)

DEBUG | 21:31:26 3523 [Router] handleReceived(REMOTE) (id=0xcf0b7d6f fr=0x68 to=0xad, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 WANTRESP rxtime=1724362286 rxSNR=6.5 rxRSSI=-50 hopStart=3)

DEBUG | 21:31:26 3523 [Router] Module 'position' wantsPacket=1

INFO | 21:31:26 3523 [Router] Received position from=0xfa68a768, id=0xcf0b7d6f, portnum=3, payloadlen=0

DEBUG | 21:31:26 3523 [Router] POSITION node=fa68a768 l=0 lat=0 lon=0 msl=0 hae=0 geo=0 pdop=0 hdop=0 vdop=0 siv=0 fxq=0 fxt=0 pts=0 time=0

INFO | 21:31:26 3523 [Router] updatePosition REMOTE node=0xfa68a768 time=0 lat=0 lon=0

DEBUG | 21:31:26 3523 [Router] Node status update: 14 online, 20 total

DEBUG | 21:31:26 3523 [Router] Sending location with precision 32

INFO | 21:31:26 3523 [Router] Providing time to mesh 1724362286

INFO | 21:31:26 3523 [Router] Position reply: time=1724362286 lat=475503235 lon=191250680

DEBUG | 21:31:26 3523 [Router] Partially randomized packet id 2679706997

INFO | 21:31:26 3523 [Router] Asked module 'position' to send a response

DEBUG | 21:31:26 3523 [Router] Module 'neighborinfo' wantsPacket=1

INFO | 21:31:26 3523 [Router] Received neighborinfo from=0xfa68a768, id=0xcf0b7d6f, portnum=3, payloadlen=0

DEBUG | 21:31:26 3523 [Router] Module 'neighborinfo' considered

DEBUG | 21:31:26 3523 [Router] Module 'routing' wantsPacket=1

INFO | 21:31:26 3523 [Router] Received routing from=0xfa68a768, id=0xcf0b7d6f, portnum=3, payloadlen=0

DEBUG | 21:31:26 3523 [Router] Routing sniffing (id=0xcf0b7d6f fr=0x68 to=0xad, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 WANTRESP rxtime=1724362286 rxSNR=6.5 rxRSSI=-50 hopStart=3)

DEBUG | 21:31:26 3523 [Router] Delivering rx packet (id=0xcf0b7d6f fr=0x68 to=0xad, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 WANTRESP rxtime=1724362286 rxSNR=6.5 rxRSSI=-50 hopStart=3)

DEBUG | 21:31:26 3523 [Router] Update DB node 0xfa68a768, rx_time=1724362286

DEBUG | 21:31:26 3523 [Router] Forwarding to phone (id=0xcf0b7d6f fr=0x68 to=0xad, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 WANTRESP rxtime=1724362286 rxSNR=6.5 rxRSSI=-50 hopStart=3)

DEBUG | 21:31:26 3523 [Router] Module 'routing' considered

DEBUG | 21:31:26 3523 [Router] Sending response (id=0x9fb91575 fr=0xad to=0x68, WantAck=0, HopLim=2 Ch=0x0 Portnum=3 requestId=cf0b7d6f rxtime=1724362286 priority=70)

DEBUG | 21:31:26 3523 [Router] Update DB node 0x1a8686ad, rx_time=1724362286

DEBUG | 21:31:26 3523 [Router] localSend to channel 0

DEBUG | 21:31:26 3523 [Router] Add packet record (id=0x9fb91575 fr=0xad to=0x68, WantAck=0, HopLim=2 Ch=0x0 Portnum=3 requestId=cf0b7d6f rxtime=1724362286 priority=70)

DEBUG | 21:31:26 3523 [Router] Expanding short PSK #1

DEBUG | 21:31:26 3523 [Router] Using AES128 key!

[... 33 lines cut ...]
DEBUG | 21:35:05 3742 [Button] enqueuing for send (id=0xbdb06579 fr=0xad to=0xff, WantAck=0, HopLim=7 Ch=0x8 encrypted rxtime=1724362504 hopStart=7 priority=70)

DEBUG | 21:35:05 3742 [Button] txGood=199,rxGood=390,rxBad=4

DEBUG | 21:35:05 3742 [Button] Using channel 0 (hash 0x8)

DEBUG | 21:35:05 3742 [Button] Expanding short PSK #1

DEBUG | 21:35:05 3742 [Button] Using AES128 key!

DEBUG | 21:35:05 3742 [Button] decoded message (id=0xbdb06579 fr=0xad to=0xff, WantAck=0, HopLim=7 Ch=0x0 Portnum=3 rxtime=1724362504 hopStart=7 priority=70)

WARN | 21:35:05 3742 [Button] ToPhone queue is full, dropping packet.

DEBUG | 21:35:05 3742 [Button] Screen: Sent ad-hoc position

DEBUG | 21:35:05 3742 [Button] determineMode(): refresh=SKIPPED, reason=FRAME_MATCHED_PREVIOUS, frameFlags=0x3

DEBUG | 21:35:06 3743 [Button] determineMode(): refresh=SKIPPED, reason=FRAME_MATCHED_PREVIOUS, frameFlags=0x3

DEBUG | 21:35:06 3743 [RadioIf] (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=49, time 600 ms

DEBUG | 21:35:06 3743 [RadioIf] Lora RX (id=0x64b0692f fr=0xc4 to=0xff, WantAck=0, HopLim=2 Ch=0x8 encrypted rxSNR=5.75 rxRSSI=-44 hopStart=4)

DEBUG | 21:35:06 3743 [RadioIf] AirTime - Packet received : 600ms

Szetya, 2024-08-22:
The other test tool was a Heltec Wireless tracker (also V2.5.0). It does not send the position either. 
I tried to set FIX position on both devices. But I still cannot see the position of my other device.
That's how much I could help with the testing. :)

Soupbones, 2024-08-23:
Having a similar issue with 2.5 when setting a fixed position in the Heltec V3. Fixed positio  will not save. Wiped the board and flashed on fresh, same issue.  Using Android to control the board.

jp-bennett, 2024-08-23:
Found it. With the introduction of "optional" tags on a bunch of protobuf items, we need to set has_x to true when sending them.

closed 2024-08-23 by thebentern

thebentern, 2024-08-23:
Also closed by #4542

PR #4540 (2024-08-23): Adds has_x bools to position packet.
fixes #4536