meshtastic/firmware #4541: [Bug]:

armooo, 2024-08-23:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

2.5.0.ab7de7f

### Description

Public keys are added to the nodedb on first sight, the node nodedb has a fixed size. We can use this to fully control the contents of another node's nodedb.

Target is the destination node you want to intercept.

Attacker steps:
1) Save a copy of all NodeInfos
2) Remove `p.from = 0; // We don't let phones assign nodenums to their sent messages` from MeshService.cpp. This allows the attacker to set the from field.
3) Send NodeInfo packets with want_response set to the target until it replays. It will now not send a NodeInfo for the next 5 minutes.
4) Send 100+ NodeInfo packets with unique from ids a with public_keys. The goal is to replace all boring nodes, then  all the oldest nodes until the nodedb only contains attacker controlled entries.
5) Copy the target's NodeInfo saved from step 1 and replace the public key with your own. Send the NodeInfo with from set to the target.

I think this can be extended to a full MITM if you can synchronize step 3 across both nodes. There will be a number of Routing_Error_PKI_UNKNOWN_PUBKEY and extra NodeInfo sent between the two nodes in the MITM case.

The AdminModule seems safe because the public key is pined in config.security.admin_key.bytes.

GUVWAF, 2024-08-23:
This assumes you can send decryptable NodeInfo packets in the first place, which requires knowing the key of the channel. Sharing that key only with people you trust is the only way you can ensure to get authentic NodeInfos.

However, eventually this boils down to a Denial-of-Service attack by sending a lot of NodeInfo packets, and you need to keep repeating this because nodes will eventually respond to a new NodeInfo request with correct public key. This would be similar to jamming, for which there is just no solution.

closed 2024-08-23 by thebentern

armooo, 2024-08-23:
>This assumes you can send decryptable Nodelnfo packets in the first place. 

Yes, this is true but most traffic is sent with the default key. 

>  Nodes will eventually respond to a new Nodelnfo request with correct public key.

This is not the case unless the the node is remove from the nodedb. The keys are not updated in the nodedb as long as there is an existing one.

GUVWAF, 2024-08-23:
> The keys are not updated in the nodedb as long as there is an existing one.

I see. I think it would make sense to clear a remote node's public key in the DB when you were unable to decrypt a DM towards you, and then ask for a new NodeInfo also. @jp-bennett do you agree?

armooo, 2024-08-23:
That makes it even simpler to force a NodeInfo out of the node db. You no longer need to send 100+ messages. You can send a single forged message that fails to decrypt and targets a single node. You then need to win the race to send a forged NodeInfo with your private key before the target node.

GUVWAF, 2024-08-23:
Yes, that wasn't well-thought from my side.

I think the risk of a "race" for the NodeDB entry is acceptable if you decided to use a public channel. There's no way to authenticate a user over a public channel without any other means. You can also still spoof channel messages, like position.

jp-bennett, 2024-08-23:
There are two solutions here. The simplest is to favorite nodes you care about. This ensures they are never cycled out of NodeDB. The other safeguard is that apps like meshtastic-android will retain all the seen nodes, and already is designed to warn the user if the public key for a remote node changes.