meshtastic/firmware #4543: [Bug]: not direct nodes showen as direct (signal bar)

OE5DRO, 2024-08-23:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

2.5.0

### Description

i am not shure if this is a firmware bug or a bug in all software versions:
when i open any MT software (Mac, iOS, Android) and go to the nodes list, i see lots of nodes with sinal bar and some with hopps. the ones with the signal bar should be direct received nodes, but i am shure they are not. i have a home node with outdoor antenna on the roof and a 2nd node with very small antenna. the small one in ddor thould only see the one on the roof, but shows also direct received nodes from very far away with big signals. the repeated ones from my main nood should be shown as 1 hop? or am i wrog? this phenomenon is always on all 2.4.x versions. please give me a clue, what i can do to.
[image]

GUVWAF, 2024-08-23:
The "hops away" count was only added in firmware 2.3. If this node far away has older firmware, or it routes through a node with older firmware, it won't show the hops.

OE5DRO, 2024-08-23:
ok this means, if the far away node it self or one between has an old SW i got it shown as direct, with wrong signal.

GUVWAF, 2024-08-23:
Indeed. The signal you're seeing then is from the last relayer towards you.

closed 2024-08-23 by GUVWAF

OE5DRO, 2024-08-23:
> Indeed. The signal you're seeing then is from the last relayer towards you.

ok i will check out FW versions from nodes around me, thank you!