meshtastic/firmware #4557: [Bug]: Dockerfile failure

guitarpicva, 2024-08-26:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

current

### Description

Linux Ubuntu 22.04 LTS host system, docker build per: https://meshtastic.org/docs/software/linux-native/  Using the Dockerfile from the "firmware" distro since none was specified in the referenced Docker build instructions.

docker build -t meshtastic/device .

Then produces a failure on step 8/9 as below

```
 => ERROR [builder 8/9] RUN bash -o pipefail -c "source ./bin/activate &&  0.3s 
------                                                                          
 > [builder 8/9] RUN bash -o pipefail -c "source ./bin/activate && bash ./bin/build-native.sh":                                                                 
0.262 bash: ./bin/build-native.sh: No such file or directory                    
------

 1 warning found (use docker --debug to expand):
 - LegacyKeyValueFormat: "ENV key=value" should be used instead of legacy "ENV key value" format (line 8)
Dockerfile:28
--------------------
  26 |     # trunk-ignore(terrascan/AC_DOCKER_00024): We would actually like these files to be owned by mesh tyvm
  27 |     COPY --chown=mesh:mesh . /tmp/firmware
  28 | >>> RUN bash -o pipefail -c "source ./bin/activate && bash ./bin/build-native.sh"
  29 |     RUN cp "/tmp/firmware/release/meshtasticd_linux_$(uname -m)" "/tmp/firmware/release/meshtasticd"
  30 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c bash -o pipefail -c \"source ./bin/activate && bash ./bin/build-native.sh\"" did not complete successfully: exit code: 127
```
If I comment out the steps after line 27 and tun the container, there is not file bin/build-native.sh as noted by the error.

Tawnos, 2024-09-01:
I can't repro:

```
[+] Building 201.0s (18/18) FINISHED                                                                     docker:default
 => [internal] load build definition from Dockerfile                                                               0.0s
 => => transferring dockerfile: 2.20kB                                                                             0.0s
 => [internal] load metadata for docker.io/library/debian:bookworm-slim                                            1.6s
 => [internal] load .dockerignore                                                                                  0.0s
 => => transferring context: 2B                                                                                    0.0s
 => [internal] load build context                                                                                  3.8s
 => => transferring context: 88.04MB                                                                               3.8s
 => [builder 1/9] FROM docker.io/library/debian:bookworm-slim@sha256:2ccc7e39b0a6f504d252f807da1fc4b5bcd838e83e4  23.2s
 => => resolve docker.io/library/debian:bookworm-slim@sha256:2ccc7e39b0a6f504d252f807da1fc4b5bcd838e83e4dec3e2f57  0.0s
 => => sha256:2ccc7e39b0a6f504d252f807da1fc4b5bcd838e83e4dec3e2f57b2a4a64e7214 1.85kB / 1.85kB                     0.0s
 => => sha256:70d4c04302bdcd71c4fa21b6c12e99888380a07f04e3d44452b961bca046489d 529B / 529B                         0.0s
 => => sha256:4735ff618e63fa90d91e11a6cd0ea73bcfe3859924de33076eaff2630a1f0232 1.46kB / 1.46kB                     0.0s
 => => sha256:e4fff0779e6ddd22366469f08626c3ab1884b5cbe1719b26da238c95f247b305 29.13MB / 29.13MB                  22.4s
 => => extracting sha256:e4fff0779e6ddd22366469f08626c3ab1884b5cbe1719b26da238c95f247b305                          0.7s
 => [builder 2/9] RUN apt-get update && apt-get install --no-install-recommends -y wget python3 python3-pip pyth  61.5s
 => [stage-1 2/5] RUN apt-get update && apt-get --no-install-recommends -y install libc-bin libc6 libgpiod2 liby  12.9s
 => [stage-1 3/5] RUN groupadd -g 1000 mesh && useradd -ml -u 1000 -g 1000 mesh                                    0.4s
 => [stage-1 4/5] WORKDIR /home/mesh                                                                               0.0s
 => [builder 3/9] RUN groupadd -g 1000 mesh && useradd -ml -u 1000 -g 1000 mesh && chown mesh:mesh /tmp/firmware   0.3s
 => [builder 4/9] WORKDIR /tmp/firmware                                                                            0.0s
 => [builder 5/9] RUN python3 -m venv /tmp/firmware                                                                2.2s
 => [builder 6/9] RUN source ./bin/activate && pip3 install --no-cache-dir -U platformio==6.1.14                   4.1s
 => [builder 7/9] COPY --chown=mesh:mesh . /tmp/firmware                                                           0.2s
 => [builder 8/9] RUN source ./bin/activate && chmod +x /tmp/firmware/bin/build-native.sh && ./bin/build-native  107.0s
 => [builder 9/9] RUN cp "/tmp/firmware/release/meshtasticd_linux_$(uname -m)" "/tmp/firmware/release/meshtasticd  0.4s
 => [stage-1 5/5] COPY --from=builder /tmp/firmware/release/meshtasticd /home/mesh/                                0.1s
 => exporting to image                                                                                             0.1s
 => => exporting layers                                                                                            0.1s
 => => writing image sha256:da871d11c1f1e3900f1c05f842b07b37d44b2f58779334aecb643ac42d50495c                       0.0s
 => => naming to docker.io/meshtastic/device   

lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.3 LTS
Release:        22.04
Codename:       jammy                                                  
```

guitarpicva, 2024-09-02:
mitch@ondev:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.4 LTS
Release:	22.04
Codename:	jammy
mitch@ondev:~$ uname -a
Linux ondev 6.8.0-40-generic #40~22.04.3-Ubuntu SMP PREEMPT_DYNAMIC Tue Jul 30 17:30:19 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

Tawnos, 2024-09-02:
Which folder are you in when building? I start in the root folder and even after updating to Ubuntu 22.04.04 and docker 27.2.0, there's no issue. Try a clean clone?

guitarpicva, 2024-09-02:
you mean the root of the cloned distro?  i'll give that a go.  thanks.  Definitely NOT doing mine that way.

guitarpicva, 2024-09-02:
Things seem to be progressing much more normally now, so it seems to be my error then.  Thank you for your patience with the 'operator head space'.  I do appreciate it.

guitarpicva, 2024-09-02:
```
 1 warning found (use docker --debug to expand):
 - LegacyKeyValueFormat: "ENV key=value" should be used instead of legacy "ENV key value" format (line 8)
```
That warning is the only "issue" I saw, and not a show-stopper

guitarpicva, 2024-09-02:
Running now, thanks.

closed 2024-09-02 by guitarpicva