meshtastic/firmware #4563: [Feature Request]: Add the absolute amount of received Meshtastic packets to telemetry

dm5tt, 2024-08-26:
### Platform

NRF52, ESP32, RP2040, Linux Native, other

### Description

Problem: It's at the moment it's impossible to determine the utilization of a Router. Without these data we cannot determine the health of the HF chain nor if the setup/position of the entire device can be improved.

Right now following data can be gathered using the Python script:

```
~$ /home/had/.local/bin/meshtastic --host 192.x.x.x --request-telemetry --dest \!279xxxx
Connected to radio
Sending telemetry request to !279xxxx on channelIndex:0 (this could take a while)
Telemetry received:
Battery level: 81.00%
Voltage: 4.00 V
Total channel utilization: 42.40%
Transmit air utilization: 13.42%
Uptime: 109625 s
```

It would be great to have an additional absolute counter of received, valid Meshtastic packets.

Example:

`Received Packets: 20000`

garthvh, 2024-08-26:
This is now in place available over the phone api for the node you are connected to.

https://github.com/meshtastic/firmware/pull/4475

dm5tt, 2024-08-26:
Ah..  wasn't aware of that.

My use-case is a bit different as mine station is remote and not directly accessible with BT/Wifi.

thebentern, 2024-08-26:
We could possibly make this one available over the wire as well. We need to break it out into its own module instead of leaning on Device Telemetry module as it exists now.

PR #5414 (2024-11-21): Telemetry can respond to want-response for LocalStats variant
Closes #4563
closed 2024-11-21 by thebentern