meshtastic/firmware #4564: [Feature Request]: Admin message to request a refresh of the nodedb

garthvh, 2024-08-27:
### Platform

NRF52, ESP32, RP2040, Linux Native

### Description

Now that we have this in place https://github.com/meshtastic/firmware/issues/3740

We need to add a admin message that will refresh the node db, so that apps can separate connecting and the nodedb refresh, which takes a long time with 100 nodes.

gjelsoe, 2024-11-20:
@garthvh Is this a `NodeDB::cleanupMeshDB()` call you had in mind ?

ianmcorvidae, 2024-11-20:
I believe this issue is for adding an admin message that sets off the connection/want_config stage of sending all the nodes in NodeDB up to the phone via the API. That way the clients can connect with the nodeless wantconfig (and thus not require anyone to wait for all 80/100/possibly more to come through on initial connect), and then send the admin message after connection to get any updated node information to update their local datastores with. So basically an admin message to set off the stuff within the `case STATE_SEND_OTHER_NODEINFOS` block in `PhoneAPI::available()` I think

garthvh, 2025-04-24:
@thebentern we really need this to improve the connection process.

garthvh, 2025-05-11:
> I was under the impression something similar was already existing since the MUI can do a DB resync. I'll look into the code more closely today, also to avoid double code while adding this.

I don't see anything related to the node only admin message in the 2.6 changes https://github.com/meshtastic/protobufs/pull/645/files

PR #6792 (2025-05-11): Only send nodes on want_config of 69421
Closes #4564

@garthvh you should be able to just issue a special want_config of 69421 and receive just the nodeinfo
closed 2025-05-11 by thebentern