meshtastic/firmware #4570: [Bug]: GPS not recognized on RAK4631, w/ver. 4.5.0.

magpie54, 2024-08-28:
### Category

Other

### Hardware

Rak4631

### Firmware Version

4.5.0

### Description

After updating firmware to 4.5.0 noticed that the RAK12500 was no longer recognized. Rolled firmware back to previous version and it works just fine.

garthvh, 2024-08-28:
There is no firmware 4.5.0

magpie54, 2024-08-28:
Sorry, 2.5.0.

caveman99, 2024-08-28:
can you post a serial log snippet from the first few seconds after power up?

magpie54, 2024-08-28:
I rolled back to the previous version. Give me a day or so and I’ll do it
again. I’m not totally sure how to get the serial log snippet. Thanks.

magpie54, 2024-08-30:
Tried again and the GPS seems to be working fine. The difference...
2.5.0.ab7de7f - did not work
2.5.0.d6dacty - seems to work just fine.

fifieldt, 2024-08-31:
Hi @magpie54 , thanks for the report.

To get a serial log go to https://flasher.meshtastic.org/ and click "Open Serial Monitor".

magpie54, 2024-08-31:
Tom, thanks for the info on the serial log.
Dave

fifieldt, 2024-09-23:
Couldn't find any obvious changes for GPS between those versions. However, since a firmware upgrade fixed this let's close it for now. I'll keep an eye on detection problems as while working on #4625 .

closed 2024-09-23 by fifieldt

magpie54, 2025-12-09:
> Dave - Positions work, but are a bit weird? Maybe check the precision
> settings in the channel...

Tom, thanks for responding so quickly. I’m a bit embarrassed to tell you
what I found. The module wasn’t seated well. Once I got it properly seated
it began to work just fine.

Thanks again for your response. All the best to you and your family and
Happy Holidays.

Dave