meshtastic/firmware #4579: [Bug]: Heltec (or ESP32-S3 in general?) tracker user button doesn't work while in light sleep

geeksville, 2024-08-28:
### Category

Hardware Compatibility

### Hardware

Heltec Wireless Tracker

### Firmware Version

2.5.? (a private build from today's master)

### Description

While working on #4154 I noticed that the heltec tracker board is not automatically waking  from light sleep when the user presses a button.

For instance (if board is configured as a router so that it sleeps and turns the screen off often), the last log messages seen before light sleep are:
```
DEBUG | ??:??:?? 7 [Screen] haveGlyphs=1
DEBUG | ??:??:?? 8 [Screen] haveGlyphs=1
INFO  | ??:??:?? 8 [PowerFSM] Turning off screen
INFO  | ??:??:?? 9 [PowerFSM] lsEnter begin, ls_secs=86400
INFO  | ??:??:?? 10 [PowerFSM] setup LORA_DIO1 (GPIO14) with wakeup by external source
```

The radio **is** able to wake the board (good), but also the user button on the board should also be sufficient for wake.  The wake does not occur if you press the button.  I suspect that something is wrong on the esp32-s3 gpio config for that button.  It should be configured to generate an interrupt to wake the CPU from sleep (so we can then run our regular button processing)

Alas, I don't think I'll have time to investigate this problem before I leave town.  So recording it so that it is known and perhaps someone else can dig into it as needed.

todd-herbert, 2024-08-29:
Just as an initial report: I checked Heltec Wireless Paper with 2.5.0.d6dac17 from the web flasher, and the user button is waking the from light-sleep as expected, so this might be version specific or device specific.

HarukiToreda, 2024-10-07:
I am seeing this on my nodes with cardkbs while in lite sleep.

todd-herbert, 2024-10-07:
@HarukiToreda With Heltec V3, right?

thebentern, 2024-10-07:
Seems like they're coming out of Deep Sleep just fine with the user button interrupt. Been using this on my V3 and Tracker. I wonder if we just need to A/B compare those.

geeksville, 2024-10-10:
(btw: it is taking a bit longer to setup our new rental apartment than
expected, so I won't be back to recreational github meshtastic futzing
until about 10/17 :-( )

closed 2024-12-16 by github-actions[bot]