meshtastic/firmware #4590: [Feature Request]: Routers behave as clients when they see another router, so clients will not see both and retransmit.

quimnut, 2024-08-30:
### Platform

other

### Description

My understanding is that a client will not flood route if it sees a retransmit.

In the scenario that an actor operates 2 routers, a client that hears both will not flood route.

This means a router will break any client mutes behind a client and as such breaks a mesh, it would require router mode for any intermediate hop like a roof top.

I feel an option would be a router should act as a client if it sees another router in hop 0.

Happy to be wrong, thanks for all the love. Meshtastic is a great community.

quimnut, 2024-08-30:
Thank you @GUVWAF that's a very clear summary.

quimnut, 2024-08-30:
Can we expand on why this is an edge case?

GUVWAF, 2024-08-30:
Maybe not really an edge case, but something that can happen in a specific scenario like this. In a slightly different scenario, where there would be another node that can hear 0 but not 1, it would also rebroadcast and 3 could receive that.

There's a trade-off between reliability and redundancy here. If every node would rebroadcast, this scenario would still work, as long as it doesn't create collisions.

quimnut, 2024-08-30:
How specific a case is someone running 2 nodes in router?

GUVWAF, 2024-08-30:
But how would that create this issue in scenarios where it would otherwise still work? 
Anyone rebroadcasting earlier than you causes you to not rebroadcast. So, it might get worse if someone is setting its node as router when it should be, but I don't see how 2 routers would make it happen in more scenarios.

fifieldt, 2024-08-31:
Seems related: https://en.m.wikipedia.org/wiki/Hidden_node_problem

GUVWAF, 2024-08-31:
The hidden node problem is indeed what causes collisions. This is more similar to the ["Exposed node problem"](https://en.m.wikipedia.org/wiki/Exposed_node_problem), where you prevent from sending when someone else did, while *your* neighbor didn't receive the other's transmission.

closed 2024-08-31 by thebentern