meshtastic/firmware #4603: [Bug]: name issue in preview

VE7OES, 2024-09-01:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.5.0.9

### Description

When a t-beam is upgraded to the preview release, the long and short name are fine. If there's a full reset done to the t-beam, attempting to change the long and/or short name doesn't take. The t-beam reboots with the default names. Every-time.

thebentern, 2024-09-01:
If you're doing this via one of the apps, are you able to grab some serial logs via `meshtastic --noproto` or the serial monitor https://flasher.meshstastic.org?

jp-bennett, 2024-09-01:
Which app are you using to try to set the names? And what version of that app, if you can find it?