meshtastic/firmware #4604: [Feature Request]: Power Saving Feature - Reduce LoRa TX power based on dusk/dawn-times or USB==1/0

dm5tt, 2024-09-01:
### Platform

NRF52, ESP32, RP2040, Linux Native, other

### Description

Solar charge controllers (TI BQ25185/etc.) support a Power Path'ing which means that the device can be powered directly from the solar panel as soon as the battery is fully charged It also mixes (70% solar, 30% battery)  the power  during dusk or cloudy phases.

Example, I marked the spots where the device switches to direct pass through:

[image]

Feature Ideas:

1. It would be great to have a switchable feature which reduces the TX power to a defined level after it cannot powered directly from solar any more (USB==0). This would greatly extend the battery lifetime during bad weather or even the upcoming winter.  The entire switching process should be rather slow so we don't end up in flapping around.

2. As we  introduced a  system time lately we could even make this with calculated dusk/dawn times based on GPS coordinates. This might make it even more robust and would supersede feature request 1.

b8b8, 2024-09-02:
What would be the situation where you would want lower transmit power but not say just turning the transmitter off completely when a certain battery threshold is met? Does the intermittent transmission or relay of messages significantly effect the battery life compared to the CPU and other peripherals running? I can see turning the transmission to off at say 10%, but 50% battery could still be 2 weeks of power on an NRF device. I guess my question is what is the use of a lower power transmit unless you are using one of the multi-watt station G2 edition devices.

dm5tt, 2024-09-02:
> What would be the situation where you would want lower transmit power but not say just turning the transmitter off completely when a certain battery threshold is met? 

According to my voltage recordings: Full power during sunrise until sunset as the device most likely can be directly powered from the solar panel. Reduced during the nighttime.

The LoRa transmitter of your NRF will also eat quite a bit of energy. Peak current according to Semtec (+20dB + PA-Boost) you can plan for ~120mA peak. With only +12dBm it's only ~30mA.

b8b8, 2024-09-02:
ok, then i ask is it good to have a node that may only be reachable at some times of the day knowing that further away nodes wouldnt be able to communicate with it at night? This is an interesting idea, i just trying to see the point of lowering power at all if you care not near a completely depleted battery and the next day it should charge up a bit.

dm5tt, 2024-09-02:
During the summer it's most likely no problem to keep it running on full TX power the entire time as you going to reach 100% SoC every mid day. 

But during the winter and autumn it can give you a lot of margin to keep your node running when there's bad weather around.

b8b8, 2024-09-02:
would be interesting to run an experiment to see 2 identical nodes, one set for 10dbm transmit and one set for 22dbm and measure how much of a difference there is over a week.

dm5tt, 2024-09-02:
I will start that experiment! Have to build two stations anyway and then will have to "burn them in" for 1-2 weeks. Going to set one to maximum power and one to +12dBm.