meshtastic/firmware #4605: [Feature Request]: Expose an I2C interface for Heltec Mesh Node T114

todd-herbert, 2024-09-01:
### Platform

NRF52

### Description

(Opened on behalf of LumpusWorax on the Meshtastic Discord server. I don't myself have this device)

The one I2C interface currently used by Heltec Mesh Node T114 is connect directly to the PCF8563TS RTC (?)
The pins (`P0.26` and `P0.27`) are not exposed via the headers or the jumper connectors.
If I understand it correctly, the NRF52840 datasheet suggests it should be possible to define a second I2C bus.

Could we create a second I2C bus, mapped to the device's header pins? Are there any technical reasons why this wouldn't be possible or would be unwise?

lyusupov, 2024-09-01:
> The one I2C interface currently used by Heltec Mesh Node T114 is connect directly to the PCF8563TS RTC (?)

[image]

[image]

todd-herbert, 2024-09-01:
Ah I had a funny feeling I'd heard that the RTC was actually populated on the boards that were being shipped. If it's not, it might be an option to just remap the pins for the first I2C bus? Although there could be an argument for leaving that first I2C bus where Heltec has defined it too, in case that RTC does get included in future revisions?

lyusupov, 2024-09-01:
Heltec have borrowed mapping for most of nRF52840 GPIO pins from LilyGO T-Echo to ease porting of the firmware.
RTC and FLASH are using the same GPIOs as T-Echo as well.
Heltec just decided to save some money on the components that they think are not in use.

thebentern, 2024-09-01:
> I don't myself have this device

Can you request a couple in the budget room? Probably any of the other Heltec Vision Master hardware missing in your inventory as well.

lyusupov, 2024-09-12:
> Ah I had a funny feeling I'd heard that the RTC was actually populated on the boards that were being shipped.

The board delivered to me yesterday has neither RTC nor SPI FLASH.

[image]

HarukiToreda, 2024-09-14:
> ed to me yesterday has neither RTC nor SPI FLASH.

According to Heltec, these are installed upon request when you place the order.

todd-herbert, 2024-09-14:
I'm definitely going to attempt to set up a second I2C bus next week when my T114 arrives. If it turns out to be straightforward, it could be good future proofing. Otherwise, one user reported already that remapping the first bus works fine.

lyusupov, 2024-09-14:
> According to Heltec, these are installed upon request when you place the order.

Nether Heltec own Store nor AliExpress source gave an **option for RTC and/or FLASH**:

**Heltec online Store**

[image]

<br>
<br>

**Heltec Store on AliEXpress**

[image]

todd-herbert, 2024-09-14:
I realize that there is currently no option available when purchasing through the store. @HarukiToreda has spoken directly with Heltec staff though. If I understood correctly, the components can be populated at customer's request. It just strikes me as good future proofing to explore leaving I2C pins routed to that pad, so long as it doesn't end up causing a hassle. I haven't looked into the option yet but will when my hardware arrives this week.

HarukiToreda, 2024-09-14:
future proofing is certainly a good idea, currently there is no use for flash on an nrf board, but having the ability to request it once it is needed will make developing new features possible. I agree with @todd-herbert

lyusupov, 2024-09-14:
>  If I understood correctly, the components can be populated at customer's request.

It is not a very smart behavior to interpret words "... can be populated at customer's request ..."  to a certain product's batch of few hundred units been already manufactured and packaged for retail sales purpose. Nobody is ready to send them back to factory in order to solder few extra components on the double sided PCB.

Words  "... can be populated at customer's request ..."  have a chance to be applied to a new batch of this product to be manufactured in the future, provided that 
- the customer is ready to pay for the full batch or
- the customer can guarantee that this specific batch has a potential to be sold out by Heltec within a reasonable time frame.

HarukiToreda, 2024-09-14:
> It is not a very smart behavior to interpret words "... can be populated at customer's request ..." to a certain product's batch of few hundred units been already manufactured and packaged for [...]

Heltec is notorious for making various iterations to their devices when they first come out, this has been seen on all previous boards they've released, from personal experience i can say with certainty that preparing for possible additions listed on their schematic diagram is the way to go. Just because your current boards do not have it does not mean they won't in future ones. 

Not doing so will cause various version of boards in the meshtastic firmware that can be avoided, see as an example Heltec V3/v2, tracker v1.0/1.1. Vision master E290 V1.0

If they don't add the chips, nothing is lost, but if they do add them, users may already be using other devices attached that may break on future firmware updates.

todd-herbert, 2024-09-14:
> If they don't add the chips, nothing is lost, but if they do add them, users may already be using other devices attached that may break on future firmware updates.

That was all I was thinking too, although to be fair if we did have to reconfigure for a scenario like this at some point, it probably wouldn't be a breaking change that splits a hardware variant.
At least, hopefully.

Mark Birss did suggest too that if a board revision does come out, there could be other changes which would force a variant. 

Just thinking out loud!

lupusworax, 2024-09-16:
Thanks @ todd Herbert for the test  Firmware file. I2C seems to work accordingly
[image]
.

todd-herbert, 2024-09-16:
Oh yeah, I built that a week or two ago, right? Was definitely just a quick test at that stage. Good to see there weren't any other issues lurking though.

closed 2024-09-23 by caveman99

todd-herbert, 2024-09-23:
@lupusworax just be aware that the pin mapping has [changed slightly](https://github.com/meshtastic/firmware/pull/4745#issuecomment-2358398414) from that test build
* SCL P0.13
* SDA P0.16
___
I'll make sure to follow this one up with some docs in the near future.

lupusworax, 2024-09-29:
> @lupusworax just be aware that the pin mapping has [changed slightly](https://github.com/meshtastic/firmware/pull/4745#issuecomment-2358398414) from that test build
> 
> * SCL P0.13
> * SDA P0.16
> 
> ~I'll make sure to follow this one up with some docs in the near future.~ Wait until we know the future of the device?

Got it Thanks!