meshtastic/firmware #4615: [Bug]: T1000-E Boot Loop

n5qm, 2024-09-02:
### Category

Other

### Hardware

Other

### Firmware Version

2.4.3.91d6612

### Description

I attempted to upgrade to this version of the firmware which seems to have been successful however the device is now in a boot loop.

I have attempted to utilize the nrf_erase2.uf2 image to clear the storage on the device and that appears to fail as the device reboots after that and no longer presents a serial interface.

### Relevant log output

```Shell
This log output is constantly repeating.  The device emits tones via the buzzer, I receive this log output, then it appears to reboot, emits a tone via the buzzer, and repeats the same process over and over.

meshtastic --noproto
WARNING file:mesh_interface.py _sendToRadio line:880 Not sending packet because protocol use is disabled by noProto
Connected to radio
WARNING file:mesh_interface.py _sendPacket line:712 Not sending packet because protocol use is disabled by noProto
WARNING file:mesh_interface.py _sendToRadio line:880 Not sending packet because protocol use is disabled by noProto
INFO  | ??:??:?? 0

//\ E S H T /\ S T / C

DEBUG | ??:??:?? 0 Filesystem files:
DEBUG | ??:??:?? 1  meshtastic.txt (288 Bytes)
DEBUG | ??:??:?? 1  config.proto (104 Bytes)
DEBUG | ??:??:?? 1  channels.proto (57 Bytes)
DEBUG | ??:??:?? 1  db.proto (2618 Bytes)
DEBUG | ??:??:?? 1  module.proto (100 Bytes)
DEBUG | ??:??:?? 1 Using analog input 2 for battery level
INFO  | ??:??:?? 1 Scanning for i2c devices...
DEBUG | ??:??:?? 1 Scanning for I2C devices on port 1
INFO  | ??:??:?? 1 No I2C devices found
DEBUG | ??:??:?? 1 acc_info = 0
INFO  | ??:??:?? 1 S:B:71,2.4.3.91d6612
DEBUG | ??:??:?? 1 Reset reason: 0x0
DEBUG | ??:??:?? 1 Setting random seed 504867195
INFO  | ??:??:?? 1 Initializing NodeDB
INFO  | ??:??:?? 1 Loading /prefs/db.proto
INFO  | ??:??:?? 1 Loaded /prefs/db.proto successfully
INFO  | ??:??:?? 1 Loaded saved devicestate version 23, with nodecount: 25
INFO  | ??:??:?? 1 Loading /prefs/config.proto
INFO  | ??:??:?? 1 Loaded /prefs/config.proto successfully
INFO  | ??:??:?? 1 Loaded saved config version 23
INFO  | ??:??:?? 1 Loading /prefs/module.proto
INFO  | ??:??:?? 1 Loaded /prefs/module.proto successfully
INFO  | ??:??:?? 1 Loaded saved moduleConfig version 23
INFO  | ??:??:?? 1 Loading /prefs/channels.proto
INFO  | ??:??:?? 1 Loaded /prefs/channels.proto successfully
INFO  | ??:??:?? 1 Loaded saved channelFile version 23
ERROR | ??:??:?? 1 Could not open / read /oem/oem.proto
DEBUG | ??:??:?? 1 cleanupMeshDB purged 0 entries
DEBUG | ??:??:?? 1 Using nodenum 0xa39c27ab
DEBUG | ??:??:?? 1 Expanding short PSK #1
INFO  | ??:??:?? 1 Wanted region 1, using US
DEBUG | ??:??:?? 1 Using GPIO06 for button
DEBUG | ??:??:?? 1 Set Timezone to GMT0
INFO  | ??:??:?? 1 GPS power state moving from OFF to ACTIVE
DEBUG | ??:??:?? 1 NeighborInfoModule is disabled
DEBUG | ??:??:?? 1 Rescanning for I2C keyboard
DEBUG | ??:??:?? 1 Scanning for I2C devices on port 1
DEBUG | ??:??:?? 1 Scanning address 0x1f
DEBUG | ??:??:?? 1 Scanning address 0x55
DEBUG | ??:??:?? 1 Scanning address 0x5f
ERROR | ??:??:?? 1 Could not open / read /prefs/ringtone.proto
INFO  | ??:??:?? 1 Initializing External Notification Module
WARN  | ??:??:?? 1 LR11x0Interface(cs=12, irq=33, rst=42, busy=7)
DEBUG | ??:??:?? 1 LR11X0_DIO3_TCXO_VOLTAGE defined, using DIO3 as TCXO reference voltage at 1.600000 V
INFO  | ??:??:?? 1 Starting meshradio init...
DEBUG | ??:??:?? 1 (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=0, time 231 ms
DEBUG | ??:??:?? 1 (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=253, time 2115 ms
INFO  | ??:??:?? 1 Radio freq=906.875, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 1 Set radio: region=US, name=LongFast, config=0, ch=19, power=30
INFO  | ??:??:?? 1 Radio myRegion->freqStart -> myRegion->freqEnd: 902.000000 -> 928.000000 (26.000000 MHz)
INFO  | ??:??:?? 1 Radio myRegion->numChannels: 104 x 250.000kHz
INFO  | ??:??:?? 1 Radio channel_num: 20
INFO  | ??:??:?? 1 Radio frequency: 906.875000
INFO  | ??:??:?? 1 Slot time: 42 msec
INFO  | ??:??:?? 1 Set radio: final power level=22
DEBUG | ??:??:?? 1 Setting DIO RF switch
INFO  | ??:??:?? 2 LR11x0 init result 0
INFO  | ??:??:?? 2 Frequency set to 906.875000
INFO  | ??:??:?? 2 Bandwidth set to 250.000000
INFO  | ??:??:?? 2 Power output set to 22
INFO  | ??:??:?? 2 Set RX gain to boosted mode; result: 0
INFO  | ??:??:?? 2 LR1110 Radio init succeeded, using LR1110 radio
DEBUG | ??:??:?? 2 (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=237, time 2000 ms
DEBUG | ??:??:?? 2 LoRA bitrate = 118.500000 bytes / sec
INFO  | ??:??:?? 2 PowerFSM init, USB power=1
DEBUG | ??:??:?? 2 Enter state: BOOT
DEBUG | ??:??:?? 2 [Power] Battery: usbPower=1, isCharging=1, batMv=3100, batPct=0
DEBUG | ??:??:?? 2 [GPS] Probing for GPS at 115200
DEBUG | ??:??:?? 2 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | ??:??:?? 2 [GPS] No GPS lock
DEBUG | ??:??:?? 2 [GPS] onGPSChanged() pos@0 time=0 lat=0 lon=0 alt=0
INFO  | ??:??:?? 2 [GPS] updatePosition LOCAL pos@0 time=0 lat=0 lon=0 alt=0
DEBUG | ??:??:?? 2 [GPS] Setting local position: lat=0 lon=0 time=0 timestamp=0
DEBUG | ??:??:?? 2 [GPS] Node status update: 25 online, 25 total
INFO  | ??:??:?? 2 [RangeTestModule] Range Test Module - Disabled
DEBUG | ??:??:?? 5 [PowerFSM] Initializing NRF52 Bluetooth
WARNING file:stream_interface.py __reader line:203 Meshtastic serial port disconnected, disconnecting... read failed: [Errno 6] Device not configured
```

n5qm, 2024-09-02:
I was provided, indirectly, a new bootloader from Seeed to recover the device and that seems to have worked.  I flashed the new bootloader and then installed the latest 2.4.2 Beta and it seems to be functioning as expect.

adafruit-nrfutil --verbose dfu serial --package Downloads/t1000_e_bootloader-0.9.1-5-g488711a_s140_7.3.0.zip -b 115200 -p /dev/tty.usbmodem101
Upgrading target on /dev/tty.usbmodem101 with DFU package /Users/r/Downloads/t1000_e_bootloader-0.9.1-5-g488711a_s140_7.3.0.zip. Flow control is disabled, Dual bank, Touch disabled
Opened serial port /dev/tty.usbmodem101
Starting DFU upgrade of type 3, SoftDevice size: 152728, bootloader size: 39000, application size: 0
Sending DFU start packet
Sending DFU init packet
Sending firmware file
########################################
########################################
########################################
########################################
########################################
########################################
########################################
########################################
########################################
###############
Activating new firmware

DFU upgrade took 20.723652839660645s
Device programmed.

circuitmike, 2024-09-03:
This happened to me as well, but that fix didn't work for me. I found [the bootloader file](https://files.seeedstudio.com/wiki/SenseCAP/Meshtastic/t1000_e_bootloader-0.9.1-5-g488711a_s140_7.3.0.hex) on Seeed's site, installed adafruit-nrfutil, converted the .hex file to a .zip package, and attempted to flash it to the device. I'm on Linux and dmesg shows the serial device as /dev/ttyACM2 and drive as /dev/sdc. Results look like this (and yes, the device is in DFU mode and I do have write access to /dev/ttyACM2):

```$ adafruit-nrfutil --verbose dfu serial --package dfu-package.zip -p /dev/ttyACM2 -b 115200
Upgrading target on /dev/ttyACM2 with DFU package /home/mike/meshtastic/dfu-package.zip. Flow control is disabled, Dual bank, Touch disabled
Opened serial port /dev/ttyACM2
Starting DFU upgrade of type 4, SoftDevice size: 0, bootloader size: 0, application size: 1034328
Sending DFU start packet

Failed to upgrade target. Error is: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
Traceback (most recent call last):
  File "/home/mike/meshtastic/adafruit-nrfutil/.venv/lib/python3.10/site-packages/adafruit_nrfutil-0.5.3.post16-py3.10.egg/nordicsemi/__main__.py", line 296, in serial
    dfu.dfu_send_images()
  File "/home/mike/meshtastic/adafruit-nrfutil/.venv/lib/python3.10/site-packages/adafruit_nrfutil-0.5.3.post16-py3.10.egg/nordicsemi/dfu/dfu.py", line 235, in dfu_send_images
    self._dfu_send_image(HexType.APPLICATION, self.manifest.application)
  File "/home/mike/meshtastic/adafruit-nrfutil/.venv/lib/python3.10/site-packages/adafruit_nrfutil-0.5.3.post16-py3.10.egg/nordicsemi/dfu/dfu.py", line 199, in _dfu_send_image
    self.dfu_transport.send_start_dfu(program_mode, softdevice_size, bootloader_size,
  File "/home/mike/meshtastic/adafruit-nrfutil/.venv/lib/python3.10/site-packages/adafruit_nrfutil-0.5.3.post16-py3.10.egg/nordicsemi/dfu/dfu_transport_serial.py", line 179, in send_start_dfu
    self.send_packet(packet)
  File "/home/mike/meshtastic/adafruit-nrfutil/.venv/lib/python3.10/site-packages/adafruit_nrfutil-0.5.3.post16-py3.10.egg/nordicsemi/dfu/dfu_transport_serial.py", line 243, in send_packet
    ack = self.get_ack_nr()
  File "/home/mike/meshtastic/adafruit-nrfutil/.venv/lib/python3.10/site-packages/adafruit_nrfutil-0.5.3.post16-py3.10.egg/nordicsemi/dfu/dfu_transport_serial.py", line 264, in get_ack_nr
    temp = [x for x in self.serial_port.read(6)]
  File "/home/mike/meshtastic/adafruit-nrfutil/.venv/lib/python3.10/site-packages/serial/serialposix.py", line 595, in read
    raise SerialException(
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
```

Since that failed I decided to take a chance on flashing the bootloader found [here](https://wiki.seeedstudio.com/flash_meshtastic_kit/) onto the device using USB storage mode (flashing the UF2 file, of course). That also fails - the device just keeps booting into DFU mode, which is the same thing it was doing before.

This device looks pretty thoroughly bricked. Any suggestions would be greatly appreciated.

n5qm, 2024-09-03:
@circuitmike the bootloader I used and the instructions that worked for me are located here.  I put the device in DFU mode and then just followed the directions.  After the bootloader flashed, it rebooted back into DFU mode and I was able to copy the firmware over and it rebooted again and all was well.

https://wiki.seeedstudio.com/sensecap_t1000_e/#flash-the-bootloader

lyusupov, 2024-09-03:
@n5qm 
Does it mean that official [**Release 0.9.2**](https://github.com/adafruit/Adafruit_nRF52_Bootloader/releases/tag/0.9.2) of the [**Adafruit nRF52 Bootloade**r](https://github.com/adafruit/Adafruit_nRF52_Bootloader) is not good enough?

markbirss, 2024-09-03:
@lyusupov

yes, it is the latest available, i suggest add/use these extra command line options "--singlebank --touch 1200"

circuitmike, 2024-09-05:
The new bootloader and instructions from the Seeed wiki fixed it for me!

closed 2024-09-11 by n5qm