meshtastic/firmware #4625: [Bug]: T-Deck Plus very inconsistent GPS module detection

MisterC925, 2024-09-04:
### Category

Hardware Compatibility

### Hardware

T-Deck

### Firmware Version

2.4.2.5b45303 Beta

### Description

The T-Deck Plus is the T-Deck in a case with a battery and soldered on GPS module (same pins as the port on the original T-Deck.

Often the unit will fail to detect the GPS module (not fail to get a fix but report there is no module "No GPS Present")

I've tried it with and without the phone reporting location (just in case that was overriding something) 

I noticed in the log it looks like it's detected but then is disabled.  I've attached the relevant log entries (that contain "GPS" let me know if you need anything else)

### Relevant log output

```Shell
INFO | ??:??:?? 7 GPS power state moving from OFF to ACTIVE
DEBUG | ??:??:?? 7 Using GPIO44 for GPS RX
DEBUG | ??:??:?? 7 Using GPIO43 for GPS TX
DEBUG | ??:??:?? 8 [GPS] Probing for GPS at 9600
WARN | ??:??:?? 14 [GPS] Failed to find UBlox & MTK GNSS Module using baudrate 9600
DEBUG | ??:??:?? 16 [GPS] Probing for GPS at 4800
DEBUG | ??:??:?? 16 [GPS] Setting Baud to 4800
INFO | ??:??:?? 18 [GPS] To Radio onwrite
INFO | ??:??:?? 18 [GPS] Client wants config, nonce=720
DEBUG | ??:??:?? 18 [GPS] Got 4 files in manifest
INFO | ??:??:?? 18 [GPS] Starting API client config
INFO | ??:??:?? 18 [GPS] getFromRadio=STATE_SEND_MY_INFO
INFO | ??:??:?? 18 [GPS] getFromRadio=STATE_SEND_OWN_NODEINFO
INFO | ??:??:?? 18 [GPS] getFromRadio=STATE_SEND_METADATA
INFO | ??:??:?? 18 [GPS] getFromRadio=STATE_SEND_CHANNELS
INFO | ??:??:?? 19 [GPS] getFromRadio=STATE_SEND_CHANNELS
INFO | ??:??:?? 19 [GPS] getFromRadio=STATE_SEND_CHANNELS
INFO | ??:??:?? 19 [GPS] getFromRadio=STATE_SEND_CHANNELS
INFO | ??:??:?? 19 [GPS] getFromRadio=STATE_SEND_CHANNELS
INFO | ??:??:?? 19 [GPS] getFromRadio=STATE_SEND_CHANNELS
INFO | ??:??:?? 19 [GPS] getFromRadio=STATE_SEND_CHANNELS
INFO | ??:??:?? 19 [GPS] getFromRadio=STATE_SEND_CHANNELS
INFO | ??:??:?? 19 [GPS] getFromRadio=STATE_SEND_CONFIG
INFO | ??:??:?? 19 [GPS] getFromRadio=STATE_SEND_CONFIG
INFO | ??:??:?? 19 [GPS] getFromRadio=STATE_SEND_CONFIG
INFO | ??:??:?? 20 [GPS] getFromRadio=STATE_SEND_CONFIG
INFO | ??:??:?? 20 [GPS] getFromRadio=STATE_SEND_CONFIG
INFO | ??:??:?? 20 [GPS] getFromRadio=STATE_SEND_CONFIG
INFO | ??:??:?? 20 [GPS] getFromRadio=STATE_SEND_CONFIG
INFO | ??:??:?? 20 [GPS] getFromRadio=STATE_SEND_CONFIG
INFO | ??:??:?? 20 [GPS] getFromRadio=STATE_SEND_CONFIG
INFO | ??:??:?? 20 [GPS] getFromRadio=STATE_SEND_MODULECONFIG
INFO | ??:??:?? 20 [GPS] getFromRadio=STATE_SEND_MODULECONFIG
INFO | ??:??:?? 20 [GPS] getFromRadio=STATE_SEND_MODULECONFIG
INFO | ??:??:?? 20 [GPS] getFromRadio=STATE_SEND_MODULECONFIG
INFO | ??:??:?? 21 [GPS] getFromRadio=STATE_SEND_MODULECONFIG
INFO | ??:??:?? 21 [GPS] getFromRadio=STATE_SEND_MODULECONFIG
INFO | ??:??:?? 21 [GPS] getFromRadio=STATE_SEND_MODULECONFIG
INFO | ??:??:?? 21 [GPS] getFromRadio=STATE_SEND_MODULECONFIG
INFO | ??:??:?? 21 [GPS] getFromRadio=STATE_SEND_MODULECONFIG
INFO | ??:??:?? 21 [GPS] getFromRadio=STATE_SEND_MODULECONFIG
INFO | ??:??:?? 21 [GPS] getFromRadio=STATE_SEND_MODULECONFIG
INFO | ??:??:?? 21 [GPS] getFromRadio=STATE_SEND_MODULECONFIG
INFO | ??:??:?? 21 [GPS] getFromRadio=STATE_SEND_MODULECONFIG
INFO | ??:??:?? 21 [GPS] getFromRadio=STATE_SEND_OTHER_NODEINFOS
INFO | ??:??:?? 21 [GPS] nodeinfo: num=0xcfad94fa, lastseen=1725485826, id=!cfad94fa, name=Meshtastic 94fa
WARN | ??:??:?? 21 [GPS] Failed to find UBlox & MTK GNSS Module using baudrate 4800
DEBUG | ??:??:?? 23 [GPS] Probing for GPS at 38400
DEBUG | ??:??:?? 23 [GPS] Setting Baud to 38400
INFO | ??:??:?? 29 [GPS] Found a UBlox Module using baudrate 38400
DEBUG | ??:??:?? 30 [GPS] Module Info :
DEBUG | ??:??:?? 30 [GPS] Soft version: ROM SPG 5.10 (7b202e)
DEBUG | ??:??:?? 30 [GPS] Hard version: 000A0000
DEBUG | ??:??:?? 30 [GPS] Extensions:5
DEBUG | ??:??:?? 30 [GPS]  FWVER=SPG 5.10
DEBUG | ??:??:?? 30 [GPS]  PROTVER=34.10
DEBUG | ??:??:?? 30 [GPS]  MOD=MIA-M10Q
DEBUG | ??:??:?? 30 [GPS]  GPS;GLO;GAL;BDS
DEBUG | ??:??:?? 30 [GPS]  SBAS;QZSS
DEBUG | ??:??:?? 30 [GPS] Protocol Version:34.10
DEBUG | ??:??:?? 30 [GPS] ProtVer=34
INFO | ??:??:?? 30 [GPS] UBlox GNSS probe succeeded, using UBlox MIA-M10Q GNSS Module
WARN | ??:??:?? 31 [GPS] Unable to disable NMEA messages for M10 GPS RAM.
WARN | ??:??:?? 32 [GPS] Unable to disable NMEA messages for M10 GPS BBR.
WARN | ??:??:?? 32 [GPS] Unable to disable Info messages for M10 GPS RAM.
WARN | ??:??:?? 33 [GPS] Unable to disable Info messages for M10 GPS BBR.
WARN | ??:??:?? 33 [GPS] Unable to enable powersaving for M10 GPS RAM.
WARN | ??:??:?? 33 [GPS] Unable to enable powersaving for M10 GPS BBR.
WARN | ??:??:?? 34 [GPS] Unable to enable Jamming detection M10 GPS RAM.
WARN | ??:??:?? 34 [GPS] Unable to enable Jamming detection M10 GPS BBR.
WARN | ??:??:?? 35 [GPS] Unable to disable SBAS M10 GPS RAM.
INFO | ??:??:?? 36 [GPS] To Radio onwrite
DEBUG | ??:??:?? 36 [GPS] PACKET FROM PHONE (id=0xea6607c6 fr=0x00 to=0x28, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 priority=10)
DEBUG | ??:??:?? 36 [GPS] Enqueued local (id=0xea6607c6 fr=0x00 to=0x28, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 priority=10)
WARN | ??:??:?? 36 [GPS] Unable to disable SBAS M10 GPS BBR.
WARN | ??:??:?? 37 [GPS] Unable to enable messages for M10 GPS BBR.
WARN | ??:??:?? 38 [GPS] Unable to enable messages for M10 GPS RAM.
WARN | ??:??:?? 40 [GPS] Unable to save GNSS module configuration.
WARN | ??:??:?? 40 [GPS] GPS FactoryReset requested
INFO | ??:??:?? 41 [GPS] Saving /prefs/db.proto
DEBUG | 21:42:34 67 [GPS] GPS is not communicating, trying factory reset on next bootup.
INFO | 21:42:34 67 [GPS] getFromRadio=STATE_SEND_PACKETS
INFO | 21:42:34 67 [GPS] Saving /prefs/db.proto
INFO | 21:42:34 67 [GPS] getFromRadio=STATE_SEND_PACKETS
DEBUG | 21:42:34 67 [GPS] phone downloaded packet (id=0xb7ad4363 fr=0x28 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 WANTRESP rxtime=1725486154 hopStart=3 priority=10)
INFO | 21:42:34 67 [GPS] GPS power state moving from ACTIVE to OFF
```

fifieldt, 2024-09-05:
Thanks for the report. Do you happen to know which GPS chip is in the T-Deck Plus?

fifieldt, 2024-09-05:
I see MIA-M10Q in the logs.

MisterC925, 2024-09-06:
I'll open it up and double check (they don't list it on their site) It might be tomorrow but I'll update as soon as I have it.

el-samiyel, 2024-09-07:
I get this too, using an M10Q:

```
Probing for GPS at 115200 
Found a UBlox Module using baudrate 115200
Module Info : 
Soft version: ROM SPG 5.00 (0dbd69)
Hard version: 000A0000
Extensions:4
 FWVER=SPG 5.00
PROTVER=34.00
GPS;GLO;GAL;BDS
SBAS;QZSS
Protocol Version:34.00
ProtVer=34
Got NAK for class 06 message 8A
Unable to enable powersaving for M10 GPS RAM.
Got NAK for class 06 message 8A
Unable to enable powersaving for M10 GPS BBR.
GNSS module configuration saved!
GPS is not communicating, trying factory reset on next bootup.
Saving /prefs/db.proto
GPS power state moving from ACTIVE to OFF
```

Module is missing from the log as it cant pick it up but u-cent 2 shows it as:

Receiver information
Generation: u-blox generation 10 (000A0000)
Current FW: ROM SPG 5.00 (0dbd69)

GPSFan, 2024-09-08:
The default baud rate for the M10Q is 38400, so the config of the module has been altered if it is being detected at 115200. (The default baud for the M10S is 9600)
There was some work on the M10 init and operating in config PR #4062 which may need some looking at.

GPSFan, 2024-09-08:
Page 5 of the release notes for the 5.10 firmware:
https://content.u-blox.com/sites/default/files/u-blox-M10-ROM-5.10_ReleaseNotes_UBX-22001426.pdf
indicate changes to the power save mode from 5.0.0:
3.1.6 Power save modes (PSM) Power  save  modes  for  on/off  operation  (PSMOO)  and  for  cyclic  tracking  (PSMCT)  have  been  added. The 1 Hz cyclic tracking PSM reduces the power consumption by about 50% compared to normal mode. 
This may have some bearing, as I believe #4062 changed to PSMOO so this the corresponding command in the init sequence may generate a NAK. 
Found what I think is another problem with the code from #4062, the changed version of the VALSET_PM_RAM and VALSET_PM_BBR have a constellation config command within which will cause a receiver restart, which can take .5-.8 seconds, this will most likely make the receiver miss the next command VALSET_PM_BBR and probablu the next on too, VALSET_ITFM_RAM.
In addition, the reason for the PR #4062 was that the M10 was not observed to sleep when the PMREQ command was sent, the fix was to put the M10 into a different power save mode. The real reason for the PMREQ not working was address in a different PR #4161. I may have a go at reverting part or all of #4062.

GPSFan, 2024-09-08:
Many rabbit holes later, I loaded 2.4.3 via the web flasher, and did a full erase, 
got the following as it found the M10:

DEBUG | 20:25:36 20 [GPS] Probing for GPS at 38400 
DEBUG | 20:25:36 20 [GPS] Setting Baud to 38400
INFO  | 20:25:41 25 [GPS] Found a UBlox Module using baudrate 38400
DEBUG | 20:25:42 26 [GPS] Module Info : 
DEBUG | 20:25:42 26 [GPS] Soft version: ROM SPG 5.10 (7b202e)
DEBUG | 20:25:42 26 [GPS] Hard version: 000A0000
DEBUG | 20:25:42 26 [GPS] Extensions:4
DEBUG | 20:25:42 26 [GPS]   FWVER=SPG 5.10
DEBUG | 20:25:42 26 [GPS]   PROTVER=34.10
DEBUG | 20:25:42 26 [GPS]   GPS;GLO;GAL;BDS
DEBUG | 20:25:43 26 [GPS]   SBAS;QZSS
DEBUG | 20:25:43 26 [GPS] Protocol Version:34.10
DEBUG | 20:25:43 26 [GPS] ProtVer=34
INFO  | 20:25:47 31 [GPS] GNSS module configuration saved!
DEBUG | 20:25:47 31 [Power] Battery: usbPower=1, isCharging=1, batMv=4869, batPct=100
DEBUG | 20:25:47 31 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | 20:25:47 31 [GPS] No GPS lock
DEBUG | 20:25:47 31 [GPS] onGPSChanged() pos@0 time=1725827147 lat=0 lon=0 alt=0
INFO  | 20:25:47 31 [GPS] updatePosition LOCAL pos@0 time=1725827147 lat=0 lon=0 alt=0
DEBUG | 20:25:47 31 [GPS] Setting local position: lat=0 lon=0 time=1725827147 timestamp=0
DEBUG | 20:25:47 31 [GPS] Node status update: 4 online, 4 total

I have not been able to replicate your issue, I restarted my t-deck at least a dozen times and it always found the M10.
 
I'll keep poking at it as I have time,

fifieldt, 2024-09-09:
@MisterC925 , are you confident in flashing firmware? We could provide one with GPS_DEBUG enabled...

MisterC925, 2024-09-09:
@fifieldt I am comfortable with flashing.

Thanks

fifieldt, 2024-09-10:
OK @MisterC925 , here's the latest firmware with GPS_DEBUG enabled for the t-deck: 

http://tomfifield.net/twmesh/tdeck-2.5.0-gps-debug.bin

if you could run that and send logs that would be most helpful.

MisterC925, 2024-09-10:
Thank you, here are the logs.

[meshtastic-log-2024-09-10T04-38-57.598Z.log](https://github.com/user-attachments/files/16939583/meshtastic-log-2024-09-10T04-38-57.598Z.log)

GPSFan, 2024-09-10:
@fifeldt
Here is the interesting part of the log:

DEBUG | ??:??:?? 28 [GPS] Module Info : 
DEBUG | ??:??:?? 28 [GPS] Soft version: ROM SPG 5.10 (7b202e)
DEBUG | ??:??:?? 28 [GPS] Hard version: 000A0000
DEBUG | ??:??:?? 28 [GPS] Extensions:5
DEBUG | ??:??:?? 28 [GPS]   FWVER=SPG 5.10
DEBUG | ??:??:?? 28 [GPS]   PROTVER=34.10
DEBUG | ??:??:?? 28 [GPS]   MOD=MIA-M10Q
DEBUG | ??:??:?? 28 [GPS]   GPS;GLO;GAL;BDS
DEBUG | ??:??:?? 28 [GPS]   SBAS;QZSS
DEBUG | ??:??:?? 28 [GPS] Protocol Version:34.10
DEBUG | ??:??:?? 28 [GPS] ProtVer=34
INFO  | ??:??:?? 28 [GPS] UBlox GNSS probe succeeded, using UBlox MIA-M10Q GNSS Module
DEBUG | ??:??:?? 29 [GPS] 2C39392E39392C39392E39392C39392E39392C312A33330D0A24474E4753412C412C312C2C2C2C2C2C2C2C2C2C2C2C2C39392E39392C39392E39392C39392E39392C332A33310D0A24474E4753412C412C312C2C2C2C2C2C2C2C2C2C2C2C2C39392E39392C39392E39392C39392E39392C342A33360D0A24474E4753412C412C312C2C2C2C2C2C2C2C2C2C2C2C2C39392E39392C39392E39392C39392E39392C352A33370D0AB56205010200068A98C1
INFO  | ??:??:?? 29 [GPS] Got ACK for class 06 message 8A in 274 millis.
DEBUG | ??:??:?? 29 [GPS] B56205010200068A98C1
INFO  | ??:??:?? 29 [GPS] Got ACK for class 06 message 8A in 50 millis.
DEBUG | ??:??:?? 30 [GPS] B56205010200068A98C1
INFO  | ??:??:?? 30 [GPS] Got ACK for class 06 message 8A in 48 millis.
DEBUG | ??:??:?? 30 [GPS] B56205010200068A98C1
INFO  | ??:??:?? 30 [GPS] Got ACK for class 06 message 8A in 49 millis.
DEBUG | ??:??:?? 30 [GPS] B56205010200068A98C1
INFO  | ??:??:?? 30 [GPS] Got ACK for class 06 message 8A in 77 millis.
DEBUG | ??:??:?? 30 [GPS] B56205010200068A98C1
INFO  | ??:??:?? 30 [GPS] Got ACK for class 06 message 8A in 77 millis.
DEBUG | ??:??:?? 31 [GPS] 4E50C6C6C6F312084A424A4B4A4A487E4AC6C6C6F2DEC8F2DEF2D66D
WARN  | ??:??:?? 31 [GPS] No response for class 06 message 8A
WARN  | ??:??:?? 31 [GPS] Unable to enable Jamming detection M10 GPS RAM.
DEBUG | ??:??:?? 31 [GPS] 09C6C6C6DEC8F2DE0854E7104A4A4A4A4B6A4A4B6A487E4AC6C6C6F2DEC8F2DEF2FE
WARN  | ??:??:?? 31 [GPS] No response for class 06 message 8A
WARN  | ??:??:?? 31 [GPS] Unable to enable Jamming detection M10 GPS BBR.
DEBUG | ??:??:?? 32 [GPS] CE4FC8F2
WARN  | ??:??:?? 32 [GPS] No response for class 06 message 8A
WARN  | ??:??:?? 32 [GPS] Unable to disable SBAS M10 GPS RAM.
DEBUG | ??:??:?? 32 [GPS] 1A4A626EA9C60243438427E84E50C6C6C6F312084A424A4B4A4A487E4AC6C6C6F2DEC8F2DEF2D66D09C6C6C6DEC8F2DE0854E7104A4A4A4A4B6A4A4B6A487E4AC6C6C6F2DEC8F2DEF2FE4E50C6C6C6F312084A424A4B4A4A487E4AC6C6C6F2DEC8F2DEF2D66D09C6C6C6DEC8F2DE0854E7104A4A4A4A4B6A4A4B6A487E4AC6C6C6F2DEC8F2DEF2FE4E50C6C6C6F312084A424A4B4A4A487E4AC6C6
WARN  | ??:??:?? 33 [GPS] No response for class 06 message 8A
WARN  | ??:??:?? 33 [GPS] Unable to disable SBAS M10 GPS BBR.
DEBUG | ??:??:?? 33 [GPS] C6F2DEC8F2DEF2D66D09C6C6C6DEC8F2DE0854E7104A4A4A4A4B6A4A4B6A487E4AC6C6C6F2DEC8F2DEF2FE4E50C6C6C6F312084A424A4B4A4A487E4AC6C6C6F2DEC8F2DEF2D66D09C6C6C6DEC8F2DE0854E7104A4A4A4A4B6A4A4B6A487E4AC6C6C6F2DEC8F2DEF2FE4E50C6C6C6F312084A424A4B4A4A487E4AC6C6C6F2DEC8F2DEF2D66D09C6C6C6DEC8F2DE0854E7104A4A4A4A4B6A4A4B6A487E4AC6C6C6F2DEC8F2DE
WARN  | ??:??:?? 34 [GPS] No response for class 06 message 8A
WARN  | ??:??:?? 34 [GPS] Unable to enable messages for M10 GPS BBR.
DEBUG | ??:??:?? 34 [GPS] F2FE4E50C6C6C6F312084A424A4B4A4A487E4AC6C6C6F2DEC8F2DEF2D66D09C6C6C6DEC8F2DE0854E7104A4A4A4A4B6A4A4B6A487E4AC6C6C6F2DEC8F2DEF2FE
WARN  | ??:??:?? 34 [GPS] No response for class 06 message 8A
WARN  | ??:??:?? 34 [GPS] Unable to enable messages for M10 GPS RAM.
DEBUG | ??:??:?? 35 [GPS] CE4FC8F21A4A626EA9C60243438427E84E50C6C6C6F312084A424A4B4A4A487E4AC6C6C6F2DEC8F2DEF2D66D09C6C6C6DEC8F2DE0854E7104A4A4A4A4B6A4A4B6A487E4AC6C6C6F2DEC8F2DEF2FE4E50C6C6C6F312084A424A4B4A4A487E4AC6C6C6F2DEC8F2DEF2D66D09C6C6C6DEC8F2DE0854E7104A4A4A4A4B6A4A4B6A487E4AC6C6C6F2DEC8F2DEF2FE4E50C6C6C6F312084A424A4B4A4A487E4AC6C6C6F2DEC8F2DEF2D66D09C6C6C6DEC8F2DE0854E7104A4A4A4A4B6A4A4B6A487E4AC6C6C6F2DEC8F2DEF2FE4E50C6C6C6F312084A424A4B4A4A487E4AC6C6C6F2DEC8F2DEF2D66D09C6C6C6DEC8F2DE0854E7104A4A4A4A4B6A4A4B6A487E4AC6C6C6F2DEC8F2DEF2FE
WARN  | ??:??:?? 36 [GPS] No response for class 06 message 09
WARN  | ??:??:?? 36 [GPS] Unable to save GNSS module configuration.

All commands are ACKd untill:
 "WARN  | ??:??:?? 31 [GPS] No response for class 06 message 8A
  WARN  | ??:??:?? 31 [GPS] Unable to enable Jamming detection M10 GPS BBR."

As I noted before, the new config is sending a command that results in a receiver reset, and following commands may be missed since it takes a long time for the receiver to come back online. 

First thing to try is add a .75 sec delay after the VALSET_PM_RAM and VALSET_PM_BBR commands.
I won't be around for the next few days, DayJob calls, ;>))

MisterC925, 2024-09-10:
No worries, let me know when anything else to try pops up.

varlocko, 2024-09-11:
I can try firmware if you need a guinea pig

quadjojo, 2024-09-12:
also got a T-Deck Plus and volunteering to try firmware if needed

GPSFan, 2024-09-12:
@varlocko Could you post a log of your startup please.

lewisxhe, 2024-09-13:
The latest version 2.5.1 can find the GPS, but it takes a long time to find the correct baud rate

lewisxhe, 2024-09-13:
If the device loses power and restarts, it still needs to find the GPS again because the baud rate is not locked at 9600, but restored to 38400.

```
«    @ INFO  | ??:??:?? 5  

//\ E S H T /\ S T / C

  INFO  | ??:??:?? 5  Booted, wake cause 0 (boot count 1), reset_reason=reset
  DEBUG | ??:??:?? 5  Filesystem files (16384/1048576 Bytes):
  DEBUG | ??:??:?? 5   /prefs/channels.proto (57 Bytes)
  DEBUG | ??:??:?? 5   /prefs/config.proto (163 Bytes)
  DEBUG | ??:??:?? 5   /prefs/db.proto (373 Bytes)
  DEBUG | ??:??:?? 6  Using analog input 4 for battery level
  INFO  | ??:??:?? 6  ADCmod: ADC Characterization based on Two Point values and fitting curve coefficients stored in eFuse
  INFO  | ??:??:?? 6  Scanning for i2c devices...
  DEBUG | ??:??:?? 6  Scanning for I2C devices on port 1
  DEBUG | ??:??:?? 6  I2C device found at address 0x40
  DEBUG | ??:??:?? 6  Wire.available() = 2
  DEBUG | ??:??:?? 6  Register MFG_UID: 0xffff
  INFO  | ??:??:?? 6  INA219 sensor found at address 0x40
  DEBUG | ??:??:?? 6  I2C device found at address 0x55
  INFO  | ??:??:?? 6  T-Deck keyboard found
  DEBUG | ??:??:?? 6  I2C device found at address 0x5d
  INFO  | ??:??:?? 6  LPS22HB sensor found
  INFO  | ??:??:?? 6  3 I2C devices found
  DEBUG | ??:??:?? 6  acc_info = 0
  DEBUG | ??:??:?? 6  found i2c sensor meshtastic_TelemetrySensorType_INA219
  DEBUG | ??:??:?? 6  found i2c sensor meshtastic_TelemetrySensorType_LPS22
  DEBUG | ??:??:?? 6  No SD_MMC card detected
  INFO  | ??:??:?? 6  S:B:50,2.5.1.3d72fbb1
  DEBUG | ??:??:?? 6  Total heap: 272272
  DEBUG | ??:??:?? 6  Free heap: 230468
  DEBUG | ??:??:?? 6  Total PSRAM: 8386295
 [34m DEBUG | ??:??:?? 6  Free PSRAM: 8386295
  DEBUG | ??:??:?? 6  NVS: UsedEntries 75, FreeEntries 555, AllEntries 630, NameSpaces 4
  DEBUG | ??:??:?? 6  Setup Preferences in Flash Storage
  DEBUG | ??:??:?? 6  Number of Device Reboots: 5
  DEBUG | ??:??:?? 6  No OTA firmware available
  INFO  | ??:??:?? 6  Initializing NodeDB
  INFO  | ??:??:?? 6  Loading /prefs/db.proto
  INFO  | ??:??:?? 6  Loaded /prefs/db.proto successfully
  INFO  | ??:??:?? 6  Loaded saved devicestate version 23, with nodecount: 1
  INFO  | ??:??:?? 6  Loading /prefs/config.proto
  INFO  | ??:??:?? 6  Loaded /prefs/config.proto successfully
  INFO  | ??:??:?? 6  Loaded saved config version 23
  ERROR | ??:??:?? 6  Could not open / read /prefs/module.proto
  INFO  | ??:??:?? 6  Installing default ModuleConfig
  INFO  | ??:??:?? 6  Loading /prefs/channels.proto
  INFO  | ??:??:?? 6  Loaded /prefs/channels.proto successfully
  INFO  | ??:??:?? 6  Loaded saved channelFile version 23
  ERROR | ??:??:?? 6  Could not open / read /oem/oem.proto
  DEBUG | ??:??:?? 6  cleanupMeshDB purged 0 entries
  DEBUG | ??:??:?? 6  Using nodenum 0x99bb9020 
  DEBUG | ??:??:?? 6  Old Pubkey:   fe   f9   e5   8e   99   a6   93   db   89   b5   5c   a7   65   12   5d   04   a3   8e   da   89   29   e8   f2   44   26   0b   85   74   23   a0   fa   7a   
  INFO  | ??:??:?? 6 [32m Using saved PKI keys
  DEBUG | ??:??:?? 6  Number of Device Reboots: 5
  DEBUG | ??:??:?? 6  Expanding short PSK #1
  INFO  | ??:??:?? 6  Wanted region 0, using UNSET
  INFO  | ??:??:?? 6  Saving /prefs/db.proto
  DEBUG | ??:??:?? 6  Using GPIO00 for button
  DEBUG | ??:??:?? 6  SPI.begin(SCK=40, MISO=38, MOSI=41, NSS=9)
  DEBUG | ??:??:?? 6  TFTDisplay!
  DEBUG | ??:??:?? 6  Set Timezone to GMT0
  DEBUG | ??:??:?? 6  Read RTC time as 6
  INFO  | ??:??:?? 6  GPS power state moving from OFF to ACTIVE
  DEBUG | ??:??:?? 6  Using GPIO44 for GPS RX
  DEBUG | ??:??:?? 6  Using GPIO43 for GPS TX
  DEBUG | ??:??:?? 6  Starting audio thread
  DEBUG | ??:??:?? 6  NeighborInfoModule is disabled
  DEBUG | ??:??:?? 6  Trackball GPIO initialized (3, 15, 1, 2, 0)
[... 150 lines cut ...]
  DEBUG | 06:16:40 36 [NodeInfoModule]  No modules interested in portnum=4, src=LOCAL
  DEBUG | 06:16:40 36 [NodeInfoModule]  localSend to channel 0
  DEBUG | 06:16:40 36 [NodeInfoModule]  Add packet record (id=0x55157f94 fr=0x20 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=4 WANTRESP rxtime=1726208200 priority=10)
  DEBUG | 06:16:40 36 [NodeInfoModule]  Expanding short PSK #1
  DEBUG | 06:16:40 36 [NodeInfoModule]  Using AES128 key!
  WARN  | 06:16:40 36 [NodeInfoModule]  send - lora tx disable because RegionCode_Unset
  INFO  | 06:16:55 51 [DeviceTelemetryModule]  (Sending): air_util_tx=0.000000, channel_utilization=0.000000, battery_level=101, voltage=4.231000, uptime=51
  DEBUG | 06:16:55 51 [DeviceTelemetryModule]  Partially randomized packet id 3911978901
  DEBUG | 06:16:55 51 [DeviceTelemetryModule]  updateTelemetry LOCAL
  DEBUG | 06:16:55 51 [DeviceTelemetryModule]  Node status update: 1 online, 1 total
  INFO  | 06:16:55 51 [DeviceTelemetryModule]  Sending packet to mesh
  DEBUG | 06:16:55 51 [DeviceTelemetryModule]  Update DB node 0x99bb9020, rx_time=1726208215
  DEBUG | 06:16:55 51 [DeviceTelemetryModule]  handleReceived(LOCAL) (id=0xe92c0f95 fr=0x20 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 rxtime=1726208215 priority=10)
  DEBUG | 06:16:55 51 [DeviceTelemetryModule]  No modules interested in portnum=67, src=LOCAL
  DEBUG | 06:16:55 51 [DeviceTelemetryModule]  localSend to channel 0
  DEBUG | 06:16:55 51 [DeviceTelemetryModule]  Add packet record (id=0xe92c0f95 fr=0x20 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 rxtime=1726208215 priority=10)
  DEBUG | 06:16:55 51 [DeviceTelemetryModule]  Expanding short PSK #1
  DEBUG | 06:16:55 51 [DeviceTelemetryModule]  Using AES128 key!
  WARN  | 06:16:55 51 [DeviceTelemetryModule]  send - lora tx disable because RegionCode_Unset
[0m  DEBUG | 06:16:56 52 [Power]  Battery: usbPower=1, isCharging=1, batMv=4236, batPct=100
  INFO  | 06:17:07 63 [PowerFSM]  Turning off screen
  DEBUG | 06:17:16 72 [Power]  Battery: usbPower=1, isCharging=1, batMv=4255, batPct=100
 

```

GPSFan, 2024-09-13:
At several lines before 6:16:37 you will see:

  INFO  | ??:??:?? 27 [GPS]  UBlox GNSS probe succeeded, using UBlox MIA-M10Q GNSS Module
  WARN  | ??:??:?? 29 [GPS]  Unable to disable NMEA messages for M10 GPS RAM.
  WARN  | ??:??:?? 32 [GPS]  Got NAK for class 06 message 09
  WARN  | ??:??:?? 32 [GPS]  Unable to save GNSS module configuration.

the failure is due to the config command within which will cause a receiver restart, which can take .5-.8 seconds, this will most likely make the receiver miss the next command VALSET_PM_BBR and probably the next on too, VALSET_ITFM_RAM. as well as the config save command.

GPSFan, 2024-09-13:
I was able to partially replicate the above problem with an M10S at 9600 baud on my Chatter 2 platform. I'll try to revert the commit that caused the problem when I get some time, maybe as late as Mon of Tues, sorry too much DayJob... The M10's backup battery had run down so it reverted to defaults, that's why I couldn't replicate it on the t-deck, it had been configured properly and the backup battery had never run down so it maintained it's config after teh t-deck was updated to 2.5.

fifieldt, 2024-09-17:
@MisterC925 here is a new firmware with a few more waits built in - if you have time to test and post logs that would be appreciated:
https://tomfifield.net/twmesh/tdeck-2.5.1-gps-750wait-debug.bin

MisterC925, 2024-09-17:
Hi I just flashed it and did a power cycle.  No GPS present.  Attached is the log.

Thanks!
[meshtastic-log-2024-09-17T20-42-07.258Z.log](https://github.com/user-attachments/files/17034779/meshtastic-log-2024-09-17T20-42-07.258Z.log)

GPSFan, 2024-09-17:
well, that certainly made it worse... ;>((

fifieldt, 2024-09-18:
Thanks @MisterC925 , that was actually an improvement. The chip was detected flawlessly and a couple of the setup messages did succeed where they didn't before. I'm now going to try adding the delay on all of the setup messages. It means it will be slower to get setup, but more reliable. I think this one has a good chance of working, if you have time to keep testing:

http://tomfifield.net/twmesh/firmware-2.5.1-tdeck-all-delays-debug.bin

If this one works, then will ship you one without DEBUG turned on :)

GPSFan, 2024-09-18:
@fifieldt I figured out what is happening, when the VALSET_PM_RAM command is sent, it causes a receiver restart, since the config had not been saved yet, the receiver reverts to 38400 baud, poor Meshtastic is still at 9600 so nothing works from then on.
Try reverting https://github.com/meshtastic/firmware/pull/4062 and also change the message_SAVE command to have a device mask of 0x01 vs 0x17. (the M10 does not like trying to save to non-extistant  eeprom or SPI flash)

fifieldt, 2024-09-19:
Awesome! @GPSFan , do you know why we go to 9600 for Ublox? Can we keep it at whatever it's detected at (eg 38400) instead?

GPSFan, 2024-09-19:
That was before my time, 9600 has been default for lots of receivers for a very long time, when there was less data to send every second. With small buffers, low processing power (esp32 @80MHz) and low baud rate, limiting the number of sentences is critical.
Sorry for the late responses, I only have about an hour a day to look at Meshtastic stuff these days.

MisterC925, 2024-09-19:
Still comes up no gps present.  I tried a couple resets just to see and they all failed.

[meshtastic-log-2024-09-19T18-58-30.493Z.log](https://github.com/user-attachments/files/17065071/meshtastic-log-2024-09-19T18-58-30.493Z.log)

GPSFan, 2024-09-19:
@MisterC925 what firmware is that log from, the one @fifieldt sent you on the 17th?
@fifieldt I need to look at the source that you built that firmware from, what branch is it on your github repo?

MisterC925, 2024-09-19:
> @MisterC925 what firmware is that log from, the one @fifieldt sent you on the 17th?

Yes, the firmware-2.5.1-tdeck-all-delays-debug.bin

Thanks

GPSFan, 2024-09-20:
@MisterC925 yeah, that is not going to do you any good, no matter how many times you reset it. I've built a version which works on my t-deck and I'd like you to try the .bin.
[firmware.zip](https://github.com/user-attachments/files/17069137/firmware.zip)
Please post the log.

MisterC925, 2024-09-20:
Thanks

It came right up with the gps present (I'm inside so no sats I'll take it outside later

[meshtastic-log-2024-09-20T17-05-09.236Z.log](https://github.com/user-attachments/files/17078414/meshtastic-log-2024-09-20T17-05-09.236Z.log)

fifieldt, 2024-09-22:
@GPSFan , sorry for the late reply - I'm working over here: https://github.com/fifieldt/meshtastic-firmware/tree/ublox-cleanup

As you're busy, if you run a diff I'm happy to do the git stuff :)

GPSFan, 2024-09-22:
@fifieldt I'll get you something tomorrow, not a big change, but the reasons for the reversion of that PR go deeper than this bug.

closed 2024-09-24 by thebentern

captainchaosrules, 2024-10-04:
> @MisterC925 yeah, that is not going to do you any good, no matter how many times you reset it. I've built a version which works on my t-deck and I'd like you to try the .bin. [firmware.zip](https://github.com/user-attachments/files/17069137/firmware.zip) Please post the log.

It's locking up for me....
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3
Features: Wi-Fi,BLE
Crystal is 40MHz
MAC: 48:ca:43:31:46:70
Uploading stub...
Running stub...
Stub running...

mechovation, 2024-10-04:
T-Deck Plus, flashed 2.5.4.8d288d5, then went into configuration set GPS pins 44 for RX and 43 for TX and enabled GPS.  Seems to be working.

devinberg, 2024-11-19:
Not sure if this is new information. However, I recently acquired a T-Deck Plus, which came with firmware 2.5.8 installed and GPS working fine. I just flashed 2.5.11.8e2a3e5 and GPS was not working by default. I was able to manually enable GPS and configure pins 44 and 43 as described above to get it working.

dzienisz, 2025-08-28:
Soo why is that? Can't it be fixed?