meshtastic/firmware #4626: [Bug]: T1000E - GPS is not configured

fifieldt, 2024-09-05:
### Category

Hardware Compatibility

### Hardware

Other

### Firmware Version

2.5.0.ff40a3f1

### Description

Meshtastic only uses RMC & GGA NMEA sequences, and there is code in gps.cc for the AG3335 to disable the others. However, the GPS in the T-1000E is producing all of: RMC, VTG, GGA, GSA, GSV, GLL, ZDA in the logs. This increases the amount of processing that the gps chip and firmware has to do, which probably has a negative impact on power consumption.

### Relevant log output

```Shell
DEBUG | ??:??:?? 122 phone downloaded packet (id=0x6ca90faa fr=0x78 to=0xff, WantAck=0, HopLim=0 Ch=0x0 Portnum=67 rxSNR=-4.5 rxRSSI=-108 hopStart=5)

DEBUG | ??:??:?? 122 [GPS] $GNGGA,000008.012,,,,,0,0,,,M,,M,,*5D#

$GNGLL,,,,,000008.012,V,N*6F#

$GNGSA,A,1,,,,,,,,,,,,,,getFromRadio=STATE_SEND_PACKETS

DEBUG | ??:??:?? 122 [GPS] phone downloaded packet (id=0x5a5bc45c fr=0x58 to=0xff, WantAck=0, HopLim=0 Ch=0x0 Portnum=67 rxSNR=-4.75 rxRSSI=-108 hopStart=3)

DEBUG | ??:??:?? 122 [GPS] ,,1*1D#

$GNGSA,A,1,,,,,,,,,,,,,,,,2*1E#

$GNGSA,A,1,,,,,,,,,,,,,,,,3*1F#

$GNGSA,A,1,,,,,,,,,,,,,,,,4*18#

$GNGSA,A,1,,,,,,,,,,,,,,,,6*1A#

$GPGSV,1,1,00,1*64#

$GLGSV,1,1,00,1*78#

$GAGSV,1,1,00,7*73#

$GBGSV,1,1,00,1*76#

$GIGSV,1,1,00,1*7D#

$GNRMC,000008.012,V,,,,,,,060180,,,N,V*2D#

$GNVTG,,T,,M,,N,,K,N*32#

$GNZDA,000008.012,06,01,1980,,*44#

getFromRadio=STATE_SEND_PACKETS

DEBUG | ??:??:?? 122 phone downloaded packet (id=0x5a5bc45d fr=0x58 to=0xff, WantAck=0, HopLim=0 Ch=0x0 Portnum=3 rxSNR=-5.25 rxRSSI=-108 hopStart=3)

INFO | ??:??:?? 123 getFromRadio=STATE_SEND_PACKETS
```

b8b8, 2024-09-05:
Please don't turn off the other constellations when working on this fix. Its really nice to get 12 satellites while the T1000E is in my pocket.

Dylanliacc, 2024-09-05:
These commands in testing~ ,I think we will get some results as soon as possible

fifieldt, 2024-09-08:
@b8b8 - fix in that PR for you to test and make sure your pocket can still get all the satellites it wants :) I have left the constellations according to the default settings: GPS+GALILEO+NAVIC

closed 2024-09-09 by caveman99