meshtastic/firmware #4629: sync time

cracky22, 2024-09-05:
Automatic synchronization of the time even if no real-time clock is installed

cracky22, 2024-09-05:
> Automatic synchronization of the time even if no real-time clock is installed

Or a function in the Android app that allows you to set / synchronize the time manually

smurfy, 2024-09-05:
A function like this would be awesome, i have two heltec v3, without gps and one of them is not connected to the internet so cant connect to a timeserver.

If i where able to update the time via app would be great. Especially usefull as @cracky22 mentioned an automatic time sync on app connect or something.

cracky22, 2024-09-05:
> A function like this would be awesome, i have two heltec v3, without gps and one of them is not connected to the internet so cant connect to a timeserver.
>
> If i where able to update the time via app [...]

Even my Heltec V3 with GPS and RTC does not keep the time...