meshtastic/firmware #4633: When toggling the GPS by tripple pressing selection button, the state on the integrated display still show the former state

SquishyBubble, 2024-09-05:
### Category

Other

### Hardware

T-Echo

### Firmware Version

2.5.0

### Description

1. Set the display on the T-Echo to the screen showing number of connections, main channel and in the upper right corner GPS status (the second to last one).
2. Triple-press the selection button to toggle the GPS.
3. The state of the GPS is changed, but the upper right corner is still showing the old state.

The correct state will show if repeatedly pressing the selection button until fully rotated back to the same screen. This problem is present on other hardware and firmware versions as well.

todd-herbert, 2024-09-05:
> This problem is present on other hardware and firmware versions as well.

As for other hardware, do you mean other devices which use E-Ink displays, or have you noticed the problem also on devices with other types of display?

SquishyBubble, 2024-09-08:
> > This problem is present on other hardware and firmware versions as well.
> 
> As for other hardware, do you mean other devices which use E-Ink displays, or have you noticed the problem also on devices with other types of display?

Hi, sorry for the late reply. I have only seen the issue with other E-Ink displays.

PR #4637 (2024-09-06): Update E-Ink GPS icon immediately when triple-pressing user button 
Fixes #4633

Quick work-around for one special case: result of a weakness in the code which parses the OLED output into discrete E-Ink refreshes.