meshtastic/firmware #4648: What is difference between the firmware without tcxo vs those who has

kwccoin, 2024-09-07:
```
%ls firmware-tlora-v2-1-1_6* #btw the site use -1. but actually it is 1_
...
firmware-tlora-v2-1-1_6-2.4.2.5b45303.bin		firmware-tlora-v2-1-1_6-tcxo-2.4.2.5b45303.bin
```

I suspect my LilyGo 433/470 MHz  T3_V1.6.1 20210104 is not the tcxo which I found some other chip info and probably is a totally different chip.

```
LoRa1280 FCC & CE TCXO LoRa 2.4G RF module Long distance 2.4G lora RF module range measurement SX1280 chip
```

kwccoin, 2024-09-07:
Second issue is the 

```
./device-install.sh firmware-tlora-v2-1-1_6-2.4.2.5b45303.bin
```

would not work as under macOS one has to use esptool.py erase ... etc.

kwccoin, 2024-09-07:
I assume based on my reading of the shell script is that for my first 2 device both flora (but I also have T-deck and others), it would be 

	"$PYTHON" -m esptool erase_flash
	"$PYTHON" -m esptool write_flash 0x00 ${FILENAME}

or 

esptool.py erase_flash
esptool.py write_flash 0x00 firmware-tlora-v2-1-1_6-2.4.2.5b45303.bin

Please confirm and then help what should I do for T-deck.  Not very good in shell script logic.

garthvh, 2024-09-07:
sh device-install.sh -f firmware-tlora-v2-1-1_6-2.4.2.5b45303.bin

closed 2024-09-07 by garthvh