meshtastic/firmware #4655: [Bug]: Flip Screen not working on Heltec T114

TheBamse, 2024-09-08:
### Category

Other

### Hardware

Other

### Firmware Version

2.4.3.91d6612 and 2.5.0.9e55e6b

### Description

When setting the Flip Screen option via either python CLI (version 2.4.0) or Android app (version 2.4.4) to true it has no effect on device.

Tried factory erasing and a reflash with both firmware versions on both devices with identical results (an unflipped screen :))

### Relevant log output

```Shell
Example CLI --info excerpt:
 "display": {
    "screenOnSecs": 60,
    "flipScreen": true,
    "headingBold": true,
    "gpsFormat": "DEC",
    "autoScreenCarouselSecs": 0,
    "compassNorthTop": false,
    "units": "METRIC",
    "oled": "OLED_AUTO",
    "displaymode": "DEFAULT",
    "wakeOnTapOrMotion": false,
    "compassOrientation": "DEGREES_0"
  }
```

todd-herbert, 2024-09-08:
I think that setting's original intention might have been to fix display output coming out mirrored on some OLED displays, but you're keen to have it rotate the display image 180 degrees right? That'd probably something that's pretty achievable in the near future, although would have to check that no other TFT displays use it to fix mirroring.

TheBamse, 2024-09-08:
The docs is a bit ambigiuos on this;
"Flip Screen
If enabled, the screen will be rotated 180 degrees, for cases that mount the screen upside down"

It could be interpreted as mirroring but since I was after rotating it to flip it upside-down I just assumed the documentation was referring to that ;)

todd-herbert, 2024-09-08:
> The docs could perhaps be a bit ambiguous on this(?);
> If enabled, the screen will be rotated 180 degrees, for cases that mount the screen upside down"

Oh, you could totally be right on that! In any case I'll definitely look into in the near future.

closed 2024-09-20 by thebentern