meshtastic/firmware #4656: [Feature Request]: Add ability to adjust BLE advertising & connection intervals

powerfeatherdev, 2024-09-08:
### Platform

ESP32

### Description

Having the ability to adjust advertising & connection interval is good for power saving, and enables the user to determine on balance how much throughput/responsiveness loss they are willing to tolerate for that.

This is mostly in preparation for ESP32 boards that might use custom Arduino libs and external 32Khz crystal to let the BLE modem sleep in between the intervals.

powerfeatherdev, 2024-09-08:
I've done some initial work with this on my board: https://github.com/PowerFeather/meshtastic-firmware/commit/d9c760b2d75bc0989bffb46d3feb9964662a402b

But it's hardcoded right now, it would be nice if this is something that could be adjusted from the app as well.

garthvh, 2024-09-08:
BLE can't sleep, it can only be on or off.

powerfeatherdev, 2024-09-09:
BLE can do modem sleep with light sleep, as per: https://github.com/espressif/esp-idf/issues/947#issuecomment-500312453

Requires custom Arduino libs with modified sdkconfig and external 32 khz crystal, however

garthvh, 2024-09-09:
BLE is totally off in deep sleep as mentioned above n your screenshot.

powerfeatherdev, 2024-09-10:
Yes, it needs to be off during deep sleep. 

But I was responding to your statement `BLE can't sleep, it can only be on or off.`, where clearly it can do shallower sleep levels, like modem sleep and light sleep.

garthvh, 2024-09-10:
Esp32 is not using a real BLE chip, there is no sleep state for BLE.

garthvh, 2024-09-10:
Making BLE worse is not really a goal when most usage is via phone.

closed 2024-09-10 by garthvh

powerfeatherdev, 2024-09-12:
Fair enough, modem sleep not a sleep mode in a traditional sense in a microcontroller where the CPU is stopped/slowed down. But the state of turning off the RF circuit (radio and baseband) is a distinct power mode, and is even mentioned in the technical reference manual as that modem sleep:

[image]

Combine modem sleep + light sleep (where the CPU *is* clock gated) is even more power saving, which is what the ESP-IDF comment I linked to allows.

powerfeatherdev, 2024-09-13:
However, going back to the original point of this  suggestion...

> Making BLE worse is not really a goal when most usage is via phone.

 The suggestion was to allow users to increase the advertising and connection intervals to the point that 

1. they can still tolerate the throughput/latency decrease, or
2. they still won't have connection stability issues

I was not suggesting to change the defaults, which is of course, going to offer the best experience. The progress I posted was merely hardcoded because I don't know how to add these options to the firmware & app.

powerfeatherdev, 2024-09-13:
> Esp32 is not using a real BLE chip

I don't even know where this statement is coming from... (I can kind of guess knowing a little how Espressif made the ESP32's quite cheap). But if this statement is not intended to be hyperbole, then I think it's taking it too far...

garthvh, 2024-09-13:
It uses the same chip for WiFi and BLE, not a purpose built ble5 chip no.

powerfeatherdev, 2024-09-14:
That's actually a different 'reason' than what I am thinking.

But as far as I know, only the RF part of the modem is shared. 
See [this]( https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/coexist.html) for more details.

It's not a reason to think it's not a 'real' BLE chip IMO.